espressif/esp-idf #1647: AVRCP. Connection from speaker with esp32 to smartfone

KryvosheyaIvan, 2018-02-21:
There is an example a2dp_sink by means of which one can establish communication between smartfone and esp32. But what about if i want to play/stop/backward/pause/etc by means of buttons located on speaker?
I know it could be done own to bluetooth avrcp. Is it right?
Are there any examples ?

Weijian-Espressif, 2018-02-24:
@blueMoodBHD Please help to answer the question.

blueMoodBHD, 2018-02-27:
@KryvosheyaIvan Yes,it could be done own to bluetooth avrcp. These audio-related features we will open in another project, but now is not convenient to provide, thank you for understanding.

blueMoodBHD, 2018-03-20:
@KryvosheyaIvan Is there any other problem? if not , can we close this issue?

KryvosheyaIvan, 2018-03-21:
@blueMoodBHD, all OK!

closed 2018-03-21 by KryvosheyaIvan