espressif/esp-idf #1648: Symbol 'TclClass' could not be resolved

Samradnyee, 2018-02-22:
/* The AODV code developed by the CMU/MONARCH group was optimized and tuned by Samir Das and Mahesh Marina, University of Cincinnati. The work was partially done in Sun Microsystems. Modified for gratuitous replies by Anant Utgikar, 09/16/02.
*///#include <ip.h>
#include <aodv/aodv.h>
#include <aodv/aodv_packet.h>
#include <random.h>
#include <cmu-trace.h>
#include <mac.h>
#include <ip.h>
//#include <energy-model.h>

#define max(a,b)        ( (a) > (b) ? (a) : (b) )
#define CURRENT_TIME    Scheduler::instance().clock()
//#define DEBUG
//#define ERROR
#ifdef DEBUG
static int route_request = 0;
#endif

//int my_nn;

/* TCL Hook */

int hdr_aodv::offset_;
static class AODVHeaderClass : public PacketHeaderClass {
public:
        AODVHeaderClass() : PacketHeaderClass("PacketHeader/AODV",sizeof(hdr_all_aodv)) {
	    bind_offset(&hdr_aodv::offset_);
	} 
} class_rtProtoAODV_hdr;

static class AODVclass : public TclClass {
public:
        AODVclass() : TclClass("Agent/AODV") {}
        TclObject* create(int argc, const char*const* argv) {
          assert(argc == 5);
          //return (new AODV((nsaddr_t) atoi(argv[4])));
	  return (new AODV((nsaddr_t) Address::instance().str2addr(argv[4])));
        }
} class_rtProtoAODV;

int AODV::command(int argc, const char*const* argv) {

  if(argc == 2) {
  Tcl& tcl = Tcl::instance();
    
    if(strncasecmp(argv[1], "id", 2) == 0) {
      tcl.resultf("%d", index);
      return TCL_OK;
    }
/////////////////////////////////////////////////////////////////////////////////////
    if(strcmp(argv[1],"hacker")==0)  {
    	malicious=true;
    	return TCL_OK;
    }

    if(strcmp(argv[1],"leader")==0)  {
    	leader=true;
       	return TCL_OK;
       }
/////////////////////////////////////////////////////////////////////////////////////

    if(strncasecmp(argv[1], "start", 2) == 0) {
    	///////////////////sandycode//////////////////
    	//printf("\n65%d",my_nn);
    	//bind("my_nn", &my_nn);
    	//my_nn=50;
    	//printf("\n65%d",my_nn);
    	///////////////////sandycode//////////////////

      btimer.handle((Event*) 0);
////////////////////////////////////////////////////////////////////////////////////
      maltimer.handle((Event*) 0);// code for malicious node
////////////////////////////////////////////////////////////////////////////////////
      uncertimer.handle((Event*) 0); // code for uncertainty timer.
      leadertimer.handle((Event*) 0); //  leader declaration _timer.
////////////////////////////////////////////////////////////////////////////////////
#ifndef AODV_LINK_LAYER_DETECTION
        htimer.handle((Event*) 0);
        ntimer.handle((Event*) 0);
#endif // LINK LAYER DETECTION

      rtimer.handle((Event*) 0);
      return TCL_OK;
     }               
  }
  else if(argc == 3) {
    if(strcmp(argv[1], "index") == 0) {
      index = atoi(argv[2]);
      return TCL_OK;
    }

    else if(strcmp (argv[1],"install-tap")==0){  // newly added for promiscuous mode
    					mac_ =(Mac*)TclObject::lookup(argv[2]);
    					if(mac_==0)return TCL_ERROR;
    					mac_->installTap(this);
    					return TCL_OK;
    	}

    else if(strcmp(argv[1], "log-target") == 0 || strcmp(argv[1], "tracetarget") == 0) {
      logtarget = (Trace*) TclObject::lookup(argv[2]);
      if(logtarget == 0)
    	return TCL_ERROR;
      return TCL_OK;
    }
    else if(strcmp(argv[1], "drop-target") == 0) {
    int stat = rqueue.command(argc,argv);
      if (stat != TCL_OK) return stat;
      return Agent::command(argc, argv);
    }
    else if(strcmp(argv[1], "if-queue") == 0) {
    ifqueue = (PriQueue*) TclObject::lookup(argv[2]);
      
      if(ifqueue == 0)
	return TCL_ERROR;
      return TCL_OK;
    }
    else if (strcmp(argv[1], "port-dmux") == 0) {
    	dmux_ = (PortClassifier *)TclObject::lookup(argv[2]);
	if (dmux_ == 0) {
		fprintf (stderr, "%s: %s lookup of %s failed\n", __FILE__,
		argv[1], argv[2]);
		return TCL_ERROR;
	}
	return TCL_OK;
    }
  }
  return Agent::command(argc, argv);
}

igrr, 2019-04-10:
@Samradnyee Do you still have this issue? If yes, please provide the code required to reproduce it.

Alvin1Zhang, 2019-10-12:
@Samradnyee Thanks for reporting and sorry for the slow turnaround. Would you please help more details to reproduce the issue? Thanks.

Alvin1Zhang, 2020-08-05:
Thanks for reporting, will close now, feel free to reopen with more details/updates. Thanks.

closed 2020-08-05 by Alvin1Zhang