espressif/esp-idf #1649: Error when try to Get ESP-IDF

mikekgr, 2018-02-22:
Dear Sirs,
today I did a new installation in my Windows 10 64bit laptop, following the instructions: https://esp-idf.readthedocs.io/en/latest/get-started/windows-setup.html
What I did:
1) I download the Toolchain using  ==> https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20180110.zip
2) I unzipped / copied the including msys2 directory to my drive e:  so after a while I got E:\msys32
3) I started E:\msys32\mingw32.exe and the bash started normally
4) I proceed to section Get ESP-IDF: ==> 
https://esp-idf.readthedocs.io/en/latest/get-started/index.html#get-started-get-esp-idf
5) then inside the mingw32 bash terminal I gave the commands: 
cd ~/esp
git clone --recursive https://github.com/espressif/esp-idf.git
As you can see bellow the git clone --recursive does not worked despite my many tries and Laptop shutdown / restart etc 
Bellow you can see the error

kranidis@KRANIDIS MINGW32 ~/esp
$ pwd
/home/kranidis/esp

kranidis@KRANIDIS MINGW32 ~/esp
$ git clone --recursive https://github.com/espressif/esp-idf.git
Cloning into 'esp-idf'...
      1 [main] git 10360 child_info_fork::abort: E:\msys32\usr\bin\msys-iconv-2.dll: Loaded to different address: parent(0x3040000) != child(0x3050000)
error: cannot fork() for git-remote-https: Resource temporarily unavailable

Please instruct me what to do in order to solve the problem.

Thanks and Best Regards,
Mike Kranidis

negativekelvin, 2018-02-22:
You can try rebooting, rebaseall, or turn off aslr in Windows defender

mikekgr, 2018-02-22:
@negativekelvin  thanks a lot for the tips.
The rebooting does nothing unfortunately... What is the  rebaaseall  that you suggested me?
I will try aslr off at Windows Defender tomorrow when I will be in my Windows 10 laptop.
I tested the "problematic" installation, using Windows 7 64bit laptop this time ( I just put the external USB hard disk ) and everything worked fine so yes, something wrong with Windows 10 64bit ...
Again, thanks a lot for your kind support.
Best Regards,
Mike Kranidis

negativekelvin, 2018-02-22:
If you Google mingw rebaseall you will find some guides

negativekelvin, 2018-02-23:
There is an autorebase.bat file in msys2 directory

mikekgr, 2018-02-23:
Dear @negativekelvin 
Y E S   this  did the trick !!!
I strongly appreciated your fine support. If someday come to Athens/Greece I will buy your dinner !
I will commute this solution to esp32.com forum to be a very fine sollution for everybody like me.

Very Best Regards,
Mike Kranidis

mikekgr, 2018-02-23:
Using the @negativekelvin suggestion, everything is working as it should now!

closed 2018-02-23 by mikekgr

Milan991, 2018-12-08:
In my case in's the Avast antivirus, when I've disabled it and everything was ok.