espressif/esp-idf #1651: Deadlock in spi_master.c spi_device_transmit

vonnieda, 2018-02-22:
I am experiencing a deadlock in spi_device_transmit(). The deadlock happens while waiting for the transaction response, in this code from spi_master.c:

```
esp_err_t spi_device_get_trans_result(spi_device_handle_t handle, spi_transaction_t **trans_desc, TickType_t ticks_to_wait)
{
    BaseType_t r;
    spi_trans_priv trans_buf;
    
    SPI_CHECK(handle!=NULL, "invalid dev handle", ESP_ERR_INVALID_ARG);
    // DEADLOCK HERE
    r=xQueueReceive(handle->ret_queue, (void*)&trans_buf, ticks_to_wait);
```

This only seems to happen when I am running many transactions on both the HSPI_HOST and VSPI_HOST.

I have previously reported this problem in this thread: https://esp32.com/viewtopic.php?f=2&t=4374 but did not get any further responses.

Initially, I was able to get around this issue by using a mutex on all my SPI access, so I would never use both HSPI_HOST and VSPI_HOST at the same time. Without that mutex the deadlock happens almost immediately. With the added mutex it is much more rare, but I can produce it within about 2 minutes by running thousands of transactions on both busses.

At the most basic, it seems like the issue is that the transaction gets submitted to the controller but then the response is never received. Since I am using SPI master, I'm not sure what the conditions are where this could happen. The master should either successfully receive the response or it should time out.

I'm happy to provide any further detail you need, please just let me know.

negativekelvin, 2018-02-23:
Does it only happen when one spi is using dma and one is not? Does it happen in single core mode? What commit is your idf on?

vonnieda, 2018-02-23:
Hi @negativekelvin, thank you for your response. Here's some answers:

1. Yes, happens when one is using DMA and one is not, or when both are using DMA. My normal use case is the SPI Flash using DMA and the LCD is not. I typically only send commands to the LCD, no data, so I don't use DMA and my transfer length is 0.

2. I normally run two cores. I tested today with one core and that does indeed seem to fix it.

3. IDF on a5eb369126e5cef31715d2755931cf933bb98bee

I also tested a few other things this morning, and I noticed that if I add `vTaskDelay(2 / portTICK_PERIOD_MS);` between calls to `spi_device_transmit` it improves the issue, but does not completely fix it. Instead of it locking up within a second or two, it takes 10-15 seconds.

I'm very interested to know if any of those points give you any ideas as to what is happening.

Thanks!

vonnieda, 2018-02-23:
Update on this: I converted my LCD code entirely to DMA so that both the SPI Flash and LCD are both using DMA and this seems to have either solved or hidden the problem. I am no longer getting deadlocks even while accessing both busses at the same time. 

For what it's worth, the biggest change that this made is that I now send my updates about 4x faster, so I am holding the bus for less time, and I am sending far fewer total transactions. Previously I would send about 532 single byte transactions per frame, and now I send 8.

I'll close this, and reopen it if I start to notice the problem again.

vonnieda, 2018-02-23:
I spoke too soon, a few minutes later it locked again. So, I've switched from spi_device_transmit to calling spi_device_queue_trans and spi_device_get_trans_result myself, so that I can specify a shorter timeout and try to recover. That eventually generated the following error:

```Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x400820be  PS      : 0x00060933  A0      : 0x8013f468  A1      : 0x3ffde9b0  
0x400820be: esp_intr_enable at /Users/jason/esp/esp-idf/components/esp32/./intr_alloc.c:768

A2      : 0x3ffdfcd8  A3      : 0x00000000  A4      : 0x3ffcec50  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0xbed194f7  A9      : 0x3ffde980  
A10     : 0x00000001  A11     : 0x3ffdeb4c  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060920  A15     : 0x00000000  SAR     : 0x00000005  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbed194f7  LBEG    : 0x40099c64  LEND    : 0x40099c92  LCOUNT  : 0xffffffff  
0x40099c64: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:168

0x40099c92: memcpy at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memcpy.S:201

Backtrace: 0x400820be:0x3ffde9b0 0x4013f465:0x3ffde9d0 0x400dba82:0x3ffdea10 0x400dc89c:0x3ffdea70 0x400dd82d:0x3ffdeaa0 0x400dd8ba:0x3ffdeac0
0x400820be: esp_intr_enable at /Users/jason/esp/esp-idf/components/esp32/./intr_alloc.c:768

0x4013f465: spi_device_queue_trans at /Users/jason/esp/esp-idf/components/driver/./spi_master.c:725

0x400dba82: fb_hal_lcd_send_buffer_unsafe at /Users/jason/Projects/xxx/xxx-esp32-firmware/main/./fb_hal_v1r1.c:1043

0x400dc89c: fb_hal_lcd_flush at /Users/jason/Projects/xxx/xxx-esp32-firmware/main/./fb_hal_v1r1.c:1203

0x400dd82d: draw_screen at /Users/jason/Projects/xxx/xxx-esp32-firmware/main/./fb_lcd.c:294

0x400dd8ba: fb_lcd_refresh_task at /Users/jason/Projects/xxx/xxx-esp32-firmware/main/./fb_lcd.c:461

Rebooting...
```

My code for sending a transaction to the LCD is below:

```
static esp_err_t fb_hal_lcd_send_buffer_unsafe(uint8_t* data, size_t data_len, bool command) {
	esp_err_t err = gpio_set_level(LCD_IO_A0, command ? 0 : 1);
	if (err != ESP_OK) {
		return err;
	}

	memcpy(lcd_tx_buffer, data, data_len);

	spi_transaction_ext_t t = {
			.base = {
				.length = data_len * 8,
				.flags = SPI_TRANS_VARIABLE_ADDR
					| SPI_TRANS_VARIABLE_CMD,
				.tx_buffer = lcd_tx_buffer,
				.rx_buffer = lcd_rx_buffer,
			},
			.command_bits = 0,
			.address_bits = 0,
	};

    err = spi_device_queue_trans(lcd_spi_device, (spi_transaction_t*) &t, 1000 / portTICK_PERIOD_MS);
    if (err != ESP_OK) {
    		FB_LOG_ERROR(err, "spi_device_queue_trans");
    		return err;
    }

	spi_transaction_t *ret_trans;
    err = spi_device_get_trans_result(lcd_spi_device, &ret_trans, 1000 / portTICK_PERIOD_MS);
    if (err != ESP_OK) {
    		FB_LOG_ERROR(err, "spi_device_get_trans_result");
    		return err;
    }

//    err = spi_device_transmit(lcd_spi_device, (spi_transaction_t*) &t);
//	if (err != ESP_OK) {
//		return err;
//	}

	return ESP_OK;
}

```

So now what happens is after a few minutes I get a timeout error on `spi_device_get_trans_result` and then within a few seconds I get the above core dump.

vonnieda, 2018-02-23:
I added some code to attempt to reinitialize the bus when this timeout happens, and got some interesting errors:

```
E (58417) FB_HAL_V1R (0107) spi_device_get_trans_result
E (58417) FB_HAL_V1R (0107) LCD bus failure detected, attempting to reinitialize
E (58427) spi_master: spi_bus_remove_device(317): Have unfinished transactions
E (58427) spi_master: spi_bus_free(221): not all CSses freed
E (58437) spi_master: spi_bus_initialize(131): host already in use
```

and here is the code that produced those errors:

```
	if (lcd_spi_device) {
		spi_bus_remove_device(lcd_spi_device);
		lcd_spi_device = NULL;
		spi_bus_free(HSPI_HOST);
	}
```

So it seems the transaction is still in the queue, but it's not completing, and I can't de-init the bus to try again. So how do I handle this situation?

negativekelvin, 2018-02-23:
You are still unable to reproduce in single core mode?  What about if all spi threads pinned to same core?

vonnieda, 2018-02-23:
@negativekelvin Correct, it works fine in single core mode. I'll try pinning the threads, but that will take a bit since the SPI Flash access is coming from the VFS layer and I have several threads accessing it.

In the meantime, can you tell me if there is a reason for the questions regarding two core? Is there a known issue, or something you've seen before?

Thanks,
Jason

negativekelvin, 2018-02-23:
There have been some SMP issues with dport and interrupts so running related peripheral interrupts on separate cores could trigger something. Will have to ask some experts...
@ginkgm @Spritetm @igrr

vonnieda, 2018-02-24:
@negativekelvin I've confirmed that if I run the app in single core mode, or if I pin all tasks that use SPI to a single core in two core mode it works fine. So, it does seem to relate to SMP and interrupts, as you said.

ginkgm, 2018-03-01:
Hi Jason @vonnieda ,
I've done a simple test: two tasks access the bus (to the same pin but one HSPI and another VSPI) continuously, using the LCD example and it switches between tasks fluently.
So I wonder if the problem is in the spi flash part.
Can you explain the lib/functions you used to access the spi flash more detailedly??

vonnieda, 2018-03-01:
Hi @ginkgm, I will try to put together a self contained example today and post it here. In the meantime, I want to mention that when this happens it has never been the SPI Flash. It's always the LCD. I've checked this both with logging and with JTAG, and it is always the LCD thread locking up.

ginkgm, 2018-03-02:
Hi @vonnieda ,
  thanks, that would help us a lot to know the cause.
   But as far as i know, ESP-IDF don't have library supporting SPI flash on HSPI or VSPI, the `esp_flash_` functions are used to access the ROM flash which holds the code. And sometimes accessing the flash involves a pause to code in a wide-range (iirc all the code except those ISR in the IRAM) since you cannot fetch code from the flash until the flash is ready again.

   If you're using your own flash lib, maybe you can check the SPI controller you're using in the flash lib, or paste them here for us to help you.

vonnieda, 2018-03-02:
Hi @ginkgm, I am using my own SPI flash access routines. I will post these as part of the example as soon as I have it ready. Right now I am trying to get it to reproduce on a dev board (WROVER-KIT) so that it will be easy for you to test and reproduce.

In summary, here is what we have:
LCD: ST7565 on HSPI_HOST @ 1 MHz
SPI FLASH: W25Q128 on VSPI_HOST @ 40 MHz

No other devices are on either bus.

The SPI FLASH is used for a LittleFS filesystem via VFS, also using my own routines.

Sorry for the delay in getting the example posted. I'll get it up as soon as I can. If you'd like to close this until I am able to provide a test case, I understand.

vonnieda, 2018-04-20:
Hi @negativekelvin, @ginkgm, @Spritetm, @igrr: I am now experiencing this at a much higher frequency on both SPI busses, and I'm at a loss to understand why. Here is a short summary of the problem:

LCD: ST7565 on HSPI_HOST @ 1 MHz
SPI FLASH: W25Q128 on VSPI_HOST @ 40 MHz

Once in a while, and randomly, spi_device_transmit() will deadlock. It never returns. According to the docs, this should be impossible. The function should either return with success or failure. In debugging this issue, the deadlock happens in spi_device_transmit() in spi_master.c, on this line:

```
ret=spi_device_get_trans_result(handle, &ret_trans, portMAX_DELAY);
```

This seems to indicate to me that the interrupt handler that receives the transaction complete notification is never firing. 

I also want to reiterate that this only happens in dual core mode - in single core mode it seems to be fine.

**Can you tell me if there is any situation in which spi_device_transmit() should fail to return?**

If there is any other information I can supply, please let me know. I have tried reproducing this on a dev board, but there is so much software behind it that it will take a long time to write something that accurately reproduces the issue. That being said, I want to reiterate that this deadlock is within the ESP-IDF code.

Thanks,
Jason

negativekelvin, 2018-04-21:
Have you tried forcing spi_bus_initialize for both buses to be called on the same core so that both interrupts are enabled on the same cpu?

vonnieda, 2018-04-21:
Hi @negativekelvin, yes, all my init code runs from app_main(). 

app_main() calls `fb_hal_initialize()`

Here is that function, somewhat trimmed down:

```
esp_err_t fb_hal_initialize() {
	FB_ERROR_CHECK_NO_RET(fb_hal_bulk_storage_initialize());
	FB_ERROR_CHECK_NO_RET(fb_hal_lcd_initialize());
	return ESP_OK;
}
```

And here are those two init functions, also somewhat trimmed down:

```
static esp_err_t fb_hal_bulk_storage_initialize() {
	bulk_storage_spi_mutex = xSemaphoreCreateMutex();

	bulk_storage_rx_buffer = heap_caps_calloc(
			BULK_STORAGE_SPI_MAX_TRANSFER_BYTES,
			sizeof(uint8_t),
			MALLOC_CAP_DMA);
	bulk_storage_tx_buffer = heap_caps_calloc(
			BULK_STORAGE_SPI_MAX_TRANSFER_BYTES,
			sizeof(uint8_t),
			MALLOC_CAP_DMA);
	spi_bus_config_t bus_config = {
		.mosi_io_num = BULK_STORAGE_IO_SPI_MOSI,
		.miso_io_num = BULK_STORAGE_IO_SPI_MISO,
		.sclk_io_num = BULK_STORAGE_IO_SPI_SCLK,
		.quadwp_io_num = BULK_STORAGE_IO_SPI_QWP,
		.quadhd_io_num = BULK_STORAGE_IO_SPI_QHD,
		.max_transfer_sz = BULK_STORAGE_SPI_MAX_TRANSFER_BYTES,
	};

	FB_ERROR_CHECK_RET(spi_bus_initialize(VSPI_HOST, &bus_config, BULK_STORAGE_SPI_DMA_CHANNEL));

	spi_device_interface_config_t dev_config = {
		.clock_speed_hz = BULK_STORAGE_SPI_SPEED_HZ,
		.mode = 0,
		.spics_io_num = BULK_STORAGE_IO_SPI_CS,
		.queue_size = 7,
		.flags = SPI_DEVICE_HALFDUPLEX,
	};

	FB_ERROR_CHECK_RET(spi_bus_add_device(VSPI_HOST, &dev_config, &bulk_storage_spi_device));

	return ESP_OK;
}

static esp_err_t fb_hal_lcd_initialize() {
	lcd_spi_mutex = xSemaphoreCreateMutex();

	lcd_tx_buffer = heap_caps_calloc(
			LCD_WIDTH * LCD_HEIGHT / 8,
			sizeof(uint8_t),
			MALLOC_CAP_DMA);

	esp_err_t err;

	// Configure the SPI bus
	spi_bus_config_t bus_config = {
			.mosi_io_num = LCD_IO_MOSI,
			.miso_io_num = -1,
			.sclk_io_num = LCD_IO_SCK
	};
	err = spi_bus_initialize(LCD_SPI_BUS, &bus_config, LCD_SPI_DMA_CHANNEL);

	// Configure the LCD SPI device
    spi_device_interface_config_t dev_config = {
        .clock_speed_hz = LCD_SPI_SPEED_HZ,
		.mode = 0,
        .spics_io_num = LCD_IO_CS,
        .queue_size = 7,
    };
	err = spi_bus_add_device(LCD_SPI_BUS, &dev_config, &lcd_spi_device);

	// setup GPIO for A0
	gpio_config_t g = {
		.pin_bit_mask = (1ULL << LCD_IO_A0),
		.mode = GPIO_MODE_OUTPUT
	};
	err = gpio_config(&g);

	return ESP_OK;
}
```

negativekelvin, 2018-04-21:
Ok well then have you tried forcing spi_bus_initialize for each bus to be called on different cores so that both interrupts are not enabled on the same cpu?

vonnieda, 2018-04-22:
Hi @negativekelvin, I've run the test as you described and it looks like it might fix the problem. I've been running now for about 41 minutes without a deadlock, where previously it would lock usually within 3-5 minutes.

Here's how I changed my code:

```
volatile static bool initialize_cpu_0_done = false;
volatile static bool initialize_cpu_1_done = false;

void initialize_cpu_0(void* pw) {
	FB_LOG_INFO(0, "Initializing Bulk Storage on core %d", xPortGetCoreID());
	FB_ERROR_CHECK_NO_RET(fb_hal_bulk_storage_initialize());
	initialize_cpu_0_done = true;
	FB_LOG_INFO(0, "Initializing Bulk Storage on core %d DONE", xPortGetCoreID());
	vTaskDelete(NULL);
}

void initialize_cpu_1(void* pv) {
	FB_LOG_INFO(0, "Initializing LCD on core %d", xPortGetCoreID());
	FB_ERROR_CHECK_NO_RET(fb_hal_lcd_initialize());
	initialize_cpu_1_done = true;
	FB_LOG_INFO(0, "Initializing LCD on core %d DONE", xPortGetCoreID());
	vTaskDelete(NULL);
}

esp_err_t fb_hal_initialize() {
	xTaskCreatePinnedToCore(initialize_cpu_0,
			"init_0",
			5 * 1024,
			NULL,
			tskIDLE_PRIORITY,
			NULL,
			0);
	while (!initialize_cpu_0_done) {
		vTaskDelay(500 / portTICK_PERIOD_MS);
	}

			xTaskCreatePinnedToCore(initialize_cpu_1,
					"init_1",
					5 * 1024,
					NULL,
					tskIDLE_PRIORITY,
					NULL,
					1);
			while (!initialize_cpu_1_done) {
				vTaskDelay(500 / portTICK_PERIOD_MS);
			}

	return ESP_OK;
}

```

So, this seems like it might be a workaround, but I'd like to understand why the problem is happening in the first place. Does this give you any clue as to what might be wrong?

Thanks,
Jason

negativekelvin, 2018-04-23:
It seems like it happens when there are overlapping interrupts on the same CPU and the other CPU is also active but still unclear exactly why.

vonnieda, 2018-04-23:
Thanks @negativekelvin, would you recommend this workaround for now, then?

negativekelvin, 2018-04-23:
Yes because you can see if it really prevent the issue 100%.  Hopefully you can give some minimal code to reproduce it.  Maybe you could also get logic capture to see exactly where it stops in terms of spi outputs.

vonnieda, 2018-04-25:
Hi @negativekelvin, I've just seen this post: https://esp32.com/viewtopic.php?f=2&t=5505&sid=3421566ef8d889acd849d972481fcf87 and then the documentation that it references at https://esp-idf.readthedocs.io/en/latest/api-reference/system/intr_alloc.html#external-peripheral-interrupts.

I looked at spi_master.c and it does indeed call esp_intr_alloc outside of a critical section, so, does this mean that I need to be sure that any task that calls into the spi_master driver is on the same core that allocated it?

This reminds me that when I tried your suggestion a while back of pinning all the SPI tasks to one core it seemed to fix it. Is this a requirement for ESP-IDF, then?

Thanks,
Jason

wuyuanyi135, 2023-02-06:
I just ran into a similar issue and the discussion in this issue seems helped a lot. 

I am having multiple thread accessing the same SPI device. The access are guarded by std::mutex and I could confirm no racing condition. However, I still got deadlock after a few hours. I crashed the program when deadlock and collected the coredump. Supurisingly, all blocking threads were stuck at `spi_device_acquire_bus`, an unexpected indication that a transaction was not ending properly. My SPI bus/locks are managed by C++ RAII pattern so they are guaranteed to be released after use. 

After reading through this issue, I realize that even though my bus access is mutually exclusive, the threads are not guaranteed to run on the same core (i.e., not pinned). I refactored my code to ensure all SPI transaction were run on the same core. Now the issue was gone in the short test (a few hours). I will keep it on and see if this really is the root issue.

If this is the case, I would argue that the [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html) should be improved.

> The SPI master driver has the concept of multiple Devices connected to a single bus (sharing a single ESP32 SPI peripheral). As long as each Device is accessed by only one task, the driver is thread safe. However, if multiple tasks try to access the same SPI Device, the driver is not thread-safe. In this case, it is recommended to either:

> Refactor your application so that each SPI peripheral is only accessed by a single task at a time. You can use [spi_bus_config_t::isr_cpu_id](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_master.html#_CPPv4N16spi_bus_config_t10isr_cpu_idE) to register the SPI ISR to the same core as SPI peripheral related tasks to ensure thread safety.

> Add a mutex lock  `around` the shared Device using [xSemaphoreCreateMutex](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#c.xSemaphoreCreateMutex).

Clearly, the second recommendation is not sufficient to guarantee the bus ISR and the task runs on the same core. Could you please confirm this? @negativekelvin

KaeLL, 2023-02-06:
I was having a problem that looks exactly like this one (#6624) that I gave up digging about it as I deemed it to be above my paygrade, but someone apparently nailed it (#8179) and seems to be solved now.  
I suggest you test it again with revisions which include that fix, and provided that there's mutual exclusion at the device level, calling any amount of SPI functions from either core in any way should work, otherwise it's a bug and you should open an issue.

AxelLin, 2023-02-07:
> I just ran into a similar issue 

What is your esp-idf version? (git describe --tags output)

wuyuanyi135, 2023-02-07:
@AxelLin v5.1-dev-3223-ge1ebda17c4

AxelLin, 2023-02-07:
> @AxelLin v5.1-dev-3223-ge1ebda17c4

It already includes 3c8f6dd1cfebaee3860fbc66ff985528ba56b447.
Could you provide more detail about the issue? e.g. sdkconfig, backtrace log. 
(Maybe you should create a new issue instead if your issue is not exactly the same as this one.)

wuyuanyi135, 2023-02-07:
@AxelLin The issue might not be exactly the same but pinning task to the same core as ISR did fix the issue. I have left my device running for 24 hrs without issue otherwise it will be stuck in deadlock in couple of hours. I will try to reproduce the error later and start a new issue. Thanks

KaeLL, 2023-02-07:
> I will try to reproduce the error later

Please do that.

KaeLL, 2024-03-27:
@wuyuanyi135 Did you do it?

wuyuanyi135, 2024-04-11:
Sorry for not responding. Probably due to environment inconsistent and the randomness I could not reproduce the issue. After updating to v5.2 and sticking to the same core rule I have never run into the same issue.