espressif/esp-idf #1653: unable to disable bluedroid having SMP disabled

mkonnov, 2018-02-23:
The issue makes itself know when I've updated esp-idf and all related modules from commit
914ab075672216c15a5564d9ae54fc257b237c13 to 2abba131729e6975085b49171251167748ef6baa

```C
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_bt.h"
#include "esp_bt_main.h"
#include "esp_bt_device.h"

void app_main(void) {

        nvs_flash_init();

	esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
	esp_bt_controller_init(&bt_cfg);
	esp_bt_controller_enable(ESP_BT_MODE_BTDM);
	esp_bluedroid_init();
	esp_bluedroid_enable();

        ESP_LOGI("", "free heap mem: %d", xPortGetFreeHeapSize());

	esp_bluedroid_disable();
	esp_bluedroid_deinit();
	esp_bt_controller_disable();
	esp_bt_controller_deinit();

	vTaskDelay(NULL);
	while (true);
}
```

SMP turned __off__
```
I (490) : free heap mem: 155888
assertion "config != NULL" failed: file "/home/f77e/projects/package-guard-device-v2/esp-idf/components/bt/bluedroid/btc/core/btc_config.c", line 312, function: btc_config_flush
abort() was called at PC 0x400d41ff on core 0
```
firmware size is **493744** bytes

SMP turned __on__
```
I (573) : free heap mem: 148624
```
firmware size is **659968** bytes

Weijian-Espressif, 2018-02-24:
@mkonnov if you want to disable SMP, you must cancel selection SMP and classic bluetooth case in menuconfig.

mkonnov, 2018-02-24:
Obviously, I'm toggling SMP via menuconfig...
Classic BT is disabled in both cases.

chegewara, 2018-02-24:
If classic BT is disabled in menuconfig then you shouldnt initialize bluetooth with this command:
`esp_bt_controller_enable(ESP_BT_MODE_BTDM);`
`ESP_BT_MODE_BTDM` initializes bluetooth in dual mode (classic BT + LE). 
Maybe this is not the case but worth to try.

mkonnov, 2018-02-24:
@chegewara thanks, I'll try.
Anyway, it has worked before I've updated esp-idf.

Weijian-Espressif, 2018-02-24:
@chegewara you are right 👍 
@mkonnov you should do that  
esp_err_t ret;

    // Initialize NVS.
    ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    **ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));**

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s initialize controller failed\n", __func__);
        return;
    }

    ret = esp_bt_controller_enable(**ESP_BT_MODE_BLE**);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable controller failed\n", __func__);
        return;
    }
    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s init bluetooth failed\n", __func__);
        return;
    }
    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable bluetooth failed\n", __func__);
        return;
    }

mkonnov, 2018-02-24:
Ok, will try in a half an hour.
Why do we need to call _esp_bt_controller_mem_release() ?_

chegewara, 2018-02-24:
@mkonnov You dont have to, but it will let you to release about 30kB heap from bluetooth classic driver.

Weijian-Espressif, 2018-02-24:
it releases the memory by mode, if never use the bluetooth mode . it can release the .bbs, .data and other section to heap

chegewara, 2018-02-24:
@Weijian-Espressif BTW, there is no more info about `esp_bt_controller_mem_release` in documentation, whole section about bluetooth controller functions is gone in all release versions.
http://esp-idf.readthedocs.io/en/v3.0-rc1/api-reference/bluetooth/controller_vhci.html

Weijian-Espressif, 2018-02-24:
sorry, we have no much time to update the document. you can refer to esp_bt.h , sorry.

mkonnov, 2018-02-24:
Works !
Thanks everyone.

closed 2018-02-24 by mkonnov

spli, 2018-05-15:
Hi

I'm now getting the same problem? Is this is a regression.

Initializing with

```
esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s initialize controller failed\n", __func__);
        return DING_SETUP_BLE_ERROR;
    }
    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable controller failed\n", __func__);
        return DING_SETUP_BLE_ERROR;
    }

    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s init bluetooth failed\n", __func__);
        return DING_SETUP_BLE_ERROR;
    }

    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable bluetooth failed\n", __func__);
        return DING_SETUP_BLE_ERROR;
    }
```

and disabling with:
```
esp_bluedroid_disable();
esp_bluedroid_deinit();
esp_bt_controller_disable();
esp_bt_controller_deinit();
```

Crash is 
```
assertion "config != NULL" failed: file "/Users/simon/esp/esp-idf/components/bt/bluedroid/btc/core/btc_config.c", line 312, function: btc_config_flush
abort() was called at PC 0x400d4233 on core 0
0x400d4233: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x40090e97:0x3ffb3410 0x40090fef:0x3ffb3430 0x400d4233:0x3ffb3450 0x4019b71f:0x3ffb3480 0x4019b72f:0x3ffb34a0 0x4018639d:0x3ffb34c0 0x40186426:0x3ffb34e0
0x40090e97: invoke_abort at /Users/simon/esp/esp-idf/components/esp32/panic.c:648

0x40090fef: abort at /Users/simon/esp/esp-idf/components/esp32/panic.c:648

0x400d4233: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x4019b71f: btc_config_flush at /Users/simon/esp/esp-idf/components/bt/bluedroid/btc/core/btc_config.c:329

0x4019b72f: btc_config_shut_down at /Users/simon/esp/esp-idf/components/bt/bluedroid/btc/core/btc_config.c:329

0x4018639d: btc_disable_bluetooth at /Users/simon/esp/esp-idf/components/bt/bluedroid/btc/core/btc_main.c:47
 (inlined by) btc_main_call_handler at /Users/simon/esp/esp-idf/components/bt/bluedroid/btc/core/btc_main.c:105

0x40186426: btc_task at /Users/simon/esp/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:104

```

It works fine if I enable SMP, which I don't want to do.

Thanks

Weijian-Espressif, 2018-05-16:
hi @spli  I fix the bug in the latest IDF.
[image]

spli, 2018-05-17:
That works. Thanks. 

When will this change appear in master?

Weijian-Espressif, 2018-05-18:
@spli This bug has been fixed in the latest IDF.