espressif/esp-idf #1656: Coredump into flash freezes chip

michprev, 2018-02-24:
Hello,
I am occasionaly experiencing problems with coredumping into flash memory.

After assert(0) in my code panic handler gets invoked and starts coredump at https://github.com/espressif/esp-idf/blob/165e3edd4949586dc90fa682f907d971fee488fc/components/esp32/panic.c#L567
I have inserted `ESP_COREDUMP_LOGE("Reached erase part")` into `esp_core_dump_flash_write_prepare` function right before this line https://github.com/espressif/esp-idf/blob/9274814268f51333bd8aed65202621161e877a6a/components/esp32/core_dump.c#L334

After printing "Reached erase part" ESP chip freezes for a while and then reboots or even gets stuck up forever!

Log:
```
assertion "0" failed: file "/home/michal/git/flyhero-esp32/Data_Analysis/main/./main.cpp", line 67, function: void imu_task(void*)
abort() was called at PC 0x400d2dbb on core 1
0x400d2dbb: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x4008dc3c:0x3ffbeee0 0x4008de13:0x3ffbef00 0x400d2dbb:0x3ffbef20 0x400d2b87:0x3ffbef50
0x4008dc3c: invoke_abort at /home/michal/git/flyhero-esp32-idf/components/esp32/./panic.c:651

0x4008de13: abort at /home/michal/git/flyhero-esp32-idf/components/esp32/./panic.c:651

0x400d2dbb: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x400d2b87: imu_task(void*) at /home/michal/git/flyhero-esp32/Data_Analysis/main/./main.cpp:67

I (6109) esp_core_dump: Save core dump to flash...
D (6109) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 4008a67a 60030 801199a9 3ffbfe20
D (6110) esp_core_dump: Stack len = 636 (3ffbfd60 3ffbffdc)
D (6111) esp_core_dump: Current task EXIT/PC/PS/A0/SP 32646231 4008dc3c 60f30 8008de16 3ffbeee0
D (6112) esp_core_dump: Stack len = 440 (3ffbee20 3ffbefd8)
D (6113) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 400d26c2 60530 8008ba84 3ffbca70
D (6114) esp_core_dump: Stack len = 364 (3ffbc9b0 3ffbcb1c)
D (6115) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 400d26c2 60630 8008ba84 3ffbc510
D (6116) esp_core_dump: Stack len = 352 (3ffbc450 3ffbc5b0)
D (6116) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 4008aa9d 60630 80119b54 3ffc0d00
D (6117) esp_core_dump: Stack len = 484 (3ffc0c40 3ffc0e24)
D (6118) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 4008c58c 60430 8008c67f 3ffbd4b0
D (6119) esp_core_dump: Stack len = 396 (3ffbd3f0 3ffbd57c)
D (6120) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 4008aa9d 60830 800ed506 3ffc2790
D (6121) esp_core_dump: Stack len = 444 (3ffc26d0 3ffc288c)
D (6121) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 4008aa9d 60730 80103bbd 3ffc7c50
D (6122) esp_core_dump: Stack len = 408 (3ffc7b90 3ffc7d28)
D (6123) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 4008aa9d 60430 800d1fa3 3ffafba0
D (6124) esp_core_dump: Stack len = 388 (3ffafae0 3ffafc64)
D (6125) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 400813ee 60630 8008aa9d 3ffc3990
D (6126) esp_core_dump: Stack len = 460 (3ffc38d0 3ffc3a9c)
D (6127) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 4008aa9d 60830 8008142b 3ffba0d0
D (6128) esp_core_dump: Stack len = 384 (3ffba010 3ffba190)
D (6128) esp_core_dump: Task EXIT/PC/PS/A0/SP 40082384 4008aa9d 60d30 8008142b 3ffba690
D (6129) esp_core_dump: Stack len = 384 (3ffba5d0 3ffba750)
D (6130) esp_core_dump: Core dump len = 9568 (12 0)
E (6131) esp_core_dump: Reached erase part
```
I think that the problem might be related to the fact that spi_flash_erase_range function is placed in IRAM. Any thoughts how to figure out what is the chip doing without JTAG?

gerekon, 2018-03-07:
Hi @michprev , 
What is the value of `Panic handler behaviour` menuconfig item from `Component config->ESP32-specific`?
Are you able to retrieve core dump using `espcoredump.py` tool even when chip gets stuck or rebooted?

michprev, 2018-03-08:
Hi @gerekon ,
it is set to `Print registers and reboot` however I am experiencing this problem even with `Print registers and halt` or `Silent reboot` settings.

I am not able to retrieve crash dump, I am getting `A fatal error occurred: Invalid start magic number!` and when I try to download coredump flash region via esptool it is just empty.

I am attaching coredump log from crash when this issue does NOT occur. I guess that it is normal but just to make sure (my IDF is up-to-date with https://github.com/espressif/esp-idf/commit/6acb38af4c273077f6cfd54d6f84af6d5f1e20f9).
[freeze_issue.log](https://github.com/espressif/esp-idf/files/1794997/freeze_issue.log)

I can provide example code but it is currently needed to attach I2C or SPI IMU sensor (MPU6050 / MPU6000) to reproduce this issue.

gerekon, 2018-03-12:
@michprev , Could you attach `sdkconfig` and `partition table CSV`? 
Also could you insert debug statements after `spi_flash_erase_range` too? To be sure that process is stuck exactly in that function.

> I think that the problem might be related to the fact that spi_flash_erase_range function is placed in IRAM.

It should be there, due to several reasons. One of them is because CPU cache must be disabled when writing to flash so we can not execute in-flash code.

michprev, 2018-03-12:
@gerekon I can confirm that `After erase flash` does not get printed out with 
```
    ESP_COREDUMP_LOGE("Before erase flash");
    err = spi_flash_erase_range(s_core_flash_config.partition.start + 0, sec_num * SPI_FLASH_SEC_SIZE);
    ESP_COREDUMP_LOGE("After erase flash");
``` 
when this issue occurs.

[partitions_singleapp_coredump.txt](https://github.com/espressif/esp-idf/files/1804339/partitions_singleapp_coredump.txt)

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/1804343/sdkconfig.txt)

gerekon, 2018-03-13:
@michprev, 
Since it is not easy to reproduce the problem at my side. Could you insert some debug output to help me diagnose the problem?

I need the folowing statement before call to `spi_flash_erase_range`:
`ESP_COREDUMP_LOGE("Before erase flash start=%x, sec num=%d", s_core_flash_config.partition.start, sec_num);`

gerekon, 2018-03-13:
@michprev,

Can we arrange live debug session via TeamViewer or something like this?

gerekon, 2018-03-13:
@michprev,

I found that generating core dump using assert(0) and the code below behaves differently. As a last attempt before screen sharing session )).
Could you check once more with the code snippet below instead of assert statement?

`
        int *p = 0;
        *p = 0;
`

michprev, 2018-03-13:
@gerekon 
The problem still occurs. You can also check out https://github.com/espressif/esp-idf/issues/1394. It migh be related to this issue and it is fairly easy to reproduce.

andersjohnsen, 2019-11-08:
We're seeing this as well, but observed that it only happens when the panic is on Core 1.

pumelo, 2020-10-01:
I've just come across this issue in 4.2.

pumelo, 2020-10-01:
I didnt share the worst part: Even with the watchdog enabled it freezes and does not recover/reset.
I'm not sure about it, but I think in my case the crash happens inside an isr which runs on core 1. I'll post more details if I'm able to get them.

jakkra, 2020-10-15:
We see similar problem on 4.0.1 release. We will also do more dubugging and provide more info. For us also device never recovers and we need to power cycle.

igrr, 2020-10-15:
@jakkra @pumelo  Thanks, please open a new issue if you get more information. This issue is quite old, and both the SPI flash implementation and most of the core dump code are different in v4.x compared to v3.x.

doothez, 2021-01-25:
I'm using esp-idf v3.3.2 and having same problem like https://github.com/espressif/esp-idf/issues/3857

> abort() was called at PC 0x400d862a on core 0
> 0x400d862a: restart at /Users/sung.lee/Code/esp/bov950_esp/components/brg_console/brg_console.c:219
> ELF file SHA256: 4e45125e97495c88
> Backtrace: 0x400914dc:0x3f90eb30
> 0x400914dc: invoke_abort at /Users/sung.lee/Code/esp/bov950_esp/amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715
> I (13192) esp_core_dump_flash: Save core dump to flash...
> I (13199) esp_core_dump_common: Found tasks: (17)!
> I (13204) esp_core_dump_port: Current task 3ffc37b8 EXIT/PC/PS/A0/SP 38363230 400914dc 60330 8009179c 3f90eb30
> I (13215) esp_core_dump_port: Crashed task will be skipped!
> I (13221) esp_core_dump_port: Corrupted TCB 3ffc37b8: stack len 500, top 3f90ea70, end 3f90ec64!
> I (13231) esp_core_dump_common: Found tasks: #00 TCD is ok but stack is corrupted ! 
> I (13239) esp_core_dump_port: Bad TCB addr 3f800c64!
> I (13245) esp_core_dump_common: Found tasks: #01 is invalid which means bad TCD.
> I (13253) esp_core_dump_port: Task 3ffbcb60 EXIT/PC/PS/A0/SP 0 80095831 60b20 80183ff5 3ffbcaa0
> I (13262) esp_core_dump_port: Stack len = 248 (3ffbca60 3ffbcb58)
> I (13269) esp_core_dump_port: Bad TCB addr 3f800800!
> I (13274) esp_core_dump_common: Found tasks: #03 is invalid which means bad TCD.
> I (13283) esp_core_dump_port: Task 3ffd6224 EXIT/PC/PS/A0/SP 0 80094468 60e20 800f9781 3ffd6150
> I (13292) esp_core_dump_port: Stack len = 300 (3ffd60f0 3ffd621c)
> I (13299) esp_core_dump_port: Task 3ffd6f90 EXIT/PC/PS/A0/SP 0 80094468 60720 800f9781 3ffd6ec0
> I (13308) esp_core_dump_port: Stack len = 296 (3ffd6e60 3ffd6f88)
> I (13315) esp_core_dump_port: Task 3ffd1d94 EXIT/PC/PS/A0/SP 0 800942e4 60d20 0 3ffd5330
> I (13324) esp_core_dump_port: Stack len = 252 (3ffd52d0 3ffd53cc)
> I (13330) esp_core_dump_port: Task 3ffcf33c EXIT/PC/PS/A0/SP 0 80095e10 60023 0 3ffcf230
> I (13339) esp_core_dump_port: Stack len = 324 (3ffcf1f0 3ffcf334)
> I (13346) esp_core_dump_port: Task 3ffd1604 EXIT/PC/PS/A0/SP 0 80095d4c 60023 0 3ffd1560
> I (13355) esp_core_dump_port: Stack len = 220 (3ffd1520 3ffd15fc)
> I (13361) esp_core_dump_port: Task 3ffdb0d4 EXIT/PC/PS/A0/SP 0 800942e4 60020 0 3ffdcfa0
> I (13370) esp_core_dump_port: Stack len = 276 (3ffdcf40 3ffdd054)
> I (13377) esp_core_dump_port: Task 3ffda6c0 EXIT/PC/PS/A0/SP 0 800942e4 60320 801b3060 3ffda5e0
> I (13386) esp_core_dump_port: Stack len = 312 (3ffda580 3ffda6b8)
> I (13393) esp_core_dump_port: Task 3ffde5c0 EXIT/PC/PS/A0/SP 0 800942e4 60320 80148730 3ffde500
> I (13403) esp_core_dump_port: Stack len = 280 (3ffde4a0 3ffde5b8)
> I (13409) esp_core_dump_port: Task 3ffecc04 EXIT/PC/PS/A0/SP 0 800942e4 60020 0 3ffecb50
> I (13418) esp_core_dump_port: Stack len = 268 (3ffecaf0 3ffecbfc)
> I (13425) esp_core_dump_port: Task 3ffea584 EXIT/PC/PS/A0/SP 0 800942e4 60520 0 3ffea4c0
> I (13434) esp_core_dump_port: Stack len = 284 (3ffea460 3ffea57c)
> I (13440) esp_core_dump_port: Task 3ffbb83c EXIT/PC/PS/A0/SP 0 800942e4 60d20 0 3ffbb7a0
> I (13449) esp_core_dump_port: Stack len = 244 (3ffbb740 3ffbb834)
> I (13456) esp_core_dump_port: Task 3ffb5f7c EXIT/PC/PS/A0/SP 0 800942e4 60b20 8016d380 3ffb5ec0
> I (13465) esp_core_dump_port: Stack len = 276 (3ffb5e60 3ffb5f74)
> I (13472) esp_core_dump_port: Task 3ffb7134 EXIT/PC/PS/A0/SP 0 80094468 60020 0 3ffb70a0
> I (13481) esp_core_dump_port: Stack len = 236 (3ffb7040 3ffb712c)
> I (13488) esp_core_dump_common: Core dump len = 9352 (17 3)
> ets Jun  8 2016 00:22:57
> rst:0x9 (RTCWDT_SYS_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:6980
> load:0x40078000,len:12944
> load:0x40080400,len:7084
> entry 0x40080774

So I tried to patch below one but that was for esp-idf v4.0 or higher, so I got below error and could not figure it out.
https://gitlab.unizar.es/761347/esp-idf/commit/e5ff431b35b21fffe486cf6bbaa835d74e66e22a

[ 83%] Built target idf_component_openssl
[ 83%] Built target idf_component_openssl_sections_info
[ 83%] Built target idf_component_protobuf-c
[ 83%] Built target idf_component_protobuf-c_sections_info
[ 84%] Built target idf_component_spiffs
[ 84%] Built target idf_component_spiffs_sections_info
[ 84%] Built target ldgen_section_infos
[ 85%] Generating esp32.project.ld
linker script generation failed for /Users/sung.lee/Code/esp/bov950_esp/amazon-freertos/vendors/espressif/esp-idf/components/esp32/ld/esp32.project.ld.in
ERROR: Parse error in linker fragment /Users/sung.lee/Code/esp/bov950_esp/amazon-freertos/vendors/espressif/esp-idf/components/espcoredump/linker.lf: error at line 9 col 1 (char 185)
make[2]: *** [amazon-freertos/esp-idf/esp32/esp32.project.ld] Error 1
make[1]: *** [amazon-freertos/esp-idf/esp32/CMakeFiles/ldgen_esp32.project.ld_script.dir/all] Error 2
make: *** [all] Error 2

Seem like esp-idf v3.3.2 linker setting is not good for the patch which means esp_flash module is not exist and just copying all the spi_flash from v4.x to v3.3.2 is not buildable because of linker and other differences. 

Are there any backport for v3.3.2 ?

adradamczyk, 2021-11-02:
I think I ran into the same problem with IDF v4.3.1:
```
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4013884c: 22428c00 2254dbc2 a22140ff
0x4013884c: _svfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:918

Core  1 register dump:
PC      : 0x40138852  PS      : 0x00060f30  A0      : 0x80138204  A1      : 0x3ffe8300
0x40138852: _svfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:918

A2      : 0x00000000  A3      : 0x3f409c64  A4      : 0xffffffff  A5      : 0x3ffe8310
A6      : 0x3ffe8690  A7      : 0x00000008  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000001  A11     : 0x3f41828b  A12     : 0x000000ff  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace:0x4013884f:0x3ffe8300 0x40138201:0x3ffe8610 0x400e4fae:0x3ffe86d0 0x400e507c:0x3ffe8780 0x400e3a68:0x3ffe87d0
0x4013884f: _svfprintf_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:918

0x40138201: sprintf at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/sprintf.c:618

0x400e4fae: <project_data> at <project_data>.c:201

0x400e507c: <project_data> at <project_data>.c:296

0x400e3a68: <project_data>_task at <project_data>.c:129

ELF file SHA256: 08c868e80c4a624d

I (12922) esp_core_dump_flash: Save core dump to flash...
```

The worst part is, as it was already mentioned, the chip freezes during core dump and requires a power cycle to restart.

Unfortunately it is hard for me to reproduce, as panic errors are random as well as core dump freeze.

jakkra, 2021-11-02:
@adradamczyk can you create a new issue please, here Espressif says that they won't do anything pretty much in this issue and to create a new one for them to investigate: https://github.com/espressif/esp-idf/issues/1656#issuecomment-709421639
We had to disable core dump for our products using v4.0.1 due to this issue.

o-marshmallow, 2021-11-17:
As @jakkra said, please create a new issue if this problem is still relevant.

Closing this (old) issue.

closed 2021-11-17 by o-marshmallow