espressif/esp-idf #1658: BLE Passive Scan Channels

hopkinskong, 2018-02-26:
BLE devices advertises on 3 channels, using frequency hopping. Does ESP32 scan the 3 channels simultaneously, or just monitor 1 channel? It seems that I can't scan some devices continously as it seems they disappear after few seconds, probably hopped to another channel, then come back up visible by ESP32 after few secs.

Weijian-Espressif, 2018-02-26:
@hopkinskong Our ESP32 can scan the all adv channels. of course you can scan one of the channels by setting scan params. you can refer to gatt_client demo in ESP-IDF.
static esp_ble_scan_params_t ble_scan_params = {
    .scan_type              = BLE_SCAN_TYPE_ACTIVE,
    .own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_interval          = 0x50,
    .scan_window            = 0x30
};
If "scan_interval" is equal to "scan_window", the scanner will scan  continously.

hopkinskong, 2018-02-26:
@Weijian-Espressif Does it also work on `BLE_SCAN_TYPE_PASSIVE`?

negativekelvin, 2018-02-26:
Advertising and scanning both rotate through the channels sequentially so there is a chance you could miss advertisements.

https://devzone.nordicsemi.com/cfs-file/__key/communityserver-blogs-components-weblogfiles/00-00-00-00-04-DZ-782/1273.Scan.PNG

hopkinskong, 2018-02-26:
@negativekelvin Is there no workaround for this issue, except deploying multiple scanners?

negativekelvin, 2018-02-26:
If your scan window is > longest advertising interval of devices you are scanning and you are continuously scanning then you should receive them

FayeY, 2018-03-01:
Hi @hopkinskong , did the above suggestion help you resolve the issue?

hopkinskong, 2018-03-02:
@FayeY Need to test, closed the issue for now. Will reopen if there is still issues.

hopkinskong, 2018-03-03:
@FayeY @Weijian-Espressif 

If `.scan_interval` == `.scan_window`, a persistent TCP connection will be timed out, and wifi connection might loss:
`I (26642) wifi: bcn_timout,ap_probe_send_start`
HTTP throughput is unacceptable (200KB transfer > 120s, and disconnection in the middle)

I am referencing to this issue #1639:
Optimizing BLE scan will affect TCP performance *a lot*
Optimizing TCP connection will affect BLE performance too...

I need to deploy my application on 16x ESP32-WROVER and I really need to get this issue solved ASAP.

Weijian-Espressif, 2018-03-05:
@hopkinskong , if you set .scan_interval == .scan_window, ble always use the  phy, it will case wifi disconnection.

hopkinskong, 2018-03-05:
@Weijian-Espressif Exactly, how to optimize the parameters for best performance of **both Wi-Fi + BLE**?
@negativekelvin I concur. I tried SW/HW COEXIST option in sdkconfig but it doesn't seem improve much performance.

Weijian-Espressif, 2018-03-05:
if you want to use both Wi-Fi + BLE, You need to pay attention to the scan parameter or adv parameter.
BLE and sharing a phy and BLE has a higher priority. When ble is using the phy, wifi will wait until the phy is idle. So You need to pay attention to the scan parameters, .scan_interval == .scan_window is not allowed in ble & wifi coexistence .

negativekelvin, 2018-03-05:
@Weijian-Espressif phy-controlled scan would be better, it could scan continuously except for rx dtim or tx.  Basically modem sleep but instead of sleep, scan.  Beacon scanner is a common use case.

hopkinskong, 2018-03-11:
@jkhax0r Good to know I am not alone.

@Weijian-Espressif Is there any recommended configuration parameters to maximize the performance of both BLE and Wi-Fi?

TianHao-Espressif, 2018-04-03:
related to #1639

hopkinskong, 2018-04-03:
@TianHao-Espressif Is it this commit?
https://github.com/espressif/esp32-wifi-lib/commit/608bf369c66c33c15b0cc198e573e421c7103e1f

What is the acceptable range for testing?  `.scan_interval == .scan_window` is still not expected to work, right?

hopkinskong, 2018-04-03:
@TianHao-Espressif Also, do I need to enable `CONFIG_SW_COEXIST_ENABLE`?

hopkinskong, 2018-04-03:
@TianHao-Espressif With `CONFIG_SW_COEXIST_ENABLE` = `YES`, the performance **is very bad**. After application start, TCP connection disconnects* in 30-45s.

With `CONFIG_SW_COEXIST_ENABLE` = `NO`, the performance is much more better, now running at least **3 minutes** without issues.

Note that my BLE scans every 1s, with the following parameters:
`.scan_type              = BLE_SCAN_TYPE_PASSIVE,
		.own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
		.scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
		.scan_interval          = 0x550, 
		.scan_window            = 0x380 `

*Disconnects means the TCP data stream is idle for more than 11s (with is REALLY high value)

TianHao-Espressif, 2018-04-03:
@hopkinskong , 
two choices:
1. if disable CONFIG_SW_COEXIST_ENABLE, you should keep "scan interval - scan windows > 150ms", it can works.
2. if enable CONFIG_SW_COEXIST_ENABLE, any scan interval and scan windows value. And for your requirement, you can try "prefer to wifi".  

For our  internal test, the tcp do not disconnect while bluetooth run A2DP for serval hours.

TianHao-Espressif, 2018-04-03:
@hopkinskong , and do you do git submodule update or could you provide me your whole log?

hopkinskong, 2018-04-06:
@TianHao-Espressif
 What do you mean "prefer to wifi"? Where can I set "prefer to wifi"?

Now my problem remains poor TCP performances. I need to send ~200KB data every 0.5~1s, they are JPEG data from the camera, the connection will just stuck due to delay and low throughput. The performances are just very bad.

negativekelvin, 2018-04-06:
https://github.com/espressif/esp-idf/commit/9b7454de0c171a7f94dd8f42a5453ef6c75370e0#diff-0b068dd317b997485fff4a56e7fb7ee8

TianHao-Espressif, 2018-04-08:
@negativekelvin, thanks.
@hopkinskong , as @negativekelvin wrote, please update your esp-idf and choose 
-> Component config                                                                                                                                                                                                               -> Wi-Fi                                                                                                                                                                                                                               -> Software controls WiFi/Bluetooth coexistence (SW_COEXIST_ENABLE [=y])                                                                                                                                                       -> WiFi/Bluetooth coexistence performance preference (SW_COEXIST_PREFERENCE [=y])

megazero1316, 2018-09-04:
hi @TianHao-Espressif i just want to ask if there's  an example to test the performance of WiFi/BLE coexistence provide in idf or would be okay to share the code you used to test the performance WiFi/BLE coexistence , as currently i have stability problem in both of them , my code is just scanning for beacons and have MQTT for another usage

lukecyca, 2018-09-16:
I've tried all combinations of settings suggested in this thread and cannot get any reliability. I wish to continuously scan for BLE broadcasts, filter and decode some of them, and transmit over MQTT.

Typically this works for about a minute, then the MQTT connection goes down, presumably because of interference from the bluetooth radio.

Does anyone have an example that is working reliably?

chegewara, 2018-09-16:
@lukecyca What scan_interval and window_interval values did you try? Usually beacons and mid performance ble device should have advertising interval around 250ms, so lets assume you may want to scan for devices with scan_interval around 250-350. Now we have 3 channels, so you can set window_interval around 3*350~=1200ms. I would start with those values:
- scan_interval = 350ms 
- window_interval = 1200ms

lukecyca, 2018-09-17:
Thanks for the reply @chegewara.

I've tried:
* `.scan_interval = 0x50, .scan_window = 0x30` as per [the example](https://github.com/espressif/esp-idf/blob/aaf12390eb14b95589acd98db5c268a2e56bb67e/examples/bluetooth/ble_ibeacon/main/ibeacon_demo.c#L44)
* `.scan_interval = 0x550, .scan_window = 0x380` as per @hopkinskong's [comment](#issuecomment-378137438) and following 
@TianHao-Espressif's advice for scan_interval - scan_window > 150.
* `.scan_interval = 0x350, .scan_window = 0x1200` but I'm pretty sure the interval must be larger than the window ([per the BLE 4.0 core spec](https://electronics.stackexchange.com/a/214013)). So I also tried the reverse in case that's what you meant.

In all cases it fails in one of two ways:
* No BLE packets are captured at all (but MQTT is working and is transmitting other telemetry); or
* Packets are captured and transmitted over MQTT for 10-20s and then MQTT connect fails. I have it reconnect after 10s and it works again for 10-20s, then fails again.

I am still using esp-idf 3.0.3 (so no software co-existance) for the above.

I also tried esp-idf 3.1 with software co-existance (preferring wifi) and repeated all of the parameters above, with the same result.

Setting a larger interval reduces the number of BLE packets received (makes sense) but otherwise seems to have no effect on whether the MQTT connection gets clobbered.

chegewara, 2018-09-17:
You're right, scan_interval should be greater value, my bad. 
When you set scan_interval to 0x50 (50ms) then you will miss most of advertising packets/devices,but its your app and you know better what you need.

I have arduino app on esp32 that is connected with aws IoT over mqtt and when is connected i am performing scan for ble thermal printer and connect to it. My app is subscribing to individual topic, then every few seconds i am publishing some value and receiving response on subscribed topic. Depends on response value i am printing or not some message on thermal printer. I had no issue with this solution except it requires a lot of heap and its very hard to fit it on wroom module.
There is few difference between my and your app, i scan only for few seconds and send just a few bytes over wifi.