espressif/esp-idf #1659: Monitor randomly repeats characters

bmakovecki, 2018-02-26:
I'm having trouble with make monitor as randomly characters repeats. 
Im runing make monitor on Win10, latest toolchain and latest IDF with CONFIG_MONITOR_BAUD=115200

BTW: I have tried other terminals like PuTTY and others .. and they all seems to work fine

What can be cause ?

Regards 
Boris Makovecki
---------------------------------------
Hello world!
This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external flash
Restarting in 10 seconds...
Restarting in 99 seconds...
Restarting in 8 secconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in  4 seconds...
Restarting inn 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.

negativekelvin, 2018-02-27:
https://github.com/espressif/esp-idf/commit/3e83cfd77c63d36bf3d3776b1dcb30130190406b

projectgus, 2018-02-27:
Yes, as @negativekelvin indicates this was a regression when fixing the crashing bug #1136. Should be fixed in master now. Thanks for taking the time to report it.

closed 2018-02-27 by projectgus

bmakovecki, 2018-02-27:
I'm confirming that issue is solved now. Thanks.