espressif/esp-idf #1660: VFS/FAT get free space

benjaminaigner, 2018-02-27:
Hi @all,

I wanted to find a way to get the free space of a FAT FS, implemented like the wear_leveling example.

There is no API call within your esp_vfs* headers, but in ff.c, there is following function:

FRESULT f_getfree (const TCHAR* path, DWORD* nclst, FATFS** fatfs);

Is there any way to get the FATFS pointer after mounting the partition?

The pointer is used by esp_vfs_fat_spiflash_mount, but discarded afterwards.

Do you have any suggestions how to implement this the best way?
Or should I wait for an official release :-)?

BTW keep up your work, impressive progress!

seaduck13, 2018-02-28:
I would also be interested in getting an answer to this.  Is it a combination of getting the free clusters as benjamineaigner suggested multiplied by the size of a cluster, or is there an easier way to get free kB or something?  Thank you.

igrr, 2018-03-05:
It looks like you don't need know the `FATFS` pointer to call `f_getfree`. Here's the example code from fatfs manual:

```c++
    FATFS *fs;
    DWORD fre_clust, fre_sect, tot_sect;

    /* Get volume information and free clusters of drive 0 */
    res = f_getfree("0:", &fre_clust, &fs);
    /* Get total sectors and free sectors */
    tot_sect = (fs->n_fatent - 2) * fs->csize;
    fre_sect = fre_clust * fs->csize;

    /* Print the free space (assuming 512 bytes/sector) */
    printf("%10lu KiB total drive space.\n%10lu KiB available.\n",
           tot_sect / 2, fre_sect / 2);
```

(The original example uses drive 1:, however if you are only using one FATFS instance, it will map to drive 0:).

benjaminaigner, 2018-03-06:
@igrr 
Works like a charm :-)

Thank you very much for the code example.

closed 2018-03-06 by benjaminaigner

snahmad, 2019-03-12:
How Do I find sector size for SDCard.

Is it CONFIG_WL_SECTOR_SIZE=??? define in menuconfig.

I specified allocation unit=64k for SD card mount

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
    sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
	 // Options for mounting the filesystem.
    // If format_if_mount_failed is set to true, SD card will be partitioned and
    // formatted in case when mounting fails.
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = true,
        .max_files = 5,
        .allocation_unit_size = 64 * 1024
    };

snahmad, 2019-03-12:
For 8 GB SD card, I am getting total bytes = 3.644850176 GB.

snahmad, 2019-03-13:
Working now. My mistakes.

cast DWORD 32-bit to 64 bit before multiple.

u64 tmp_total_bytes = (64)tot_sect * FF_SS_SDCARD;
u64 tmp_free_bytes = (64)fre_sect * FF_SS_SDCARD;