espressif/esp-idf #1661: sntp_reqeust is a private method - how can I trigger sntp request without polling mode?

uladkasach, 2018-02-27:
the SNTP component [provides public methods]
(https://github.com/espressif/esp-idf/blob/master/components/lwip/include/lwip/apps/sntp/sntp.h)
`sntp_init` and `sntp_stop`, but does not provide the method `sntp_request` publicly. Because of this I can not find a way to manually trigger the `sntp_reqeust` without starting the module in `polling` mode. 

Is there any way to manually trigger an sntp request without polling mode? If not, could we make the `sntp_request` method public?

[sntp component code for reference](https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/sntp/sntp.c)

tferrin, 2018-02-28:
sntp_request() is an internal function and not meant to be called from an application. It is called automatically when sntp is running in OPMODE_POLL mode. The only other mode is OPMODE_LISTENONLY in which case sntp passively listens for broadcast packets from an ntp server and uses those to synchronize time. So there is no need to call sntp_request(). Instead you should just configure sntp to run in OPMODE_POLL mode.

closed 2018-03-06 by FayeY

Trubochkin, 2018-08-07:
Hello
I see that request interval is set by [define](https://github.com/espressif/esp-idf/blob/4480ab6c8ce70778df4948e587e712540de237eb/components/lwip/include/lwip/apps/sntp/sntp_opts.h#L129)  I need to set request interval manually in my app. How can I do it?

tferrin, 2018-08-07:
The sntp module is pre-compiled and part of the lwip library, so the only way to make a change is to recompile sntp.c. If you are using the Arduino IDE for your build environment, you can accomplish this without rebuilding the entire lwip library by adding the new define you want for SNTP_UPDATE_DELAY to tools/sdk/lwip/include/lwipopts.h and then copying the sntp.c source file to your local sketch directory. The IDE will then compile and link the local sntp.o before linking the lwip library and thus your version will get used instead of the standard sntp. Note that the NTP protocol spec defines the minimum update delay as 15 second, so SNTP_UPDATE_DELAY cannot be less than 15000.
If you are not using the Arduino IDE then I don't know how to easily accomplish this.

Trubochkin, 2018-08-07:
Thanks, but I do not use Arduino at all... only native "C"

crkrusch, 2023-11-04:
I discovered that default SNTP_UPDATE_DELAY is actually set to call a weak function called sntp_update_delay_MS_rfc_not_less_than_15000();

Because it's a weak function, you can override it and set your own update delay by creating your own external function with that name and the linker will use your function instead:

e.g. defining the following in my main code sets the refresh to every 6 minutes.

extern uint32_t sntp_update_delay_MS_rfc_not_less_than_15000() {return 360000;};

It's also possible to get notified when sntp calls the settimeofday function by registering a callback - e.g.

    settimeofday_cb([&](bool sntp) {                                // Register callback for whenever time of day is set.
        _timeSet = 1;                                               //    Possibly called via LWIP interrupt so Keep short and simple!
        _timeSntp = sntp;
    });

sntp is set to true if sntp made the settimeofday call - it's false if called by any other method.  I used the callback to verify that sntp is setting the time of day at the frequency set via sntp_update_delay_MS_rfc_not_less_than_15000()