espressif/esp-idf #1662: WiFiServer seems to hang while ESP32 is running in SoftAP mode and dealing with large file transfers

seaduck13, 2018-02-27:
I am currently running my ESP32 in SoftAP mode with an HTTP Server listening on port 80.  My project has some basic input and outputs, with the occasional large POST from the client to the server (1-3 files of up to 256Kb each).  

After a while, or one or two file uploads using POST, the WiFi server seems to hang and not receive any more clients.  Sometimes this is not permanent and after a few seconds (10-15secs) I can enter the first if statement below, sometimes it never recovers until I disconnect from the AP and reconnect.  

I had seen that there were some issues with LWIP.  I also can't seem to tell if the client.setTimeout does anything.  I've definitely got caught in while loops indefinitely even though there is no response from my browser.

For reference I'm running esp-idf from the most up to date platformio.

Below is my main FreeRTOS task for handling http requests

```
void HTTPServer::handleHTTP(bool con, bool dis)
{
	if(httpServer != NULL && (*httpServer) && httpServer->hasClient()){
		ESP_LOGD(TAG, "Found HTTP Client");
		client.stop();
		client = httpServer->available();
		ESP_LOGD(TAG, "Client connected: %d, CLient available: %d", client.connected(),  client.available());
	}

	if(httpServer != NULL && !(*httpServer))
		ESP_LOGE(TAG, "Server stopped listening");

	if(client.connected()){
		ESP_LOGD(TAG, "Client connected");
		vTaskDelay(pdMS_TO_TICKS(10));
		client.setTimeout(5);
		client.setNoDelay(true);
		while(client.available()){
			ESP_LOGD(TAG, "Got data from browser");
			readHTTP();
			if(pageReady){
				servePage();
				vTaskDelay(pdMS_TO_TICKS(10));
				client.stop();
				pageReady = false;
				Serial.println("Page served, client disconnected");
			}
		}
	}
}
```

seaduck13, 2018-03-01:
As an update, it seems that its not necessarily tied to large file transfers, but simply a function of the amount of time that the client is connected to the ESP32 AP.  I've tried bringing down the tcpip_interface using tcpip_adapter_up and tcpip_adapter_down, but that didn't seem to help.

The best solution I've come up with so far is to use esp_wifi_deauth_sta(0) periodically.  I'm not sure what  an AP_STADISCONNECTED and AP_STACONNECTED reinitializes, but it seems to refresh the connection so that its available immediately when the client requests a page.

liuzfesp, 2018-03-06:
Hi @seaduck13 is it possible to provide the whole demo code, then I can reproduce it.

zhangyanjiaoesp, 2018-03-09:
@seaduck13 
can you provide the IDF version? Or you can see the problem #1372 . Is it the same as your problem?

seaduck13, 2018-03-12:
@liuzfesp Unfortunately no, this project is sensitive so I cannot share the code more than I have.  

@zhangyanjiaoesp I believe it's ESP-IDF 2.1.1

I believe the issue is tied to the lwip_bind and lwip_listen and the time_wait state.  I could be way off though, its just what I've been looking at as a solution.  Some people suggested using socket option SO_REUSEADDR to reuse the address during bad disconnects, but I don't think it's fixed my issues completely.  I'm still getting periodic pauses where the server won't return true for hasClient().

zhangyanjiaoesp, 2018-03-28:
@seaduck13  you can try to use idf 3.0, in another question idf 3.0 work well for the similar issue(#1372). If you insist on using 2.1 version, you need to wait for some time for updating

closed 2018-05-10 by FayeY