espressif/esp-idf #1664: i2s_adc_dac

joicetm, 2018-02-28:
Hi,

I have just tried the i2s_adc_dac example and it is working fine. it recorded the audio and played well :)
But i do not know how to save the save the data from flash. should i use SPIFFS and write the data to a file or is there any other which already implemented somewhere ?

panfeng-espressif, 2018-03-07:
Hi,
What do you want to save in flash, an audio file or other data? If it is an audio file, what format is it?

joicetm, 2018-03-07:
Hi panfeng-espressif,

Thanks for the reply.
Actually i forgot to close this issue. I was reading the data from microphone through adc and wanted to do some analysis of the data. 
now i am sending the data through a tcp connection and using a python script on my system to do the analysis.
Anyway once again thanks for the support, is it okay if i re-open this issue in case of any help in future?

closed 2018-03-07 by joicetm