espressif/esp-idf #1666: lwip PPPoS close connection problem

fabriciopk, 2018-02-28:
I'm using ESP32 with GSM modem and lwip+pppos. There is a long time(10s +) between calling the function pppapi_close(ppp, 0) and the event of closing the connection.

local IP address 10.51.139.201
remote IP address 10.64.64.64
primary DNS address 8.8.8.8
secondary DNS address 8.8.4.4
ppp phase changed[1]: phase=8
pppos_netif_output[1]: proto=0x21, len = 44
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
I (29635) [SOCKET]: ... connected
pppos_netif_output[1]: proto=0x21, len = 40
pppos_netif_output[1]: proto=0x21, len = 81
I (29645) [SOCKET]: ... socket send success
pppos_input[1]: got 49 bytes
ppp_input[1]: ip in pbuf len=44
pppos_netif_output[1]: proto=0x21, len = 40
pppos_input[1]: got 45 bytes
ppp_input[1]: ip in pbuf len=40
pppos_input[1]: got 92 bytes
ppp_input[1]: ip in pbuf len=41
ppp_input[1]: ip in pbuf len=40
pppos_netif_output[1]: proto=0x21, len = 40
I (31445) [SOCKET]: Received from server 255
pppos_netif_output[1]: proto=0x21, len = 40
**ppp_close[1]: kill_link -> lcp_close** _<<<<<<< ---------------- call pppapi_close(ppp, 0);_
ppp phase changed[1]: phase=9
ppp phase changed[1]: phase=7
sifvjcomp[1]: VJ compress enable=0 slot=0 max slot=0
sifdown[1]: err_code=5
ccp_set[1]: is_open=0, is_up=0, receive_method=0, transmit_method=0
ppp phase changed[1]: phase=4
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=0 0 A 0
ppp_recv_config[1]
pppos_write[1]: len=20
pppos_write[1]: len=20
ppp phase changed[1]: phase=10
Connection terminated.
ppp phase changed[1]: phase=0
ppp_link_terminated[1]
ppp_end[1]
**E (43015) [PPPOS CLIENT]: status_cb: User interrupt** _<<<<<<----------- closing event_
**ppp_link_terminated[1]: finished.** _<<<<<<----------- end of ppp connection_

I'm wondering if it is possible to reduce the time between call pppapi_close and the actual closing event? I already tested the modem with other setup(linux + pppd),  the time to close the connection is shorter.

liuzfesp, 2018-03-10:
Hi @fabriciopk could you share me the Demo code please? Then I can reproduce it myself.

liuzfesp, 2018-03-10:
@zhangyanjiaoesp please help to following this issue.

Alvin1Zhang, 2019-06-14:
@fabriciopk Sorry for getting back to you so late... Would you help share whether you still have the issue you mentioned in this ticket? If yes, would you kindly help share the demo code so that we could try to reproduce? Thanks.

fabriciopk, 2019-06-17:
Hi @Alvin1Zhang,
This was the demo code that I was using: 
https://github.com/fabriciopk/esp32_integrations/blob/97ed44dad56f20e7601b997b9e987976d9b6c8ac/main/Gprs.cpp#L303

suda-morris, 2019-09-29:
@fabriciopk Can you try the pppos example in esp-idf, if there's still a long time between calling the function pppapi_close(ppp, 0) and the event of closing the connection?

Alvin1Zhang, 2019-10-12:
@fabriciopk Thanks for reporting. Would you please help try the pppos example in esp-idf, if there is still a long time between calling the function pppapi_close(ppp, 0) and the event of closing the connection? Thanks.

Alvin1Zhang, 2020-05-26:
Will close due to short of feedback, feel free to reopen with more details. Thanks.

closed 2020-05-26 by Alvin1Zhang