espressif/esp-idf #1669: missing bytes in BLE advertising (Eddystone-UID)

MartinSadovy, 2018-03-01:
Hi, I'm trying to advertise Eddystone-UID. I checked this issue with reference code example (examples/bluetooth/ble_adv).

I replaced service name sequence by Eddystone-UID sequence in example code and run it.

When byte 27 is 0x(2-0)(0-f), broadcasting sequence ends on this byte. (code below)

- Expected UID: 00000000000000000000-112233445566
- Observed UID: 00000000000000000000-112233440000

Chip is ESP32D0WDQ6 (revision 0)
Tested on master & 2.1.

```c
static void hci_cmd_send_ble_set_adv_data()
{
	uint8_t adv_data[31] =
	{
          2,      // Len
	  0x01,   // Type Flags
	  0x06,   // GENERAL_DISC_MODE 0x02 | BR_EDR_NOT_SUPPORTED 0x04
	  3,      // Len
	  0x03,   // Type 16-Bit UUID
	  0xAA,   // Eddystone UUID 2 -> 0xFEAA LSB
	  0xFE,   // Eddystone UUID 1 MSB
	  19,     // Length of Beacon Data
	  0x16,   // Type Service Data
	  0xAA,   // Eddystone UUID 2 -> 0xFEAA LSB
	  0xFE, // Eddystone UUID 1 MSB
	  0x00,  // Frame type: URL
	  0x01, // Power - 1dbm - https://github.com/espressif/esp-idf/blob/master/components/bt/include/esp_bt.h
	  0x00, // NID 0-9
	  0x00,
	  0x00,
	  0x00,
	  0x00,
	  0x00,
	  0x00,
	  0x00,
	  0x00,
	  0x00,
	  0x11, // BID 0-5
	  0x22,
	  0x33,
	  0x44,
	  0x55, // 0xY5 - Y >= 2
	  0x66,
	  0x00, //RFU
	  0x00, //RFU
	};
	uint8_t adv_data_len = 31;

    uint16_t sz = make_cmd_ble_set_adv_data(hci_cmd_buf, adv_data_len, (uint8_t *)adv_data);
    esp_vhci_host_send_packet(hci_cmd_buf, sz);
}
```

chegewara, 2018-03-01:
Hi
try to compare your code and advertising settings with this example. Its edystone example and i think its better start for you:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_eddystone

chegewara, 2018-03-01:
Hi @MartinSadovy I think your data lenght is too small. It should be 23 bytes and not 19 like in your code.
This way client probably is trucating data to 19 bytes and then adds 2 0x00 bytes since last 2 bytes are reserved and need to be 0x00.

MartinSadovy, 2018-03-02:
@chegewara Thank you!

closed 2018-03-02 by MartinSadovy