espressif/esp-idf #1670: Error when trying to connect to server while Bluetooth is on: bcn_timout, alloc fail

eagi223, 2018-03-01:
I'm getting the following errors when trying to connect to a server with HTTPS while Bluetooth is running. I'm using the latest IDF v3.0 commit I believe. I ran git pull on esp-idf right before testing this. Once the issue occurs, wifi disconnects and is unable to reconnect. I've combed through forums and issues trying to find a solution, but I'm stumped. Maybe I'm doing something wrong, but I've seen these error messages pop up in the issue tracker a few times for other bugs.

I (185) cpu_start: Pro cpu up.
I (185) cpu_start: Starting app cpu, entry point is 0x40081124
0x40081124: call_start_cpu1 at C:/esp-idf/components/esp32/cpu_start.c:225

I (0) cpu_start: App cpu up.
I (188) heap_init: Initializing. RAM available for dynamic allocation:
I (195) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (201) heap_init: At 3FFD2AF8 len 0000D508 (53 KiB): DRAM
I (207) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (214) heap_init: At 3FFE4350 len 00013CB0 (79 KiB): D/IRAM
I (220) heap_init: At 40093BE8 len 0000C418 (49 KiB): IRAM
I (226) cpu_start: Pro cpu start user code
I (19) esp_apptrace: Initialized TRAX on CPU0
I (20) pm_esp32: Frequency switching config: CPU_MAX: 240, APB_MAX: 240, APB_MIN: XTAL, Light sleep: DISABLED
I (25) cpu_start: Starting scheduler on PRO CPU.
I (68) esp_apptrace: Initialized TRAX on CPU1
I (5) cpu_start: Starting scheduler on APP CPU.
I (51) Main App (main): Start App
Device Name: Gunbox Pro
Firmware Version: R003541_00aa

I (118) Bluetooth Driver (dBluetooth): Initializing Bluetooth Driver
I (125) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (381) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (422) Bluetooth Driver (dBluetooth): Bluetooth Driver Initialized
I (422) Bluetooth Interface (iBluetooth): Initializing Bluetooth Interface
I (426) Bluetooth Interface (iBluetooth): GATTS Profile Event Server reg event
I (439) Bluetooth Interface (iBluetooth): create attribute table successfully, the number handle = 16

I (447) Bluetooth Interface (iBluetooth): create attribute table successfully, the number handle = 28

I (454) Bluetooth Interface (iBluetooth): gap event handler: 0
I (460) Bluetooth Interface (iBluetooth): gap event handler: 1
I (469) Bluetooth Interface (iBluetooth): SERVICE_START_EVT, status 0, service_handle 40
I (475) Bluetooth Interface (iBluetooth): SERVICE_START_EVT, status 0, service_handle 56
I (484) Bluetooth Interface (iBluetooth): gap event handler: 6
I (490) Bluetooth Interface (iBluetooth): advertising start successfully
I (498) Bluetooth Interface (iBluetooth): Bluetooth Interface initialized successfully
I (506) WiFi Driver (dWifi): Start ESP32_Wifi_Driver_Init
I (514) wifi: wifi firmware version: ebd3e5d
I (516) wifi: config NVS flash: enabled
I (520) wifi: config nano formating: disabled
I (524) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (533) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (565) wifi: Init dynamic tx buffer num: 32
I (565) wifi: Init data frame dynamic rx buffer num: 32
I (565) wifi: Init management frame dynamic rx buffer num: 32
I (569) wifi: wifi driver task: 3ffe3c7c, prio:23, stack:4096
I (574) wifi: Init static rx buffer num: 10
I (578) wifi: Init dynamic rx buffer num: 32
I (582) wifi: wifi power manager task: 0x3ffe8c40 prio: 21 stack: 2560
I (600) wifi: mode : sta (24:0a:c4:83:13:64)
I (601) WiFi Driver (dWifi): Task started
I (601) WiFi Driver (dWifi): SYSTEM_EVENT_STA_START
I (601) ESP32 HTTPS Driver (dESP32_HTTPS): Seeding the random number generator
I (613) ESP32 HTTPS Driver (dESP32_HTTPS): Loading the CA root certificate...
I (622) ESP32 HTTPS Driver (dESP32_HTTPS): Setting hostname for TLS session...
I (628) ESP32 HTTPS Driver (dESP32_HTTPS): Setting up the SSL/TLS structure...
I (9317) Bluetooth Interface (iBluetooth): ESP_GATTS_CONNECT_EVT, conn_id = 0
I (9317) Bluetooth Interface (iBluetooth): 5a e7 22 d4 8d 6f
I (9734) Bluetooth Interface (iBluetooth): gap event handler: 20
I (9734) Bluetooth Interface (iBluetooth): update connetion params status = 0, min_int = 16, max_int = 32,conn_int = 24,latency = 0, timeout = 400
I (9825) Bluetooth Interface (iBluetooth): ESP_GATTS_MTU_EVT, MTU 500
I (10425) Bluetooth Interface (iBluetooth): ESP_GATTS_READ_EVT
E (10425) BT: GATT_INFO_READ_EVT, conn_id 0, trans_id 1, handle 42

I (10427) Bluetooth Interface (iBluetooth): Base MAC: **:**:**:**:**:**
I (10545) Bluetooth Interface (iBluetooth): ESP_GATTS_READ_EVT
[... 86 lines cut ...]
W (27492) wifi: alloc eb len=76 type=2 fail

W (27492) wifi: m f probe req l=0

W (27992) wifi: alloc eb len=76 type=2 fail

W (27993) wifi: m f probe req l=0

W (28493) wifi: alloc eb len=76 type=2 fail

W (28493) wifi: m f probe req l=0

I (28993) wifi: ap_probe_send over, resett wifi status to disassoc
I (28993) wifi: state: run -> init (1)
I (28994) wifi: pm stop, total sleep time: 0/8088689

I (28994) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (29002) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
Wifi Serv char status read val changed to : 0
I (31422) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
Wifi Serv char status read val changed to : 0
I (33833) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
Wifi Serv char status read val changed to : 0
I (36243) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
Wifi Serv char status read val changed to : 0

eagi223, 2018-03-08:
So I doubled the amount of heap that I was allocating to both the Wifi and HTTPS tasks and I'm now getting this in the log: Has anyone got a fix or suggestion for this? I've given more info at [esp32.com](https://www.esp32.com/viewtopic.php?p=21124#p21124)
`
I (18516) WiFi Driver (dWifi): Wifi Config Bit Detected

I (18557) Bluetooth Interface (iBluetooth): ESP_GATTS_READ_EVT
E (18558) BT: GATT_INFO_READ_EVT, conn_id 0, trans_id 21, handle 58

I (18643) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (19309) wifi: state: init -> auth (b0)
I (19314) wifi: state: auth -> assoc (0)
I (19318) wifi: state: assoc -> run (10)
I (19334) wifi: connected with RPHEngineering, channel 1
I (22318) wifi: pm start, type:0

I (23653) event: sta ip: 10.42.125.96, mask: 255.255.255.0, gw: 10.42.125.1
I (23653) WiFi Driver (dWifi): SYSTEM_EVENT_STA_GOT_IP
Wifi Serv char status read val changed to : 1
System got IP: wifiServVal: 1
I (23663) WiFi Driver (dWifi): Wifi Config Good Bit Detected

I (23663) ESP32 HTTPS Driver (dESP32_HTTPS): Connected to AP
I (23675) ESP32 HTTPS Driver (dESP32_HTTPS): Connecting to 54.218.40.65:443...
I (23759) ESP32 HTTPS Driver (dESP32_HTTPS): Connected.
I (23760) ESP32 HTTPS Driver (dESP32_HTTPS): Performing the SSL/TLS handshake...
I (29844) wifi: bcn_timout,ap_probe_send_start
I (32345) wifi: ap_probe_send over, resett wifi status to disassoc
I (32345) wifi: state: run -> init (1)
I (32346) wifi: pm stop, total sleep time: 0/10027329

I (32346) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (32355) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
Wifi Serv char status read val changed to : 0
I (34774) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
Wifi Serv char status read val changed to : 0
I (37184) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
Wifi Serv char status read val changed to : 0
I (39594) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
Wifi Serv char status read val changed to : 0
I (42003) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
Wifi Serv char status read val changed to : 0
I (44413) WiFi Driver (dWifi): SYSTEM_EVENT_STA_DISCONNECTED
`

TianHao-Espressif, 2018-04-03:
@eagi223 , you can try the latest master branch that we have finish wifi/bluetooth coexistence function.

eagi223, 2018-04-03:
@TianHao-Espressif Perfect, I'll give it a try. Thanks!