espressif/esp-idf #1676: WIFI readme possibly misleading

ammaree, 2018-03-04:
In the section [https://github.com/espressif/esp-idf/blob/master/docs/api-guides/wifi.rst#system_event_sta_disconnected](url) the sentence in **bold** states that the application sockets must be **_re-created_** when the SYSTEM_EVENT_STA_DISCONNECTED event is received.

In the second last paragraph of this section it is stated that the application (socket operations?) should only be restarted on the SYSTEM_EVENT_STA_CONNECTED event.

Surely this is incorrect since sockets can only be recreated once the L3 IP connectivity has been restored, event SYSTEM_EVENT_STA_GOT_IP. 

Or has improvements been made to the lower level (L1 & L2) code such that the L3 & L4 structures have not been torn down in case of a short L2 disconnect ?

liuzfesp, 2018-03-10:
HI @ammaree you are right, it's mis-leading. Since IDFv3.0/lastest IDF, it's not a must that the applications close the sockets when SYSTEM_EVENT_STA_DISCONNECTED happens and recreate it on receiving SYSTEM_EVENT_STA_CONNECTED, the application can close the socket when SYSTEM_EVENT_STA_LOST_IP and recreate it on receiving SYSTEM_EVENT_STA_GOT_IP.

We will amend the description in IDFv3.1.

ammaree, 2018-03-11:
Hi @liuzfesp 

Tried your suggestion but it does not work.
The device never sees a SYSTEM_EVENT_STA_LOST_IP event, only ..
SYSTEM_EVENT_STA_DISCONNECTED -> SYSTEM_EVENT_STA_CONNECTED -> SYSTEM_EVENT_GOT_IP

Closing sockets in SYSTEM_EVENT_STA_LOST_IP does not work since the event does not occur, hence we get a string of fatal errors when we try to re-open the sockets on event SYSTEM_EVENT_STA_GOT_IP.

What are we missing ?

ammaree, 2018-03-14:
Hi @liuzfesp 

Do you have any suggestions for why we do not see the SYSTEM_EVENT_STA_LOST_IP  event?

We have moved out code handling the closing of sockets back to SYSTEM_EVENT_STA_DISCONNECTED as a temporary measure, but not an ideal or correct solution.

Andre

liuzfesp, 2018-03-14:
Hi @ammaree, the  SYSTEM_EVENT_STA_LOST_IP will be raised after CONFIG_IP_LOST_TIMER_INTERVAL seconds if the WiFi is disconnected from the AP, or if the DHCP failed to rebind/renew the IP address. The default value of CONFIG_IP_LOST_TIMER_INTERVAL is 120s, in your example, if the WiFi is disconnected for more than 120s, the event will be raised; on the other hand, if the IP is re-got within 120s, no SYSTEM_EVENT_STA_LOST_IP.

closed 2018-04-04 by FayeY