espressif/esp-idf #1679: Ability to disable make monitor 0x4xxx xxxx address decoding

nkolban, 2018-03-04:
I'm a big fan of ESP-IDF "make monitor".  I also appreciate the decode of executable addresses to source statements when an address in the range 0x4xxx xxxx is detected.  This issue/request asks that we have an option to disable this decoding at execution time.  For example:

```
export MONITOR_DECODE=0
make monitor
```

... or some other definition.  In the work I am doing, I am actively debugging MMU and ESP32 address processing and in my "make monitor" output, I am seeing decoded addresses everywhere when I actually don't want to see any (at this time).

jkhax0r, 2018-03-07:
Its a really cool option but would be nice to see something where we can ignore the false positive results during normal messages whenever printing any other standard hex value.  I haven't looked, but for example maybe all actual LOG messages can be modified to be prefixed with 0x whereas the stack dump isn't (or vice-versa).

FHFS, 2018-03-09:
Quick and dirty comment https://github.com/espressif/esp-idf/blob/master/tools/idf_monitor.py#L425

closed 2021-01-15 by espressif-bot