espressif/esp-idf #1681: mbedtls hangs in mbedtls_mpi_mod_mpi if using a DHE-RSA cipher

SStorm, 2018-03-05:
Example project: https://github.com/imont/ESP32_ssl_webserver

Usual `make menuconfig && make flash && make monitor`

Connect with openssl s_client: 
`openssl s_client -showcerts -pause -connect 192.168.1.110:443`
Should work fine (will most likely negotiate a ECDHE-RSA cipher).

Now try forcing use of a DHE cipher:
`openssl s_client -showcerts -pause -cipher DHE-RSA-AES128-GCM-SHA256 -connect 192.168.1.110:443`

mbedtls will hang and you'll start getting notifications like:
```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: OpenSSL_demo
```
(busylooping?)

The code is stuck here:
ssl_parse_client_key_exchange()
mbedtls_dhm_calc_secret():
```
    if( f_rng != NULL )
    {
        MBEDTLS_MPI_CHK( mbedtls_mpi_mul_mpi( &ctx->K, &ctx->K, &ctx->Vf ) );
        MBEDTLS_MPI_CHK( mbedtls_mpi_mod_mpi( &ctx->K, &ctx->K, &ctx->P ) );
    }
```

in the mbedtls_mpi_mod_mpi() function.

Note sure if problem in esp32 or mbedtls, but reporting here just in case... Will try to debug what's happening and add more details.

projectgus, 2018-08-09:
Hi @SStorm ,

Sorry for not following up on this earlier. When this happens, [do you have hardware big number acceleration enabled in menuconfig](https://docs.espressif.com/projects/esp-idf/en/stable/api-reference/kconfig.html#mbedtls-hardware-mpi)? If you change this setting, does the result change?

Angus

SStorm, 2018-08-13:
Hi @projectgus 

I did have MBEDTLS_HARDWARE_MPI enabled. When I disable it, the problem does go away. I've just tested this with the very latest master as of now (30545f4cccec7460634b656d278782dd7151098e). Here's the sdkconfig [that I'm using](https://gist.github.com/SStorm/fe72a72ec9ef4d97c175076ce1e0b1c2).

Cheers,
Roman

projectgus, 2019-03-01:
Hi @SStorm ,

Sorry for the slow follow-up. An issue was fixed in 8bba3485289c7d405 which covers the behaviour you describe. Are you still seeing this problem with ESP-IDF master branch?

SStorm, 2019-03-02:
Will test, thanks.

projectgus, 2019-06-04:
Hi @SStorm ,

Did you get a chance to test this?

Angus

SStorm, 2019-06-04:
Sorry did not have a chance. Assuming it's fixed if no one else was complaining.

projectgus, 2019-06-05:
No problems, thanks for the update. If the issue reoccurs, please let us know.

closed 2019-06-05 by projectgus