espressif/esp-idf #1682: Eddystone example WDT triggered

seopyoon, 2018-03-06:
I am trying out the Eddystone example, and it does not work at all.

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5612
load:0x40078000,len:0
load:0x40078000,len:13736
entry 0x40078fc0
I (28) boot: ESP-IDF v3.1-dev-268-gd89542eb-dirty 2nd stage bootloader
I (28) boot: compile time 13:46:10
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2b044 (176196) map
I (155) esp_image: segment 1: paddr=0x0003b06c vaddr=0x3ffc0000 size=0x02a54 ( 10836) load
I (159) esp_image: segment 2: paddr=0x0003dac8 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /Users/seopyoon/Developer/esp-sdk/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (162) esp_image: segment 3: paddr=0x0003ded0 vaddr=0x40080400 size=0x02140 (  8512) load
I (174) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x682a4 (426660) map
0x400d0018: _stext at ??:?

I (327) esp_image: segment 5: paddr=0x000a82c4 vaddr=0x40082540 size=0x113e4 ( 70628) load
0x40082540: _xt_medint3 at /Users/seopyoon/Developer/esp-sdk/esp-idf/components/freertos/./xtensa_vectors.S:1238

I (357) esp_image: segment 6: paddr=0x000b96b0 vaddr=0x400c0000 size=0x00000 (     0) load
I (368) boot: Loaded app from partition at offset 0x10000
I (368) boot: Disabling RNG early entropy source...
I (369) cpu_start: Pro cpu up.
I (372) cpu_start: Starting app cpu, entry point is 0x40080f10
0x40080f10: call_start_cpu1 at /Users/seopyoon/Developer/esp-sdk/esp-idf/components/esp32/./cpu_start.c:215

I (0) cpu_start: App cpu up.
I (383) heap_init: Initializing. RAM available for dynamic allocation:
I (389) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (396) heap_init: At 3FFCAAF8 len 00015508 (85 KiB): DRAM
I (402) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (408) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (414) heap_init: At 40093924 len 0000C6DC (49 KiB): IRAM
I (421) cpu_start: Pro cpu start user code
I (103) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (187) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (447) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 0, 0
I (477) EDDYSTONE_DEMO: Register callback
I (487) EDDYSTONE_DEMO: Start scanning...
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Btc_task
CPU 1: IDLE

```

The above is the log I get from running it.

joicetm, 2018-03-07:
Usually the WDT gets triggered when the idle task does not get any time of the processor. it seems that some function is continuously using the processor. try to add some delay in the suspect function.

Weijian-Espressif, 2018-03-12:
@seopyoon thanks for your reminder, I will fix it as soon as possible, thanks.

Weijian-Espressif, 2018-03-19:
@seopyoon we have fix it in the latest IDF, the commit is 9e8d0982642669fa0c10210a0dce49d549cb2ef9, please update to the latest IDF and try, thanks.

closed 2018-03-20 by FayeY

seopyoon, 2018-03-21:
When I flash the app, it cannot be searched.  I have tried with three different beacon scanners, one including an app made by google.
Thanks.

Weijian-Espressif, 2018-03-21:
@seopyoon Our Eddystone demo is a eddystone scanner.

seopyoon, 2018-03-21:
@Weijian-Espressif Ah! Thank you.