espressif/esp-idf #1683: Bluetooth LE HID host demo ?

Sushantnaik21, 2018-03-06:
Hi all,

i am looking for HID host, i need HID in LE, i was implementing hid_hos_demo but as it turns out it on classic. i am using ESP32 hardware
The device i am pairing is having following spec:
Bluetooth®4.0 2.4GHz, HID virtual keyboard, nRF8001 μBlue Bluetooth Low Energy Platform.

i can see HID over GATT (HOG) as per comment in [topic](https://github.com/espressif/esp-idf/issues/782) keyboard and mouse demo are up in example, but i cant see HOST DEMO.

would like yo know what need to be done for HOST application, is there ready to use example/sample code.

Thanks
Sushant

chegewara, 2018-03-06:
Hi @Sushantnaik21 
to implement HID host you need to register for notification for one specific characteristic. The problem is you will have few characteristics with the same UUID. Each characteristic will differ by descriptor value. This is descriptor and values you need to test (you are looking for input value):
https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.report_reference.xml

Now you have to understand the value that will be send in this characteristic. You will get 8 bytes each time:
- byte 1 is bit mask that will tell you if special key has been pressed (shift, ctrl etc)
- byte 2 reserved (always 0x0)
- bytes 3-8 are pressed keys value (you can have up to 6 keys strokes); now, each value has to be converted to ASCII character, this may be helpful:
https://gist.github.com/MightyPork/6da26e382a7ad91b5496ee55fdc73db2

Sushantnaik21, 2018-03-08:
hi @chegewara , thanks for reply,
sorry for late reply.

how different is hid host from HOG host? actually i need flow or algo to follow something, this is actually new thing for me.

any suggestion will be helpful

chegewara, 2018-03-08:
HID is acronym used for USB devices, HOG is equivalent used for BLE devices which means HID over GATT.

Standard USB HID is wide topic and its good to read and understand it before you even start to think about writing host driver. I can try to explain you flow but you need to very good undertand bluetooth LE mechanism, how to connect, services, characteristics and descriptors dicovering and handling GAP and GATT events.

Sushantnaik21, 2018-03-08:
Yes you are right, there is no easy way for this, i was trying to get easy solution even from another driver to copy it, but thats not way. 
i did understand that HID is protocol that needed to be implemented over GATT server/client or sdp server/client in case of EDR and i need to implement client. this will be good topic to explore

Thanks for you time and words

chegewara, 2018-03-08:
No problem, ive spent a lot time to read and understand HID and HOG to write BLE HOG library. If you have question feel free to ask and good luck.

Weijian-Espressif, 2018-03-13:
@sandeepmistry we will add HID device demo in IDF v3.1,but HID host demo is  not yet in the plan

closed 2018-04-04 by FayeY

WayneKeenan, 2019-03-25:
@Weijian-Espressif @FayeY Can a HID host demo be put on the plan please?

Weijian-Espressif, 2019-04-01:
@WayneKeenan, Sorry, HID host demo is not yet in the plan.

tharunjoy, 2019-08-06:
Hi, I really required HID host in my esp32 chipset.
Is there any update in HID host demo.

tharunjoy, 2019-08-13:
@Weijian-Espressif  Do you have any update on HID host demo

mringwal, 2020-04-20:
Hi. Check out the hog_boot_host_demo on btstack's develop branch: 
https://github.com/bluekitchen/btstack/blob/develop/example/hog_boot_host_demo.c