espressif/esp-idf #1685: Long delay between a write to UART FIFO register and data appearing on the wire

rojer, 2018-03-06:
I'm not sure if i'm missing something obvious, but i ran into this issue. a have a piece of time-critical code that needs to output to UART, so i have essentially this:

1. disable ints (by entering a critical section)
2. set a gpio to 1
3. write a byte to UART1 (i got this down to literally just `WRITE_PERI_REG(UART_FIFO_AHB_REG(1), byte);` so not much can happen here)
4. enable ints, go on

the thing is, for some reason i see a delay of 26.7 ms (yes, millseconds) between 2 and 3. is this expected?
some observations i made trying to track it down:
 - during this wait time, `UART_TXFIFO_CNT` remains at 1 and transmission begins as soon as it goes to 0.
 - `UART_ST_UTX_OUT` stays at 0 (idle), and transitions to 1 (`TX_STRT`) at the same time as fifo counter goes to 0 and transmission begins.
everything seems to behave as expected, but i cannot explain the wait between TX fifo becoming non-empty and TX FSM kicking into gear.

negativekelvin, 2018-03-07:
That is interesting because this only results in a delay of 7us
```
gpio_set_level(GPIO_OUTPUT_IO_0, 1);
uart_write_bytes(UART_NUM_1, (const char *) data, len);
```

rojer, 2018-03-07:
ok, so it's not expected / known issue. i'll try to produce an example app that demonstrates the issue.

rojer, 2018-03-07:
ok, this was misconfigured `UART_IDLE_CONF_REG`, so it took a while after previous transmission before new transmission would start.
one wonders why the defaults for idle values are non-zero, but oh well.
sorry!

closed 2018-03-07 by rojer