espressif/esp-idf #1688: smartconfig problem

Sergey081977, 2018-03-07:
Hello! I have got problem with smartconfig on Samsung S8+ with Android 8.0
Log:

I (13942) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 2
I (13962) wifi: mode : sta (30:ae:a4:3b:c9:d8)
I (14012) smartconfig: SC version: V2.7.1
I (18842) wifi: ic_enable_sniffer
I (18842) sc: SC_STATUS_FINDING_CHANNEL
I (22652) smartconfig: TYPE: ESPTOUCH
I (22652) sc: SC_STATUS_GETTING_SSID_PSWD
assertion "timer_initialized(ptimer)" failed: file "/home/sergey/esp/esp-idf/components/esp32/./ets_timer_legacy.c", line 85, function: ets_timer_arm
abort() was called at PC 0x400d79fb on core 0
0x400d79fb: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x4008d07b:0x3ffc17e0 0x4008d1d3:0x3ffc1800 0x400d79fb:0x3ffc1820 0x40089a65:0x3ffc1850 0x40110282:0x3ffc1880 0x40114b24:0x3ffc18a0 0x4011555d:0x3ffc1900 0x4011045e:0x3ffc1920 0x40104939:0x3ffc1940 0x400873c9:0x3ffc1960
0x4008d07b: invoke_abort at /home/sergey/esp/esp-idf/components/esp32/./panic.c:648

0x4008d1d3: abort at /home/sergey/esp/esp-idf/components/esp32/./panic.c:648

0x400d79fb: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40089a65: ets_timer_arm at /home/sergey/esp/esp-idf/components/esp32/./ets_timer_legacy.c:85 (discriminator 1)

0x40110282: sc_channel_timer_cb at ??:?

0x40114b24: TOUCH_Find_sync_ht20 at ??:?

0x4011555d: TOUCH_Deal_with at ??:?

0x4011045e: sc_PackCallback at ??:?

0x40104939: ppProcessRxPktHdr at ??:?

0x400873c9: ppTask at ??:?

Rebooting...
ets Jun  8 2016 00:22:57

TimXia, 2018-03-09:
@Sergey081977 Could you please tell me which router did you use? What is the commit ID of IDF? Did the issue happen every time?

Sergey081977, 2018-03-09:
I repeat this bug on routers:
Mikrotik hapAC;
Asus rt-n12;
Keenetic giga II.
 I use last id of idf.
Issue happen every time.
If downgrade id of smartconfig to version 2.6.8, then smartconfig is work.

TimXia, 2018-03-10:
@Sergey081977 What is the last id of IDF? It updates every day. Did you use the smartconfig example of IDF?

Sergey081977, 2018-03-10:
Yes. IDF automatic downloading from git. Yes I use exemple from last idf. This bag i find only with samsung S8 plus with android 8. When on samsung S8 was android 7, smartconfig work is correct.

erkki, 2018-04-01:
I'm getting the same, android 8.1.0 on google pixel, random residential AP, latest esp-idf master (da2781631455b2c87c01704503f70443799a5187). Of note, before updating idf from 84788230392d0918d3add78d9ccf8c2bb7de3152  I was getting auth timed out issues, so no panic but still timer issues.

erkki, 2018-04-01:
The problem is most likely using 802.11n and MIMO. As per http://processors.wiki.ti.com/index.php/Smart_Config_Application_development_information_-_Java_Applet#Overview. I couldn't find a way to force 802.11g on android to confirm.

TimXia, 2018-07-10:
@Sergey081977 @erkki The issue has been solved in IDF v3.1. Please take a look at it. Sorry for telling you late.

erkki, 2018-07-10:
@TimXia thanks! I'm guessing though it's better to do some form of bluetooth configuration in light of this technique being fundamentally incompatible with 802.11n and MIMO (and most likely .ac and friends?)

closed 2018-08-28 by FayeY