espressif/esp-idf #1689: aws_iot has a memory leak

jaimeyu, 2018-03-07:
Diagnosing a memory leak we're seeing in 2 of our apps with the heap tracing enabled. The app connects and disconnects from AWS periodically and we notice that we leak after each time it connects to AWS.

What I found was a very slow leak and xQueueGenericCreate() seems to be called each time.

> 11 allocations trace (300 entry buffer)

> 80 bytes (@ 0x3ffedfc8) allocated CPU 1 ccount 0x433e7330 caller 0x4008a342:0x4008a5a6:0x40123577:0x401225e8:0x40120d0c
> 0x4008a342: xQueueGenericCreate at /home/esp32/esp/esp-idf/components/freertos/./queue.c:2037
> 
> 0x4008a5a6: xQueueCreateMutex at /home/esp32/esp/esp-idf/components/freertos/./queue.c:2037
> 
> 0x40123577: aws_iot_thread_mutex_init at /home/esp32/esp/esp-idf/components/aws_iot/port/threads_freertos.c:37
> 
> 0x401225e8: aws_iot_mqtt_init at /home/esp32/esp/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_mqtt_client.c:192
> 
> 0x40120d0c: aws_iot_shadow_init at /home/esp32/esp/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src/aws_iot_shadow.c:78
> 

Ok, makes sense these mutexes are being created and dispose. So lets check if [aws_iot_thread_mutex_destroy() ](https://github.com/espressif/esp-idf/blob/02fdf8271de3a6db2ab9d4d6f023c160c84ebdbe/components/aws_iot/port/threads_freertos.c#L96) is called to free the mutex with a grep. 

>$ pwd
> /home/esp32/esp/esp-idf/components/aws_iot/aws-iot-device-sdk-embedded-C/src
>
>$ grep -r aws_iot_thread_mutex_ * 
> aws_iot_mqtt_client.c:          threadRc = aws_iot_thread_mutex_trylock(pMutex);
> aws_iot_mqtt_client.c:          threadRc = aws_iot_thread_mutex_lock(pMutex);
> aws_iot_mqtt_client.c:  return aws_iot_thread_mutex_unlock(pMutex);
> aws_iot_mqtt_client.c:  rc = aws_iot_thread_mutex_init(&(pClient->clientData.state_change_mutex));
> aws_iot_mqtt_client.c:  rc = aws_iot_thread_mutex_init(&(pClient->clientData.tls_read_mutex));
> aws_iot_mqtt_client.c:  rc = aws_iot_thread_mutex_init(&(pClient->clientData.tls_write_mutex));
> 
> $ pwd 
> /home/esp32/esp/esp-idf/components/aws_iot
> 
> $ grep -r vSemaphoreDelete *
> port/threads_freertos.c:    vSemaphoreDelete(pMutex->mutex);

So from my search, looks like the aws_iot port never frees the mutexes that it creates. This would explain the slow leak that we're seeing in our apps.

I did a test and moved aws_iot_shadow_init() so it is only init'd once at start up and this solved our memory leak issue. Unfortunately this produced a new bug but it is not ESP32 related (if you're curious, the bug is [running out of client tokens due to this one line when calling aws_iot_shadow_register_delta()](https://github.com/aws/aws-iot-device-sdk-embedded-C/blob/9ad74d845905343bcc1f327e5307af39cae5d458/src/aws_iot_shadow_records.c#L119). The tokens are never zeroed until aws_iot_shadow_init() is called again).

I don't know what the solution is yet but I'm thinking of putting aws_iot_thread_mutex_destroy() into the disconnect logic and seeing if that fixes everything nicely (and properly).

mahavirj, 2018-03-14:
@jaimeyu Thanks for reporting this issue. But it looks like something to do with AWS IoT SDK, probably you have already reported and acknowledged here, https://github.com/aws/aws-iot-device-sdk-embedded-C/issues/145. Whenever fix will be available, we will pull it in IDF. For now this issue can be closed here.

jaimeyu, 2018-03-17:
Thanks for looking into it. I agree, when I opened the bug, I was basing it out of my system path but I realized afterwards the esp IDF was using a sub git project.

closed 2018-03-17 by jaimeyu