espressif/esp-idf #1691: Where to find info about the new two point adc calibration of modules/chips?

chmorgan, 2018-03-08:
I'm interested in more details about the two point adc calibration of modules/chips. Is this going to occur at the factory? Does that affect 3rd party producers of modules that buy chips? Are they expected to perform calibration based on the module power supply? (For example I'm using the ALB module with additional ram and flash). If this is a factory calibration, when is this going to be rolled out?

igrr, 2018-03-12:
Since this is more a product-related question, rather than a feature request or bug report related to IDF, i'm going to close this issue. Please subscribe to PCN notifications at https://www.espressif.com/en/subscribe to get product related changes, and use the form on the website (https://www.espressif.com/en/company/contact/pre-sale-questions-crm) for product related questions.

closed 2018-03-12 by igrr

chmorgan, 2018-03-12:
Thanks Ivan. I've subscribed to PCNs.

mws-rmain, 2018-04-20:
I discovered this post while seeking the same info. To save everyone else trying to optimize the ADC time & effort, the details are [here](http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/adc.html). Although a bit dated, there are further details in [this thread](https://www.esp32.com/viewtopic.php?t=2334)