espressif/esp-idf #1692: Access to wifi nvs routines or a way to control when the wifi radio mode is persisted to flash vs. kept in ram?

chmorgan, 2018-03-08:
I have an application that switches between sta and ap modes every few dozens of seconds until a connection is established on the sta, in which case it goes into sta+ap, or on the ap, in which case it stays in ap mode. The design is intended to automatically reconnect to an ap that went away, or present the esp32 as an ap for configuration.

The problem I've run into is that switching between these modes is causing writes to nvs as the wifi data is persisted to flash. I'm concerned about burning out the flash if the unit sits for days in this mode.

I could change the timings to reduce the frequency of radio mode changes but I'm not convinced that this is the best user experience yet.

Rather, I'd like to make the wifi nvs code smarter for my particular case, and persist configuration (wifi_config_t) to flash, but only store the radio mode (ap/sta/apsta) back to flash when a successful connection is made to the users AP. As I'm only changing the radio mode this should remove almost all of the writes to nvs.

I looked for the wifi nvs functions but they appear baked into the 80211 library.

Thoughts on the best approach to address this? I'd rather not have to reimplement all of the nvs persistence of wifi parameters myself.

liuzfesp, 2018-03-21:
HI @chmorgan why not always use APSTA mode? If you don't want to store the configuration to flash, call esp_wifi_set_storage(WIFI_STORAGE_RAM) before calling esp_wifi_set_mode or esp_wifi_set_config; if you want to store the configuration to flash, call esp_wifi_set_storage(WIFI_STORAGE_FLASH).

chmorgan, 2018-03-21:
Hi @liuzfesp.

APSTA, when not connected to an STA, results in the radio changing channels
to find the STA. This disrupts the AP connection as the AP is on whatever
radio channel is being used for the STA.

I ended up always using ram mode and saving to my own nvs variable the
wifi_config structures. It was easier than trying to figure out how to stop
the WiFi code from storing back to nvs at initialization time. Now I can
achieve zero spi flash writes.

jack0c, 2018-03-22:
@liuzfesp @FayeY looks problem solved.

closed 2018-04-03 by FayeY