espressif/esp-idf #1693: Recursive calls to vfs_fat system cause infinite lock

mill1000, 2018-03-08:
Recursive calls to the vfs_fat system will result in an infinite lock. Likely other VFS systems are equally affected.

The problem is a result of redirecting the ESP Logging to a file on the SD card. If an SD card operation fails, the ESP_LOG calls within vfs_fat and sdmmc_cmd attempt to access the VFS again which will spin indefinitely on the lock.

The sequence we've seen:
1. A task makes a called to fopen
2. VFS_FAT lock is acquired
3. The fopen call fails due to a SD card error in sdmmc_cmd
4. sdmmc_cmd makes a call to ESP_LOGE
5. esp_log_write calls s_log_print_func which has been redirected to our function
6. Our function makes a call to fopen
7. VFS_FAT is still locked and calling task spins blocks forever.

dobairoland, 2018-11-09:
Hi @mill1000! Thanks for reporting this issue and sorry for the late reply.

Yes, you are right that without releasing the lock no logger function should be called. I'll make a correction as soon as possible.

However, you will still need to guard against such situations in your code because by removing the problem with locks the application will get into an infinite recursion: `fopen` -> failure -> `ESP_LOGE` -> `fopen` -> failure -> `ESP_LOGE` and so on until stack overflow occurs. Even if you move `fopen` and `fclose` outside the logger function and keep only `vfprintf` you can get stuck: `vfprintf` -> `fwrite` -> failure -> `ESP_LOGE` -> `vfprintf` -> `fwrite` -> failure -> `ESP_LOGE` and so on. The solution could be to turn off the logging in the begining of your custom logger function and turn back on at the end:
```c
static int logger(const char *msg, va_list va)
{
    int ret = -1;
    esp_log_level_set("*", ESP_LOG_NONE);
    FILE *f = fopen("/sdcard/flog.txt", "w");
    if (f) {
        ret = vfprintf(f, msg, va);
        fclose(f);
    }
    esp_log_level_set("*", ESP_LOG_INFO);
    return ret;
}

esp_log_set_vprintf(logger);

```

Is this solution sufficient for you?

mill1000, 2018-11-16:
Hi. Thanks for getting back. Yes we ended up using a similar solution, but only changing the vfs_fat log level.

We also implemented a buffered monitor. So the calls to ESP_LOG write into a buffer, and then the writes are protected as you described. The buffered monitor had the advantage of letting us maintain monitor logging while the SD card was in error. At least for a short duration.

dobairoland, 2018-11-16:
Thanks for the feedback! I'm happy that it got solved.

This issue will be closed when the cleanup of the critical sections in VFS FATFS gets pushed into master.

closed 2018-11-18 by igrr