espressif/esp-idf #1695: Erro after every 9sec and ESP32 is rebooting!!

sjoteppagol, 2018-03-09:
Hello All,
This is very strange issue I am facing from few days, I tried to resolve but I couldn't .
Here I am not actually getting root cause for this issue.
I have 

 `    xTaskCreate(main_task, "main_task", 2048+1024, NULL, 10, NULL);    

//  xTaskCreate(sensor_task, "sensor_task", 2048, NULL, 10, NULL) ; // This task causing issue
    xTaskCreatePinnedToCore(aws_iot_task, "aws_iot_task", 15 * 1024, NULL, 15, NULL, 1);
    xTaskCreate(power_task, "power_task", 2048, NULL, 15, NULL);
    xTaskCreate(rx_task, "uart_rx_task", 2048, NULL, 10, NULL);
    ble_task();`

**E (16056) aws_iot: failed! mbedtls_ssl_handshake returned -0x10
E (16056) APP_DEBUG: aws_iot_shadow_connect returned error -4, aborting...
abort() was called at PC 0x4010cd7b on core 1**

**Backtrace: 0x4008c5f4:0x3ffdf930 0x4008c6f3:0x3ffdf950 0x4010cd7b:0x3ffdf970**

Thank you
Shridhar

negativekelvin, 2018-03-09:
https://github.com/espressif/esp-idf/issues/346

FayeY, 2018-03-19:
Hi, did the above suggestion help you resolve the issue?

sjoteppagol, 2018-03-19:
Hello FayeY,
I keep getting reboot once I get 
**E (12125) aws_iot: failed! mbedtls_ssl_handshake returned -0x4310
E (12135) APP_DEBUG: aws_iot_shadow_connect returned error -4, aborting...
abort() was called at PC 0x400d50bf on core 1**
I have attached the log for your reference.
Please advice
Thank you
[image]

negativekelvin, 2018-03-19:
Did you read the other issue about running out of memory when using aws_iot and Bluetooth together?

sjoteppagol, 2018-03-19:
Hello negativekelvin,
Yes. But how to overcome this issue !?

sjoteppagol, 2018-03-21:
Hello Negativekelvin,
Thank you. I gone through your provided link one more time and tried to decrease mbedTLS message length size and it is working now.

Thank you once again.

closed 2018-03-21 by sjoteppagol