espressif/esp-idf #1696: Deadlock @ mdns_free()

onb3891, 2018-03-09:
mdns_free() calls function _mdns_service_task_stop() 

mdns_free()  --> _mdns_service_task_stop() {
MDNS_SERVICE_LOCK()  --> Step 1 
Post ACTION_TASK_STOP to task and  wait on this while loop
          while (_mdns_service_task_handle) {                                                                   
              vTaskDelay(10 / portTICK_PERIOD_MS);                                                              
          } 

MDNS_SERVICE_UNLOCK();
}

on _mdns_service_task() {

Wait on queue messages()                --> Step 2 
if(action == ACTION_TASK_STOP)
{
   break; //exit the task
}
MDNS_SERVICE_LOCK();                                --> Step 3                                          
 _mdns_execute_action(a);                                                                      
 MDNS_SERVICE_UNLOCK();  

}

If Step 1 happens and on Step2 if there are other actions pending on queue mdns task will be stuck at Step3

This can be easily reproduced if  mdns_service_remove_all() is called just before mdns_free(). (happens 1 in 10 times)

negativekelvin, 2018-03-09:
@me-no-dev

abhimanyuv1, 2018-06-12:
Suggested fix:
[mdns.patch.tar.gz](https://github.com/espressif/esp-idf/files/2093296/mdns.patch.tar.gz)

I was able to reproduce this easily with softAP created on ubuntu machine.

david-cermak, 2019-03-13:
Hi, this bug has already been addressed in https://github.com/espressif/esp-idf/commit/6bb68a5a7567a94c3605136d44960ff060c74663 and https://github.com/espressif/esp-idf/commit/206b47c03aca0acdf40d1d9c250e18aeddfe1bd7.

There's still one possible deadlock (yet very improbable) calling `_mdns_stop_timer();` when mutex is locked (if a timer task executed before unlocking). There's one commit fixing this issue in an internal repo, so I will update the commit message to close this issue.

closed 2019-03-14 by igrr