espressif/esp-idf #1698: Guru Meditation Error: Core  0 panic'ed (LoadProhibited) . Exception was unhandled.

JLIspace, 2018-03-10:
For some reason my program, which worked without a problem liek a week ago, stopped working today.
(I also tried definitely working back-up versions)
It's based on the a2dp_sink and after connecting, this is what happens:

`I (193) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=128
I (193) I2S: PLL_D2: Req RATE: 44100, real rate: 44642.000, BITS: 16, CLKM: 14, BCK: 8, MCLK: 11289966.924, SCLK: 1428544.000000, diva: 64, divb: 11
I (203) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (513) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x400014fd  PS      : 0x00060f30  A0      : 0x8004f20f  A1      : 0x3ffb18c0
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb18d0
A10     : 0x00000003  A11     : 0x00060b23  A12     : 0x00060b20  A13     : 0x00000001
A14     : 0x00060b20  A15     : 0x00000005  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400014fd:0x3ffb18c0 0x4004f20c:0x3ffb18d0 0x4002509e:0x3ffb18f0 0x4013d409:0x3ffb1930 0x4001a637:0x3ffb1950 0x40019d11:0x3ffb1980 0x40055b
4d:0x3ffb19a0 0x4013b7bb:0x3ffb19c0 0x4013b80b:0x3ffb19e0
0x4013d409: lc_lmp_rx_handler at ??:?

0x4013b7bb: r_rw_schedule at ??:?

0x4013b80b: btdm_controller_task at ??:?

Rebooting...
`

Any ideas? ._.

JLIspace, 2018-03-10:
Got it. It seems like erasing all the flash again did the trick.

closed 2018-03-10 by JLIspace

qcarver, 2019-03-29:
how did you "erase all the flash"? Thank you. I have seen this problem too.

JLIspace, 2019-03-29:
> how did you "erase all the flash"? Thank you. I have seen this problem too.

I'm not 100% sure anymore but I think that the error was caused by using "make app-flash" after I flashed a different program (with different settings) to the esp32 before. You should definitely flash the whole thing when starting on a esp as "make app-flash" only uploads the app itself.

By using "make flash" the whole program get's compiled and flashed again (including app, bootloader and partition table)

see https://docs.espressif.com/projects/esp-idf/en/latest/get-started/make-project.html