espressif/esp-idf #1699: Reconnecting to BT SPP causes panic assertion

j-e-f-f-williams, 2018-03-10:
Have been trying to ass BT classic SPP to a project and after I disconnect from the ESP if I try to connect again I get the following assertion.  My test app is based on the SPP acceptor (attached).

- connect to the ESP32 as a BT SPP  (I use Bluetooth SPP Pro app on Android tablet)
- disconnect from BT SPP
- try to connect again to the ESP32 with BT SPP

`E (30394) BT: port_rfc_closed RFCOMM connection in state 2 closed: Closed (res: 19)
I (30394) SPP_ANDROID: SPP Event Handler
I (30394) SPP_ANDROID: ESP_SPP_CLOSE_EVT
I (30394) SPP_ANDROID: isConnected = 0
I (30404) SPP_ANDROID: connectionHandle = 0
E (31254) BT: rfc_find_lcid_mcb LCID reused LCID:0x43 current:0x0
E (31254) BT: RFCOMM_DisconnectInd LCID:0x43
I (47734) SPP_ANDROID: SPP Event Handler
I (47734) SPP_ANDROID: ESP_SPP_SRV_OPEN_EVT
I (47734) SPP_ANDROID: isConnected = 1
I (47744) SPP_ANDROID: connectionHandle = 385
I (47834) SPP_ANDROID: SPP Event Handler
I (47834) SPP_ANDROID: ESP_SPP_WRITE_EVT
assertion "!list_is_empty(list)" failed: file "C:/dev/esp-idf/components/bt/bluedroid/osi/list.c", line 82, function: list_
front
abort() was called at PC 0x400d45ff on core 0
0x400d45ff: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/
../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x4008b488:0x3ffd7110 0x4008b65f:0x3ffd7130 0x400d45ff:0x3ffd7150 0x400e8626:0x3ffd7180 0x400edb0d:0x3ffd71a0 0x
400e9232:0x3ffd71f0
0x4008b488: invoke_abort at C:/dev/esp-idf/components/esp32/panic.c:648

0x4008b65f: abort at C:/dev/esp-idf/components/esp32/panic.c:648

0x400d45ff: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/
../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x400e8626: list_front at C:/dev/esp-idf/components/bt/bluedroid/osi/list.c:214

0x400edb0d: btc_spp_cb_handler at C:/dev/esp-idf/components/bt/bluedroid/btc/profile/std/spp/btc_spp.c:503

0x400e9232: btc_task at C:/dev/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:104
`

[main.c.zip](https://github.com/espressif/esp-idf/files/1799017/main.c.zip)

negativekelvin, 2018-03-10:
https://github.com/espressif/esp-idf/issues/1537

j-e-f-f-williams, 2018-03-11:
Closed as duplicate of #1537 referenced above.  Confirm that change solves my issue as well

closed 2018-03-11 by j-e-f-f-williams