espressif/esp-idf #1701: IO09 as PWM

hvishu, 2018-03-10:
In our project (using WROOM module), we want to use three pwm's.  IO02 and IO05 are working fine.  But if we try to use IO09, the 
[mcpwm_brushed_dc_control_example.c.zip](https://github.com/espressif/esp-idf/files/1799233/mcpwm_brushed_dc_control_example.c.zip)
app gets reset.

I have modified version of "mcpwm_brushed_dc_control_example.c" to test this (see attached).  If GPIO_PWM0A_OUT is configured as 9,  I see this problem.  Is there any limitation in using IO09 as PWM?

Regards
VIshwa

negativekelvin, 2018-03-10:
GPIO6-11 are normally reserved for SPI flash

kewalmshah, 2018-03-10:
https://www.espressif.com/sites/default/files/documentation/esp-wroom-32_datasheet_en.pdf
Please refer to section 2.2 and Note on the bottom of page 4.

hvishu, 2018-03-11:
It say's not recommended, does that mean it can't be used at all?  In this project, we are kind of short on IO's.

negativekelvin, 2018-03-11:
>does that mean it can't be used at all?
Yes

closed 2018-03-19 by FayeY