espressif/esp-idf #1703: Windows Setup - make: *** No rule to make target

mrmonteith, 2018-03-10:
Trying to setup in a Windows environment.
In the Configure section I tried to run:
cd ~/esp/hello_world
make menuconfig
I get an error:
$ make menuconfig
Makefile:8: C:/mysys32/MichaelMonteith/esp/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target 'C:/mysys32/MichaelMonteith/esp/esp-idf/make/project.mk'.  Stop.
The files in the directory are:
MichaelMonteith@DESKTOP-B761DNH MINGW64 ~/esp/hello_world
$ ls -la
total 2
drwxr-xr-x 1 MichaelMonteith MichaelMonteith   0 Mar 10 17:23 .
drwxr-xr-x 1 MichaelMonteith MichaelMonteith   0 Mar 10 17:23 ..
drwxr-xr-x 1 MichaelMonteith MichaelMonteith   0 Mar 10 17:23 main
-rw-r--r-- 1 MichaelMonteith MichaelMonteith 183 Mar 10 17:23 Makefile
-rw-r--r-- 1 MichaelMonteith MichaelMonteith 170 Mar 10 17:23 README.md

negativekelvin, 2018-03-10:
Did you set IDF_PATH ?

mrmonteith, 2018-03-10:
$ echo $IDF_PATH
C:/mysys64/MichaelMonteith/esp/esp-idf

negativekelvin, 2018-03-10:
Is it "mysys32" or "mysys64" or is it actually msys32?

mrmonteith, 2018-03-11:
It's mysys64 and I was using mingw32.exe.  I tried all this first with the one zip file and then the manual way and had the same problem.  However the zip file uses mysys32 and the manual way way mysys64.   I was trying to find which way worked in case I did something wrong.

negativekelvin, 2018-03-11:
What is the output of 
'ls /c/msys32 
And
'ls /c/mysys32

mrmonteith, 2018-03-11:
On Windows you have to do:
ls C:/msys64
$ ls C:/msys64
autorebase.bat  home                 mingw32      mingw64.ini      network.xml
components.xml  InstallationLog.txt  mingw32.exe  msys2.exe        opt
dev             maintenancetool.dat  mingw32.ini  msys2.ico        tmp
dir             maintenancetool.exe  mingw64      msys2.ini        usr
etc             maintenancetool.ini  mingw64.exe  msys2_shell.cmd  var

negativekelvin, 2018-03-11:
Both work, but ok.  So you show  `IDF_PATH=C:/mysys64/MichaelMonteith/esp/esp-idf` and `ls C:/msys64`. The IDF_PATH has an extra 'y' which is what I said before.

mrmonteith, 2018-03-11:
Missed that.  Yes. I corrected the IDF path.  
$ echo $IDF_PATH
C:/msys64/MichaelMonteith/esp/esp-idf
$ ls C:/msys64
autorebase.bat  home                 mingw32      mingw64.ini      network.xml
components.xml  InstallationLog.txt  mingw32.exe  msys2.exe        opt
dev             maintenancetool.dat  mingw32.ini  msys2.ico        tmp
dir             maintenancetool.exe  mingw64      msys2.ini        usr
etc             maintenancetool.ini  mingw64.exe  msys2_shell.cmd  var

 Unfortunately that didn't solve the problem.

negativekelvin, 2018-03-11:
Probably because it is supposed to be `C:/msys64/home/MichaelMonteith/esp/esp-idf`

mrmonteith, 2018-03-11:
Good grief.  (face in shame).  Thanks  That worked.

negativekelvin, 2018-03-11:
Nice.  Go ahead and close the issue.

mrmonteith, 2018-03-11:
Corrected IDF_PATH to C:/msys64/home/MichaelMonteith/esp/esp-idf

closed 2018-03-11 by mrmonteith

anlpereira, 2018-06-11:
My dear friends, I'm facing this same problem. Can you help me?

$ echo $IDF_PATH
C:/msys32/home/anlpe@Derso-PC/esp/esp-idf

$ make menuconfig
Makefile:8: C:/msys32/home/anlpe@Derso-PC/esp/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target 'C:/msys32/home/anlpe@Derso-PC/esp/esp-idf/make/project.mk'.  Pare.

projectgus, 2018-06-12:
Hi @anlpereira ,

My guess is that the "@" in the pathname is causing issues. If you run the `pwd` command in the MSYS terminal, what path is printed there?

anlpereira, 2018-06-12:
I got it fixed. The problem was the Environment variable path that was
incorrect.

anlpereira, 2018-06-12:
I got it fixed. It was the IDF_PATH that wasn't addressed correctly. Thanks a lot.

jktriky, 2019-04-26:
Hello, there!
My case is frustrating:
Rickys-MacBook-Air:hello_world fcwijaya$ export IDF_PATH="/Users/fcwijaya/esp/esp-idf"
Rickys-MacBook-Air:hello_world fcwijaya$ printenv IDF_PATH
/Users/fcwijaya/esp/esp-idf
Rickys-MacBook-Air:hello_world fcwijaya$ ls -al /Users/fcwijaya/esp/esp-idf/make/project.mk
-rw-r--r--  1 root  staff  28112 Apr 26 11:31 /Users/fcwijaya/esp/esp-idf/make/project.mk
Rickys-MacBook-Air:hello_world fcwijaya$ sudo make menuconfig
Makefile:8: /make/project.mk: No such file or directory
make: *** No rule to make target `/make/project.mk'.  Stop.

I really don't know what's wrong. Please help.

negativekelvin, 2019-04-27:
@jktriky `sudo -E make menuconfig`

negativekelvin, 2019-05-08:
Well obviously that is a problem. what is the output of ` ls $IDF_PATH`

jktriky, 2019-05-10:
> @jktriky `sudo -E make menuconfig`
> 
> Do you have to run as sudoer?

Yes, i have to.
sudo -E make menuconfig, it works. tks a lot.

jcabad100, 2019-05-18:
Hello everyone....i somehow managed to arrive to Step 7 on the ESP32 toolchain (under WIndows 10 x64) setup according  https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#setup-toolchain ... but i was confused if  all the line commands were supposed to be input in the shell program (mingw64.exe) because when i arrived to phython dependencies i could not figure it out how to do it with it and used the DOS cmd prompt and added D:\Programs\Python37 to the Windows system PATH and that works at the cmd prompt (starts the Phyton interpreter no matter which is the local directory. Additionally i sucessfully copied the hellow world example files using DOS cmd prompt (xcopy /e /i %IDF_PATH%\examples\get-started\hello_world hello_world).   The problem i have is that it seems that the shell program (mingw64.exe) has no access to the Windows system PATH because when building/making the hello_world example it properly start to compile and create the modules objects files *.o but the build/make process stops with the following message stating that it does not "know" python: "make[1]: Leaving directory '/home/DELL/esp/esp-idf/tools/kconfig'
/usr/bin/env: ‘python’: No such file or directory
make: *** No rule to make target 'check_python_dependencies', needed by 'menuconfig'.  Stop.
"  ...please help!

jcabad100, 2019-05-18:
i just got a message that told me that only mingw32.exe is supported (despite i have a x64 machine)....i used that instead and in fact then that shell does recognize python i.e. find the system PATH to find it...i did manage to build the menu now ...

emfior, 2019-05-27:
I have the same problem and i fix it, cause my IDF_PATH was incorrect.

Matko555, 2019-07-23:
/usr/bin/env: „python“: No such file or directory

my path is definitely correct (checked for like a 100 times now),
any idea what else can it be?

saahilshroff, 2019-11-04:
Hi,
I have been following "https://exploreembedded.com/wiki/Setting_up_Bare_Metal_Development_Toolchain_for_ESP32" to be able to work on ESP-32 using Eclipse.
When I try running the "make menuconfig". it shows me the following error: "make: *** No rule to make target 'menuconfig'.  Stop."

My IDF_PATH is "D:/SaahilESP/esp-idf". 

How do I resolve this issue? Please help! TIA.

embelgit, 2020-02-29:
Hello, there.
I'm facing a problem to make the menuconfig. I´m trying everything and nothing helps. below you can see the error: 
# make menuconfig
Makefile:8: c:/msys/home/Admin/esp/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target 'c:/msys/home/Admin/esp/esp-idf/make/project.mk'.  Stop.

jpparmar14, 2024-09-07:
Hello, there.
I'm facing a problem to make the menuconfig. I´m trying everything and nothing helps. below you can see the error:

jparmar@DESKTOP-683TKEH MINGW32 ~/example/esp-idf_template
$ make menuconfig
The following Python requirements are not satisfied:
idf-component-manager>=0.2.99-beta
kconfiglib==13.7.1
ecdsa>=0.16.0
construct==2.10.54
The recommended way to install a packages is via "pacman". Please run "pacman -Ss <package_name>" for searching the package database and if found then "pacman -S mingw-w64-i686-python-<package_name>" for installing it.
NOTE: You may need to run "pacman -Syu" if your package database is older and run twice if the previous run updated "pacman" itself.
Please read https://github.com/msys2/msys2/wiki/Using-packages for further information about using "pacman"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: (not set)
    Python interpreter used: C:/msys32/mingw32/bin/python.exe
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: C:\msys32\opt\xtensa-esp32-elf\bin;C:\msys32\mingw32\bin;C:\msys32\usr\local\bin;C:\msys32\usr\bin;C:\msys32\usr\bin;C:\Windows\System32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\msys32\usr\bin\site_perl;C:\msys32\usr\bin\vendor_perl;C:\msys32\usr\bin\core_perl;C:\msys32\mingw32\bin\
make: *** No rule to make target 'check_python_dependencies', needed by 'menuconfig'.  Stop.