espressif/esp-idf #1704: Secure boot re-flashable bootloader esptool.py --flash_size detect

catalinio, 2018-03-10:
Hi,
In the case of enabling Secure Boot and re-flashable bootloader, every time a new bootloader is released, the bootloader digest (digest+bootloader.bin) has to be flashed at address 0x0.

catalinio, 2018-03-10:
I assume, I have to obtain booloader.bin with flash_size set at compile time, so, basically to obtain a bootloader for each flash size needed.
Does anyone know where is the flash_size set in the bootloader code?

negativekelvin, 2018-03-11:
@projectgus

projectgus, 2019-03-01:
Hi @catalinio ,

Sorry for the extended silence on this issue.

This is the same root cause fixed by this PR: https://github.com/espressif/esptool/pull/386

Will fix in esptool.py soon.

closed 2019-07-18 by projectgus

projectgus, 2019-07-18:
(This issue is fixed in esptool v2.7, but we haven't updated the esptool submodule in the ESP-IDF master branch yet. This should happen in the next week or so.)