espressif/esp-idf #1706: ulp.h: stage count register related ops missing

ThomasWaldmann, 2018-03-12:
could not find anything for it in ulp.h.

or any other docs that describe the instructions' layout on a bit level.

igrr, 2018-03-12:
Please refer to the "ULP coprocessor" chapter of the ESP32 Technical Reference Manual, section "instruction formats", "operations with stage count register".

Also we don't plan to update ulp.h (the macros part, that is) anymore. This approach has been deprecated in favour of using the binutils to generate ULP coprocessor code.

ThomasWaldmann, 2018-03-12:
Thanks for the hint, looks like it is SUB_OPCODE_ALU_CNT there.

Yeah, I know it is deprecated, was just searching for instruction layout specs.

Trying to implement a simple assembler in micropython.
https://github.com/ThomasWaldmann/py-esp32-ulp

closed 2018-03-12 by ThomasWaldmann

igrr, 2018-03-12:
Looks like a very useful project @ThomasWaldmann!