espressif/esp-idf #1708: UART2 not restartee

rrobinet, 2018-03-12:
Environment:
- Arduino IDE
- ESP32 (WEMOS LOLIN32).

Hi,
My application uses the UART2 (`HardwareSerial Serial2(2)` ) 
It also uses the function  `ESP.restart()` for reconfiguration.
However, I have to force a **hardware rest** to have the UART2 active again after the ESP was restarted by software..
Looking at esp_system.h
I read:
```
  * @brief  Restart PRO and APP CPUs.
  *
  * This function can be called both from PRO and APP CPUs.
  * After successful restart, CPU reset reason will be SW_CPU_RESET.
  * Peripherals (except for WiFi, BT, UART0, SPI1, and legacy timers) are not reset.
  * This function does not return.
  */
void esp_restart(void) __attribute__ ((noreturn));
```
However I need to have the UART2 restarted; 
Is there a solution?

Thanks in advance
Robert

negativekelvin, 2018-03-12:
Probably related to these and should be resolved in arduino-esp32

https://github.com/espressif/arduino-esp32/issues/1189
https://github.com/espressif/arduino-esp32/issues/1145

rrobinet, 2018-03-13:
OK sorry 
Robert

closed 2018-03-13 by rrobinet