espressif/esp-idf #1710: esp_restart never restart

CPZbre, 2018-03-12:
Some time when we calling the function esp_restart the ESP was never restarting and and nothing else happens without hard resest or power off/on.

The call it make from main loop. The wifi and bluetooth are off and all other task are done.

we are on 3.0 branch.

liuzfesp, 2018-03-13:
@CPZbre  could you share the demo code which can reproduce this issue?

CPZbre, 2018-03-13:
This error occurs randomly with different codes.

When the fault is triggered ESP is blocked until hard reset.

Is it possible to add traces in this function to see where it could block?

We are unable to provide you with a code permitting to reproduce this malfunction.

liuzfesp, 2018-03-14:
Hi @CPZbre yes, we can add debug code, but I'm afraid it's not efficient method and it may take a long time before we can found the root cause, I will debug it if you can provide some simple demo code to reproduce it, because I just don't know how to reproduce this issue myself.

FayeY, 2018-03-19:
Hi @CPZbre , is there any update about this issue? Could you provide a test project, so we can reproduce the problem and help debug it? Thanks.

CPZbre, 2018-03-19:
Hi
I try to understand in wich context the problem it throw.

CPZbre, 2018-03-21:
Hi

We are on [Tree: e1d965e4b9](https://github.com/espressif/esp-idf/tree/e1d965e4b9aab3998d503cc32662af1a9534a174)

The case where the restart does not occur seems very particular.
Our wifi task is blocked on waiting for a stop event response for more than 60 seconds following an connection with a unresponsive from DHCP request and an low heap size 56000 bytes free (it's low for multiple TLS sessions for us).
The esp_restart it's call from main loop.
If a task is blocked can this block the restart?

liuzfesp, 2018-03-27:
Hi @CPZbre  f2132238b2a8e0d0841ab4c81a9d9c26a6b2e80d  fixes an issue which relating to esp_wifi_stop, not sure wether it can fix your issue or not. 
Could you 'git pull' the latest release/v3.0 or master branch, and check whether this issue still exists?

liuzfesp, 2018-03-27:
@CPZbre, just a little bit confusing about your last reply: 
1.  "Our wifi task is blocked on waiting for a stop event response for more than 60 seconds following an connection with a unresponsive from DHCP request" does it means the wifi task calls esp_restart if failed to got IP address?  Does the 'wifi task' mean your application task that relating to WiFi?
2.  "The esp_restart it's call from main loop" does it means esp_restart is called from main loop of your WiFi task?
It will be easy for me to understand your issue if you can give relating code directly.

CPZbre, 2018-04-10:
Hi

While we use the release of 3/21/2018 that contains the '_esp32: Fix wifi stop never returns issue_' we no longer encounter the issue.

We wait and see.

FayeY, 2018-04-24:
Closing this, please feel free to reopen if the issue persists. Thanks.

closed 2018-04-24 by FayeY