espressif/esp-idf #1713: Setting esp level log doesn't work

jstar88, 2018-03-13:
I'm using last version of esp idf.
Seems like esp_log_level_set() doesn't override the log level in menuconfig

```c
esp_log_level_set("*", ESP_LOG_VERBOSE);
ESP_LOGV(TAG_I2C, "OK _sendStopBit"); // don't work
ESP_LOGD(TAG_I2C, "OK _sendStopBit"); // work
```

```c
/**
 * @brief Set log level for given tag
 *
 * If logging for given component has already been enabled, changes previous setting.
 *
 * @param tag Tag of the log entries to enable. Must be a non-NULL zero terminated string.
 *            Value "*" resets log level for all tags to the given value.
 *
 * @param level  Selects log level to enable. Only logs at this and lower levels will be shown.
 */
void esp_log_level_set(const char* tag, esp_log_level_t level);`
```

agmuntianu, 2018-03-14:
I had a similar situation, and I noticed that changing the menuconfig log level will actually result in different compiled code size, so I guess `esp_log_level_set` only applies to log levels higher than the on in menuconfig.

jstar88, 2018-03-15:
Thank you, this behavior must be better explained in the function documentation

closed 2018-05-18 by igrr