espressif/esp-idf #1716: Secure Boot, reflashable bootloader fails if debug is not enabled in esp_image_format.c

catalinio, 2018-03-13:
Hi,
Intensively testing the secure boot (with reflashable bootloader option), it seems to fail if in the file: components/bootloader_support/src/esp_image_format.c if some debug level is not added, like:
#define LOG_LOCAL_LEVEL ESP_LOG_ERROR
#include <esp_log.h>
The `secure boot fail` is produced when a bootloader-reflash-digest.bin is flashed, and only with any debug level >=  ESP_LOG_ERROR.
Note, the flash encryption is also enabled, and this bootloader-reflash-digest.bin is pre-encrypted (on computer) before flashing it, using esptool.py.
thanks

projectgus, 2019-03-01:
Hi @catalinio ,

Sorry for not getting back to you about this earlier.

I think the root cause for this was probably fixed in 326d791ebb233930, there was a bug where digest was incorrect for certain binary sizes. Changing the local log level has an immediate impact on the final binary size, and I think in this case it happened to trigger or un-trigger the bug.

projectgus, 2019-03-26:
Hi @catalinio ,

I'm going to close this but please comment again if the issue is not resolved as expected in a current ESP-IDF release.

Angus

closed 2019-03-26 by projectgus