espressif/esp-idf #1720: A2DP SBC codec

ewoo5, 2018-03-14:
Hi,

I am trying to stream 16 kHz mono audio using the a2dp profile. Is there a way to make sure my configuration for the SBC codec is set to the right configuration? I am not sure whether it is being set to 44.1 kHz, which could have caused some bugs.

Thanks.

ewoo5, 2018-03-14:
To clarify, I am using 2 esp32 modules: one to send audio, and the other to receive this audio on another end. The audio I am sending is 16 kHz sampling rate. However, this didn't work very well as the audio is distorted at some frequencies. 

I have tested every other part on their own, and they work, so I am pretty sure the problem is with the a2dp profile I am using. My guess is that the SBC encoder the is set up to be treating the audio as 48 kHz or 44.1 kHz sampling rate. This difference between the SBC codec configuration and the actual sampling rate of 16 kHz might be causing the problem.

So the question is, how do I set the configurations for the SBC codec. Is there an api for this? I want to make sure my audio doesn't get distorted just because the codec thinks that I have a different sampling rate.

Thanks.

blueMoodBHD, 2018-03-21:
@ewoo5 According to the profile, a2dp source should supported at least one of the values between 44.1 kHz and 48 kHz. 16 kHz and 32 kHz is optional. And, we set SBC codec to  44.1 kHz, now.

If you want to send 16 kHz data, you can set SBC codec to  48 kHz and try again. The changes are as follows:
1. In bta_av_co.c line 73
`    // (A2D_SBC_IE_SAMP_FREQ_44), /* samp_freq */
    (A2D_SBC_IE_SAMP_FREQ_48 | A2D_SBC_IE_SAMP_FREQ_44), /* samp_freq */` 

2 In bta_av_co.c line 94
`#if !defined(BTC_AV_SBC_DEFAULT_SAMP_FREQ)
// #define BTC_AV_SBC_DEFAULT_SAMP_FREQ A2D_SBC_IE_SAMP_FREQ_44
#define BTC_AV_SBC_DEFAULT_SAMP_FREQ A2D_SBC_IE_SAMP_FREQ_48
#endif`

3 In btc_a2dp_source.c line 498
`    /* for now hardcode 44.1 khz 16 bit stereo PCM format */
    media_feeding.cfg.pcm.sampling_freq = 48000;//44100;`

blueMoodBHD, 2018-03-21:
@ewoo5 According to the profile, a2dp source should supported at least one of the values between 44.1 kHz and 48 kHz. 16 kHz and 32 kHz is optional. And, we set SBC codec to 44.1 kHz, now.

If you want to send 16 kHz data, you can set SBC codec to 48 kHz and try again. The changes are as follows:

In bta_av_co.c line 73
`// (A2D_SBC_IE_SAMP_FREQ_44), /* samp_freq */

    (A2D_SBC_IE_SAMP_FREQ_48 | A2D_SBC_IE_SAMP_FREQ_44), /* samp_freq */`

2 In bta_av_co.c line 94
`#if !defined(BTC_AV_SBC_DEFAULT_SAMP_FREQ)

// #define BTC_AV_SBC_DEFAULT_SAMP_FREQ A2D_SBC_IE_SAMP_FREQ_44

#define BTC_AV_SBC_DEFAULT_SAMP_FREQ A2D_SBC_IE_SAMP_FREQ_48

#endif`

3 In btc_a2dp_source.c line 498
`    /* for now hardcode 44.1 khz 16 bit stereo PCM format */

    media_feeding.cfg.pcm.sampling_freq = 48000;//44100;`

blueMoodBHD, 2018-03-27:
@ewoo5 If you don't want to work with 16-bit stereo. Just change as follows:
In btc_a2dp_source.c line 499
`media_feeding.cfg.pcm.bit_per_sample = 16;`
Set the stereo you want.

blueMoodBHD, 2018-03-27:
And set audio stream to mono, The changes are as follows:
In btc_a2dp_source.c line 1008
 just set `btc_aa_src_cb.encoder.s16ChannelMode`  as `SBC_MONO`

In btc_a2dp_source.c line 499
`media_feeding.cfg.pcm.num_channel = 1; //2;`

ewoo5, 2018-03-29:
@blueMoodBHD Thank you for your quick response. With the setup you have suggested, I have managed to send 16 kHz mono audio without distortion. However, there is another issue.

This time, the problem is at the a2dp sink instead of source. When another ESP32 receives this 16 kHz mono audio as a sink, the data passed to the callback function is actually stereo with exact duplicates on both channels.
For example, if the source sends 0xAABBCCDD, the data passed to the a2dp_sink callback function is 0xAABBAABBCCDDCCDD.

Is there a way to change it such that the data remains as mono audio? This would really help with my bottleneck issues. Thanks a lot!

blueMoodBHD, 2018-03-30:
@ewoo5 I'm sorry for forgetting to tell you , to set audio stream to mono, you also to change in bta_av_co.c line 100
`A2D_SBC_IE_CH_MD_MONO,  //A2D_SBC_IE_CH_MD_JOINT,         /* ch_mode */`

And about your new problem, we just copy the data for headset.  
if you want the data received is 0xAABBCCDD ,too. You could change in btc_a2dp_sink.c line 469
`status = OI_CODEC_SBC_DecoderReset(&context, contextData, sizeof(contextData), 2, /*2*/1, FALSE);`

FayeY, 2018-04-03:
Hi @ewoo5 , did the above suggestion help you resolve the issue?

ewoo5, 2018-04-03:
Thank you so much @blueMoodBHD , it finally worked out for me. 

@FayeY , yes my issue is resolved.

closed 2018-04-04 by FayeY

BassManTL, 2025-07-23:
@ewoo5, I too am working with 16 kHz PCM data. I am confused by the instruction to configure SBC for 48 kHz and not 16 kHz. Did you have to upsample your 16 kHz data to make it work at 48 kHz? Thanks