espressif/esp-idf #1721: SPI problem with spi_device_transmit()

seopyoon, 2018-03-14:
For communicating with an NFC reader peripheral, I am using SPI protocol.  In the code, I only use synchronous communication with the SPI device (i.e. I only call spi_device_transmit() function).

Once in a while, I would get the following error.
```
assertion "ret_trans==trans_desc" failed: file "/Users/seopyoon/Developer/esp-sdk/esp-idf/components/driver/./spi_master.c", line 783, function: spi_device_transmit
```
What does it mean to have ret_trans and trans_desc to be the same?

Spritetm, 2018-03-16:
Internally, the spi_device_trans uses spi_device_queue_trans, a function to queue up transfers, and spi_device_get_trans_result, a function to get the result of finished transfers from the queue. One of the requirements for calling spi_device_trans is that these queues must be empty, so the function can queue the transaction and then wait for its result to come in. If there still was another transaction somewhere on the queue, however, spi_device_get_trans_result will pop that off instead. In that case, spi_device_trans will see that it's not the result of the transaction it itself queued, and the assert is invoked.

seopyoon, 2018-03-16:
@Spritetm thank you for your insight. So that means, previous spi_device_transmit() call hasn't received a message back, thus the queue not being empty at that time.  Am I interpreting it correctly? It must be because the queue function hasn't returned from portMAX_DELAY? Thanks.

Spritetm, 2018-03-16:
spi_device_transmit should always get a device back, or wait for the message to come back; that's what the portMAX_DELAY is for. My guess is that either you're using spi_device_queue_trans somewhere in your code without getting the result using spi_device_get_trans_result, or calling spi_device_transmit on the same SPI device from multiple threads at the same time; both would lead to this assert triggering.

seopyoon, 2018-03-16:
Since I am only using spi_device_transmit() throughout the code, it must not be the former.  When communicating with the spi slave, I perform spi related functions upon interrupt from the slave.  There must be an issue here.  Thank you very much.

closed 2018-03-16 by seopyoon