espressif/esp-idf #1722: TG1WDT_SYS_RESET gets triggered randomly

agmuntianu, 2018-03-14:
Hi,

I am using both cores.
I have started getting this watchdog reset at random moments: 
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5880
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:15516
entry 0x40078ff8
W (91) boot: PRO CPU has been reset by WDT.
W (91) boot: WDT reset info: PRO CPU PC=0x40081aaf
0x40081aaf: .check_dport_access_end at /home/munt/esp/esp-idf/components/esp32/./dport_panic_highint_hdl.S:182

W (91) boot: WDT reset info: APP CPU PC=0x400818d2
0x400818d2: esp_dport_access_stall_other_cpu_start at /home/munt/esp/esp-idf/components/esp32/./dport_access.c:102
```
My IDF is release/v3.0 daf1d0557661bdaa1d4101ad2c7ac2d7fadc8e44 .
I might be possible this error only appeared after I upgraded my IDF , but I am not 100% sure , because I was debugging several other crashes , which now are solved, and this is now the only reset cause I see.
I have erased the flash , but the problem still persists.

Thank you for any help.
BR. Andrei.

agmuntianu, 2018-03-14:
This is the heap info just before the crash:
```
I (2595591) GCP_HTTPS: heap 134848 B / 131 kB
I (2595591) GCP_HTTPS: min heap 69072 B / 67 kB
I (2595591) GCP_HTTPS: max allocable heap 42540 B / 41 kB
```

agmuntianu, 2018-03-15:
Sometimes the reset "source" is in cpu_util:
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5544
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13732
entry 0x40078fb4
W (91) boot: PRO CPU has been reset by WDT.
W (91) boot: WDT reset info: PRO CPU PC=0x40083611
0x40083611: esp_cpu_stall at /home/munt/esp/esp-idf/components/soc/esp32/cpu_util.c:24

W (91) boot: WDT reset info: APP CPU PC=0x40081cd5
0x40081cd5: .check_dport_access_end at /home/munt/esp/esp-idf/components/esp32/./dport_panic_highint_hdl.S:181
```

agmuntianu, 2018-03-15:
I tried to invoke GDBStub , but the watchdog reset  does not trigger it

agmuntianu, 2018-03-16:
I have just tried with another controller, this time it was a revision 1 , and that does not get the wdt reset. The board I have issues with is a revision 0 of ESP32Wroom .
I am unsure if I should close this ticket, if the revision 0 is considered obsolete and fixes for it are still made.

igrr, 2018-03-18:
@agmuntianu Can you give some indication what your code is doing prior to the watchdog reset? Do you see similar behavior with any of ESP-IDF examples (on rev.0 module)?

agmuntianu, 2018-03-18:
I only saw that behaviour on the code we develop.
We are doing the following : 
* reading/writing to I2c every 2ms, the FreeRTOS tick is set to 1kHz.
* sending every 3 seconds HTTPS POST requests.

I suspected a timeout on I2C so I added timing measuring code , the I2C task executes it's code between 300 and 900 microseconds. Also no errors returned when communicating on I2C , just the watchdog reseting.

BR. Andrei

agmuntianu, 2018-04-12:
@igrr , I have narrowed down the offending code that causes tg1wdt trigger:
```C
err = mbedtls_pk_sign(&ctx, MBEDTLS_MD_SHA256, hash, sizeof(hash),
         sig, &sig_len, mbedtls_ctr_drbg_random, &ctr_drbg);
//code part of signing info in a JWT 
```

The crash occurred eventually on rev1 hardware  of esp32, and is related to the *hardware accelerated SHA encryption*. It seems the workaround for the  [3.10](https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf) bug fails sometimes.
Some other people experienced similar crashes ([same issue](https://www.esp32.com/viewtopic.php?t=4260), [hardware correlation](https://www.esp32.com/viewtopic.php?t=3623), and I can confirm that disabling the hardware acceleration for encryption fixes the watchdog reset.

If I disable both watchdogs , the controller will eventually stall (this is how I found the offending code).
The error rate is like 1 in 2000 calls. An increase in usage of the two cores seems to be correlated to the increase in failure rate.

BR. Andrei

dipensls, 2018-04-25:
Hi All ESP members,

We are also facing similar problem while doing HTTP download and writing the downloaded chunks  into flash using flash erase and flash write apis from flash_ops.c in spi_flash directory. 

We identified problem area to below api, 

`esp_err_t IRAM_ATTR spi_flash_write(size_t dst, const void *srcv, size_t size)`

Currently using IDF 2.1 and Wrover Module in our products.  

I went through the above given links and still not able  to judge if the hardware accelerators problem is with SPI _Flash as well if write operation is using flash_cache. 

Any suggestion ? 

Fail case prints : 

> [Tue Apr 24 21:52:32.147 2018] dlurl http://192.168.0.154/raj_tmp/finalBIN_3_package.bin
> [Tue Apr 24 21:52:32.838 2018] Command : $dlurl http://192.168.0.154/raj_tmp/finalBIN_3_package.bin
> [Tue Apr 24 21:52:32.838 2018] 
> [Tue Apr 24 21:52:32.838 2018] I (519892) wifi: bcn_timout,ap_probe_send_start
> [Tue Apr 24 21:52:32.838 2018] 
> [Tue Apr 24 21:52:32.838 2018] 
> [Tue Apr 24 21:52:32.838 2018] 
> [Tue Apr 24 21:52:32.838 2018] URL :: http://192.168.0.154/raj_tmp/finalBIN_3_package.bin 
> [Tue Apr 24 21:52:32.838 2018] 
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL: StartFirmwareDownLoad: 1168: firmware download start
> [Tue Apr 24 21:52:32.838 2018] E HTTP_FW_DL: StartFirmwareDownLoad: 1180: connection established
> [Tue Apr 24 21:52:32.838 2018] E HTTP_FW_DL: StartFirmwareDownLoad: 1186: @
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL: check_firmware_status: 1103: inside check_firmware_status
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL: check_firmware_status: 1109: URL = http://192.168.0.154/raj_tmp/finalBIN_3_package.bin
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL: check_firmware_status: 1110: remaining_rotation = -1
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL: check_firmware_status: 1111: firmare_length = 0
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL: check_firmware_status: 1112: flash_pointer = 210000
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL: check_firmware_status: 1113: firmare_downloaded_range = 0
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL: check_firmware_status: 1114: reset cause = 0
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL:      initialize_page:  912: inside initialize_page
> [Tue Apr 24 21:52:32.838 2018] I HTTP_FW_DL: StartFirmwareDownLoad: 1205: *********
> [Tue Apr 24 21:52:32.838 2018] E HTTP_FW_DL: update_firmware_status:  677: Before store_firmware_status_in_flash call
> [Tue Apr 24 21:52:32.852 2018] E HTTP_FW_DL: store_firmware_status_in_flash:  587: !
> [Tue Apr 24 21:52:33.625 2018] ets Jun  8 2016 00:22:57
> [Tue Apr 24 21:52:33.800 2018] 
> [Tue Apr 24 21:52:33.800 2018] rst:0x8 (TG1WDT_SYS_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
> [Tue Apr 24 21:52:33.800 2018] configsip: 159678918, SPIWP:0xee
> [Tue Apr 24 21:52:33.800 2018] clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> [Tue Apr 24 21:52:33.800 2018] mode:DIO, clock div:2
> [Tue Apr 24 21:52:33.800 2018] load:0x3fff0008,len:8
> [Tue Apr 24 21:52:33.800 2018] load:0x3fff0010,len:4260
> [Tue Apr 24 21:52:33.800 2018] load:0x40078000,len:10176
> [Tue Apr 24 21:52:33.800 2018] ho 0 tail 12 room 4
> [Tue Apr 24 21:52:33.800 2018] load:0x40080000,len:252
> [Tue Apr 24 21:52:33.800 2018] entry 0x40080034
> [Tue Apr 24 21:52:33.800 2018] W (97) boot: PRO CPU has been reset by WDT.
> [Tue Apr 24 21:52:33.800 2018] W (98) boot: WDT reset info: PRO CPU PC=0x40081afa
> [Tue Apr 24 21:52:33.800 2018] W (98) boot: WDT reset info: APP CPU PC=0x40086e13
> [Tue Apr 24 21:52:33.800 2018] I (117) boot: ESP-IDF  2nd stage bootloader
> [Tue Apr 24 21:52:33.800 2018] I (132) boot: compile time 14:35:00
> [Tue Apr 24 21:52:33.800 2018] I (186) boot: Enabling RNG early entropy source...
> [Tue Apr 24 21:52:33.800 2018] I (186) boot: SPI Speed      : 40MHz
> [Tue Apr 24 21:52:33.800 2018] I (186) boot: SPI Mode       : DIO
> [Tue Apr 24 21:52:33.800 2018] I (194) boot: SPI Flash Size : 16MB
> [Tue Apr 24 21:52:33.800 2018] I (207) boot: Partition Table:
> [Tue Apr 24 21:52:33.800 2018] I (218) boot: ## Label            Usage          Type ST Offset   Length
> [Tue Apr 24 21:52:33.800 2018] I (241) boot:  0 nvs              WiFi data        01 02 00009000 00004000
> [Tue Apr 24 21:52:33.817 2018] I (264) boot:  1 otadata          OTA data         01 00 0000d000 00002000
> [Tue Apr 24 21:52:33.817 2018] I (287) boot:  2 phy_init         RF data          01 01 0000f000 00001000
> [Tue Apr 24 21:52:33.817 2018] I (310) boot:  3 factory          factory app      00 00 00010000 00100000
> [Tue Apr 24 21:52:33.817 2018] I (334) boot:  4 ota_0            OTA app          00 10 00110000 00100000
> [Tue Apr 24 21:52:33.817 2018] I (357) boot:  5 storage          Unknown data     01 82 00318000 00080000
> [Tue Apr 24 21:52:33.817 2018] I (380) boot: End of partition table
> [Tue Apr 24 21:52:33.817 2018] I (393) boot: Disabling RNG early entropy source...
[... 50 lines cut ...]
> [Tue Apr 24 21:46:32.908 2018] I $
> [Tue Apr 24 21:46:32.908 2018] I HTTP_FW_DL:          GetFirmware:  686: I am in getfirmware task ......
> [Tue Apr 24 21:46:32.908 2018] I HTTP_FW_DL:         CreateSocket:  327: socket allocate successfully
> [Tue Apr 24 21:46:32.908 2018] I HTTP_FW_DL: update_content_length:  393: request == GET /raj_tmp/finalBIN_3_package.bin HTTP/1.1
> [Tue Apr 24 21:46:32.932 2018] Host: 192.168.0.154
> [Tue Apr 24 21:46:32.932 2018] User-Agent: esp-idf/1.0 esp32
[... cut ...]
> [Tue Apr 24 21:46:33.233 2018] I flash_operation:  store_data_in_flash:  128: flash_offset_ptr ===== 212000
> [Tue Apr 24 21:46:33.265 2018] I HTTP_FW_DL:         on_status_cb:  984: percentage == 1%
> [Tue Apr 24 21:46:33.384 2018] I flash_operation:  store_data_in_flash:  128: flash_offset_ptr ===== 213000
> [Tue Apr 24 21:46:33.466 2018] I HTTP_FW_DL:         on_status_cb:  984: percentage == 2%
> [Tue Apr 24 21:46:33.570 2018] I flash_operation:  store_data_in_flash:  128: flash_offset_ptr ===== 214000_

Thanks
dipen.parmar@slscorp.com

dipensls, 2018-04-25:
@agmuntianu 
Can you provide more details on 'silicon error on esp32 3.10' so i can confirm ?

agmuntianu, 2018-10-15:
Hi @igrr,

I saw that changes to the DPORT workaround have been made in V3.1 , do you know if those changes address the hardware encryption problem I have posted here ?

Thank you in advance.

chegewara, 2018-12-17:
Hi, 
i would like to join this issue. Recently i am working on app that is using mqtt and extensively UART. I have only remote access to device that is crashing from time to time (few to 13-16 hours) but i have logs from last crash. Before crash there was no WDT triggered or any unusual behavior, just normal but intensive transmission with UART 2:
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6792
load:0x40078000,len:11356
load:0x40080400,len:6732
entry 0x400807dc
[1B][0;33mW (172) boot: PRO CPU has been reset by WDT.[1B][0m
[1B][0;33mW (172) boot: WDT reset info: PRO CPU PC=0x40084b14[1B][0m
[1B][0;33mW (173) boot: WDT reset info: APP CPU PC=0x40090693[1B][0m
[1B][0;32mI (179) boot: ESP-IDF v3.3-dev-314-gafe4c76b6 2nd stage bootloader[1B][0m
```
I am suspecting its silicon rev 0 wroom devKitC board. I will have opportunity to change board to wrover VB this week i think and i hope this is help me fix the problem.

igrr, 2019-04-10:
@chegewara Are you still seeing this issue? If yes, could you please decode the addresses in this dump using idf_monitor?

chegewara, 2019-04-10:
@igrr i think i dont have this issue anymore, thanks.

ijustwant, 2019-05-06:
Hi
I have the same problem. Here is my screen data after crash.

ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5976
load:0x40078000,len:9040
ho 0 tail 12 room 4
load:0x40080000,len:6120
entry 0x40080334
W (91) boot: PRO CPU has been reset by WDT.
W (91) boot: WDT reset info: PRO CPU PC=0x40092547
W (92) boot: WDT reset info: APP CPU PC=0x40081a1f
I (98) boot: ESP-IDF v3.1-rc1-53509c7b 2nd stage bootloader
I (104) boot: compile time 13:16:14
I (108) boot: Enabling RNG early entropy source...
I (114) boot: SPI Speed      : 40MHz
I (118) boot: SPI Mode       : DIO
I (122) boot: SPI Flash Size : 4MB
I (126) boot: Partition Table:
I (130) boot: ## Label            Usage          Type ST Offset   Length
I (137) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (145) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (152) boot:  2 MicroPython      factory app      00 00 00010000 001f0000
I (160) boot:  3 internalfs       Unknown data     01 82 00200000 00100000
I (167) boot: End of partition table
I (172) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x735b8 (472504) map
I (346) esp_image: segment 1: paddr=0x000835e0 vaddr=0x3ffb0000 size=0x05040 ( 20544) load
I (354) esp_image: segment 2: paddr=0x00088628 vaddr=0x3ffb5040 size=0x00000 (     0) load
I (355) esp_image: segment 3: paddr=0x00088630 vaddr=0x40080000 size=0x00400 (  1024) load
I (362) esp_image: segment 4: paddr=0x00088a38 vaddr=0x40080400 size=0x075d8 ( 30168) load
I (383) esp_image: segment 5: paddr=0x00090018 vaddr=0x400d0018 size=0x154b7c (1395580) map
I (872) esp_image: segment 6: paddr=0x001e4b9c vaddr=0x400879d8 size=0x0d92c ( 55596) load
I (896) esp_image: segment 7: paddr=0x001f24d0 vaddr=0x400c0000 size=0x00648 (  1608) load
I (896) esp_image: segment 8: paddr=0x001f2b20 vaddr=0x50000000 size=0x0098c (  2444) load
I (917) boot: Loaded app from partition at offset 0x10000
I (917) boot: Disabling RNG early entropy source...
I (918) cpu_start: Pro cpu up.
I (921) cpu_start: Starting app cpu, entry point is 0x400812f8
I (0) cpu_start: App cpu up.
I (932) heap_init: Initializing. RAM available for dynamic allocation:
D (939) heap_init: New heap initialised at 0x3ffae6e0
I (944) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
D (950) heap_init: New heap initialised at 0x3ffbd3a8
I (955) heap_init: At 3FFBD3A8 len 00022C58 (139 KiB): DRAM
I (961) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (967) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
D (974) heap_init: New heap initialised at 0x40095304
I (979) heap_init: At 40095304 len 0000ACFC (43 KiB): IRAM
I (985) cpu_start: Pro cpu start user code
D (997) clk: RTC_SLOW_CLK calibration value: 3138086
I (114) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
D (115) heap_init: New heap initialised at 0x3ffe0440
D (119) heap_init: New heap initialised at 0x3ffe4350
D (124) MicroPython: Entry
D (1127) MicroPython: SPIRAM: Disabled
D (1127) nvs: nvs_flash_init_custom partition=nvs start=9 count=6
D (1143) MicroPython: Configure stack
D (1143) nvs: nvs_open_from_partition MPY_NVM 1
D (1143) nvs: nvs_get MPY_StackSize 4
D (1144) nvs: nvs_get_str_or_blob MpyTimeZone
D (1148) nvs: nvs_get_str_or_blob MpyTimeZone
D (1153) MicroPython: MPy stack: 0x3ffc0d78 - 0x3ffc5d80 (20488)
D (1159) MicroPython: Configure heap
D (1162) nvs: nvs_get MPY_HeapSize 4
D (1166) MicroPython: MPy heap: 0x3ffc5d84 - 0x3ffd7dc4 (73728)
D (1172) MicroPython: Main task exit, stack used: 1392
I (1174) MicroPython: [=== MicroPython FreeRTOS task started (sp=3ffc5cd0) ===]

Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 1048576; Flash address: 0x200000]
----------------
Filesystem size: 956416 B
           Used: 8448 B
           Free: 947968 B
----------------
MicroPython ESP32_LoBo_v3.2.24 - 2018-09-06 on ESP32 board with ESP32
Type "help()" for more information.
>>>

KonstantinKondrashov, 2019-05-08:
Hi @ijustwant! Could you provide backtrace for this? 
```
W (91) boot: WDT reset info: PRO CPU PC=0x40092547
W (92) boot: WDT reset info: APP CPU PC=0x40081a1f
```
Thanks.

Alvin1Zhang, 2019-05-10:
@ijustwant Hi, could you provide backtrace for this? Thanks.

Alvin1Zhang, 2019-09-29:
@ijustwant Hi, thanks for reporting and sorry for the slow turnaround. Would you please help provide backtrace for this? Thanks.

KonstantinKondrashov, 2019-10-12:
Hi @ijustwant!
I noticed that you use the `v3.1-rc1-53509c7b` branch for your app. The `v3.1-rc1-53509c7b` does not have a fix (https://github.com/espressif/esp-idf/commit/ada3bc3f). I think your issue linked with it. Could you update your branch to the latest v3.1?
Thanks.

Alvin1Zhang, 2019-10-12:
@ijustwant Thanks for reporting. Please help try with the commit suggested by @KonstantinKondrashov . Feel free to reopen if the issue still happens. Thanks.

closed 2019-10-12 by Alvin1Zhang