espressif/esp-idf #1723: esp_wifi_get_config() not setting necessary parameters / documentation unclear about how to use

chmorgan, 2018-03-15:
I was debugging an issue where in one code path my system was able to update the WIFI_IF_STA configuration and connect to an AP, but in another code path the same configuration failed.

The issue turned out to be that **esp_wifi_get_config()** doesn't appear to set the **wifi_config_t.sta.scan_method** parameter. The original value is preserved. The code in both cases was doing something like:

```
wifi_config_t config;
esp_wifi_get_config(WIFI_IF_STA, &config);

// update ssid and password

esp_wifi_set_config(WIFI_IF_STA, &config);
```

and this was failing to join the AP because **scan_method** was set to **0**.

In both code paths **wifi_config_t** is on the stack. In the code path that worked the value of **scan_method** was something like -17763234. In the other code path the value just happened to be 0 and this caused the issue.

Should this be better documented in esp_wifi_get_config()? Should esp_wifi_get_config() set the scan_method parameter when WIFI_IF_STA is used?

XinDeng11, 2018-05-07:
Hi， @chmorgan, the scan_method, sort_method and threshold parameters are added in 3.0, we have an example about how to use the feature, but didn't add the parameters in the get_config API, we will add it later, thank you for the advice

XinDeng11, 2018-05-11:
Hi@chmorgan, we have and the scan relative parameters in the API, you can found it after 3.1 release, if it earge, you can pull down the commit f3df30f9 after it merge, it may need few days to merge for review and test

closed 2018-05-19 by igrr