espressif/esp-idf #1725: SSL_read in openssl_server_example

jfdevmoto, 2018-03-15:
Hello,
It looks like SSL_read function blocks in case the wifi connection is broken.
I have modified the code to run ESP32 in wifi AP mode and everythink works ok, however if the remote STA leaves after "SSL server read message" log, then it looks like the SSL_read function never returns.
I have tried to call SSL_shutdown from the wifi connection state callback but this has no effect and the function still never returns.
Is it normal behavior for this function ? (I could not browse through the SSL_read code that is too macro confusing for me...).
What can I do to reset the server then ? shall I kill the task ? will it free all resources ?
Is there another SSL function to use such as SSL_free ?
I see there is a Set_shutdwon mode function but the code documentation is very limited for this openSSL component...will try to understand it from openssl.org...
Thanks in advance for your advise.

agmuntianu, 2018-03-16:
have you tried `mbedtls_ssl_session_reset(&ssl);` ?

jfdevmoto, 2018-03-16:
No,
Thanks for the hint.
I understood this behavior is inherent to blocking sockets and there are a
few topics on the internet, such as using a select to check file change
before calling the read but I did not understand how to implement it in
this case.
I have tried to check available bytes with SSL_pending() before reading but
it always returns 0.
So I eventually just checked the wifi connection state before reading as a
dirty workaround.

projectgus, 2019-03-01:
Hi @jfdevmoto ,

Sorry noone got back to you about this question earlier.

TCP is designed to be very resilient about connection issues, including potentially maintaining a connection while the underlying Wi-Fi connection goes down and comes back up. These can take a long time to timeout by themselves.

The solution will be for ESP-IDF to add support for SSL_set_timeout() to the openssl layer, so it's possible to set a timeout for read/write operations.

Currently we don't have this support in the openssl layer. If you use mbedTLS API directly, or [esp_tls API](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/protocols/esp_tls.html), then it is possible to set a timeout.

closed 2021-02-01 by Alvin1Zhang