espressif/esp-idf #1726: CPU halted

GregCle, 2018-03-15:
Hi,

I have an error that rarely happens (about 1 time of 50). My code is in micropython, but the error seems deep in the code written in C.

When I start my code, I try to connect to an access point and after I start the mqtt protocol.

During startup, a network is searched and an ip address is found (See picture)

[image]

```
def Init_Connection():
    global Adr_Ip_Asept3X, ID, Ad_Ip_MQTT, Net_Name, Pass
    ID = 'Asept3x_'
    wlan.active(False)
    while wlan.ifconfig()[0] !='0.0.0.0':
        time.sleep(1)

    wlan.active(True)
    wlan.connect(Net_Name,Pass)
    #----- Attend connexion = True ------
    while wlan.isconnected() != True :
        Toggle(LED_Error)
        time.sleep_ms(500)
    time.sleep(2)
    network.telnet.start()
    #-------- Recup茅ration info IP adresse + creation ID
    Adr_Ip_Asept3X = wlan.ifconfig()[0]
    Ad_Ip_MQTT = wlan.ifconfig()[2]
    ID += Adr_Ip_Asept3X.split('.')[3]
    print(('Asept3x IP: ' + Adr_Ip_Asept3X))
    print(('Broker IP: ' + Ad_Ip_MQTT))
    print(('ID: ' + ID))
```
This problem happens a few times and it is hard to recreate.

Do you have an idea how to fix my problem?

Thank you

agmuntianu, 2018-03-16:
I see you are using a display , could you try to reproduce the error if you have the display disabled in the code ?

GregCle, 2018-03-16:
I have no display. It's strange !

 Here are the libraries I use

```
import time
import network
import machine
import os
import micropython
from machine import Pin
from machine import PWM
import _thread
import gc
```

agmuntianu, 2018-03-16:
yes, but you are importing the whole micropython :) , which has a display in it. Also you are importing firstly the whole machine module and then parts of machine module afterwards, which might be a waste .

GregCle, 2018-03-16:
Ho you just teach me something :-)

I will make my changes and see if the problem continues.

This does not explain why the problem occurs about 1 time of 50

agmuntianu, 2018-03-16:
hmm, I might be wrong , those files seem to be part of the core of micropython ....

GregCle, 2018-03-16:
I'm going to clean the code (library) and cycle the machine to see if the problem is still present. I'll be a little more today or Monday.

Thanks for the solution track

agmuntianu, 2018-03-16:
about the 1/50 error , if it is the `xQueueGenericSend` the cause, then it can be that it tries somewhere to put on a queue that has been freed , and of course , sometimes the timing is off enough to cause an issue. So it might be a bug in micropython

GregCle, 2018-03-16:
I made the proposed changes and for the moment everything works (I am about 100 restarts) :-)

I noticed something strange, why I have text at startup that appears after the execution of part of my code, is it normal ?
```
#Text Start
#...

D (132) heap_init: New heap initialised at 0x3ffe0440
D (138) heap_init: New heap initialised at 0x3ffe4350
D (143) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (158) nvs: nvs_flash_init_custom partition=nvs start=9 count=4
D (178) intr_alloc: Connected src 34 to int 3 (cpu 1)
D (189) intr_alloc: Connected src 22 to int 4 (cpu 1)

Internal FS (SPIFFS): Mounted on partition 'internalfs' [size: 1114112; Flash address: 0x2F0000]
----------------
Filesystem size: 1016576 B
           Used: 11264 B
           Free: 1005312 B
----------------
D (1429) intr_alloc: Connected src 49 to int 9 (cpu 1)
I (1848) phy: phy_version: 380.0, b77d4dc, Jan 24 2018, 17:42:07, 0, 0

#My code

ESP_IP: 192.168.0.39
Broker IP: 192.168.0.100
ID: ESP_IP_39
[RPI] Disconnected
MQTT connection failed
[RPI] Connected

#Text start

FreeRTOS running on BOTH CORES, MicroPython task running on both cores.
Running from OTA_0 partition starting at 0x10000, [MicroPython_1].

 Reset reason: Power on reset
    uPY stack: 64512 bytes
     uPY heap: 3501888/14688/3487200 bytes (in SPIRAM using heap_caps_malloc)

MicroPython ESP32_LoBo_v3.1.18 - 2017-02-07 on ESP32 board with ESP32
Type "help()" for more information.

#My code
>>> MQTT connection failed
[RPI] Connected
```
My code is executed in the main()

igrr, 2018-03-18:
Hi @GregCle, for your final question, i think it is better to ask this on the Micropython issue tracker.

closed 2018-03-18 by igrr