espressif/esp-idf #1729: Trash on console when Power Management enabled

loboris, 2018-03-15:
When `Support for power management` is **enabled** in menuconfig and `Enable dynamic frequency scaling (DFS) at startup` is **NOT enabled**, soon after application start the UART output becomes garbled (corrupt).
If `Enable dynamic frequency scaling (DFS) at startup` is **enabled**, this does not happen.

**This only happens if**  `→ Component config → ESP32-specific → CPU frequency` **is set to _240_ MHz**.

Possible workaround is to set the the CPU frequency to 160 MHz at the application start, then set it back to 250 MHz:
```
#if defined(CONFIG_PM_ENABLE) && !defined(CONFIG_PM_DFS_INIT_AUTO) && defined(CONFIG_ESP32_DEFAULT_CPU_FREQ_240)
esp_pm_config_esp32_t pm_config;
pm_config.max_cpu_freq = RTC_CPU_FREQ_160M;
pm_config.min_cpu_freq = RTC_CPU_FREQ_80M; // or RTC_CPU_FREQ_XTAL
pm_config.light_sleep_enable = false;
esp_pm_configure(&pm_config);
vTaskDelay(2);
pm_config.max_cpu_freq = RTC_CPU_FREQ_240M;
esp_pm_configure(&pm_config);
#endif
```

loboris, 2018-03-18:
Even with this workaround, after changing the max CPU frequency to 160 or 80 MHz, the console output (readable befor the frequency change) becomes unreadable, as if the boudrate changed.
Printing with **printf** works before freq change, not after.
Is REF_TICK used as clock source for the console uart ?

igrr, 2018-03-20:
Thanks for the report @loboris, the fix for this will be in master branch shortly.

closed 2018-03-21 by igrr