espressif/esp-idf #1732: spiffs - Unable to read data from mkspiffs generated filesystem

chmorgan, 2018-03-16:
@igrr 

Using latest esp-idf from about an hour or so ago (7e268ada), with updated submodules.

I've generated and written a spiffs filesystem to an esp32 module, called 'storage':

```
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00010000
I (72) boot:  1 otadata          OTA data         01 00 00019000 00002000
I (80) boot:  2 phy_init         RF data          01 01 0001b000 00001000
I (87) boot:  3 storage          Unknown data     01 82 0001c000 00014000
I (94) boot:  4 factory          factory app      00 00 00030000 00200000
I (102) boot:  5 ota_0            OTA app          00 10 00230000 00200000
I (110) boot:  6 ota_1            OTA app          00 11 00430000 00200000
```

and I'm able to mount as vfs and the total size and used size looks good:

```
D (1544) SPIFFS: page index byte len:         256
D (1544) SPIFFS: object lookup pages:         1
D (1544) SPIFFS: page pages per block:        16
D (1554) SPIFFS: page header length:          5
D (1564) SPIFFS: object header index entries: 95
D (1564) SPIFFS: object index entries:        124
D (1574) SPIFFS: available file descriptors:  10
D (1574) SPIFFS: free blocks:                 16
D (1584) SPIFFS: SPIFFS_info

I (1584) main: spiffs partition size: total: 68021, used: 7530
```

 and I'm able to stat() the expected files to see that they exist:

```
I (1674) aws: pClientCRT /spiffs/certificate.pem.crt found
...
```

And able to fopen() the files as well like fopen("/spiffs/certificate.pem.crt", "r");

But I'm unable to fread() from the files.

I also switched over to using open() and read() and I'm unable to read there as well.

Here is the debug level output from an open() and a read() of that certificate.pem.crt file:

```
D (2174) SPIFFS: SPIFFS_open '/certificate.pem.crt' 08

D (2184) SPIFFS: CACHE_FREE: free cache page 5 pix 00d0

D (2184) SPIFFS: CACHE_ALLO: allocated cache page 5 for pix 0020

D (2194) SPIFFS: CACHE_FREE: free cache page 6 pix 00e0

D (2194) SPIFFS: CACHE_ALLO: allocated cache page 6 for pix 0030

D (2214) SPIFFS: CACHE_FREE: free cache page 7 pix 00f0

D (2214) SPIFFS: CACHE_ALLO: allocated cache page 7 for pix 0032

D (2224) SPIFFS: open: fd 1 is obj id 8006

D (2224) SPIFFS: SPIFFS_read 1 64

D (2224) SPIFFS: read: load objix page 0032:0000 for data spix:0000

D (2234) SPIFFS: read: offset:0 rd:64 data spix:0000 is data_pix:ffff addr:00ffff05

D (2244) SPIFFS: SPIFFS_clearerr

I (2244) aws: bytesRead -1
I (2244) aws: data ?>ȇ
D (2244) SPIFFS: SPIFFS_close 1
```

The code for reading is:
```
    int fd = open(filename, O_RDONLY);
    if(fd == -1)
    {
        ESP_LOGE(TAG, "failed to open '%s'", filename);
    } else
    {
        bytesRead = read(fd, data, 64);
        ESP_LOGI(TAG, "bytesRead %d", bytesRead);
        ESP_LOGI(TAG, "data %s", data);
        close(fd);
    }
```

What could be going on here? I'm not sure where to go in terms of debugging.

igrr, 2018-03-17:
Could you please run `./mkspiffs --version` in your mkspiffs directory and `grep SPIFFS sdkconfig` in your project directory and paste output of both here?

chmorgan, 2018-03-17:
Hi @igrr, of course, included below.

I did read through the documentation on the mkspiffs README.md a few times to ensure at least I had tried to build the spiffs image that would match the sdkconfig settings. I also thought it was likely that the spiffs code was checking in the spiffs image header upon mount to ensure that various parameters matched up.

```
[cmorgan@CHRISTOPHERs-MacBook-Pro mkspiffs]$ ./mkspiffs --version
mkspiffs ver. 0.2.2-2-g1693a95
Build configuration name: generic
SPIFFS ver. 0.3.7-5-gf5e26c4
Extra build flags: (none)
SPIFFS configuration:
  SPIFFS_OBJ_NAME_LEN: 32
  SPIFFS_OBJ_META_LEN: 0
  SPIFFS_USE_MAGIC: 1
  SPIFFS_USE_MAGIC_LENGTH: 1
  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0
```

```
[cmorgan@CHRISTOPHERs-MacBook-Pro app]$ grep SPIFFS sdkconfig
# SPIFFS Configuration
CONFIG_SPIFFS_MAX_PARTITIONS=3
# SPIFFS Cache Configuration
CONFIG_SPIFFS_CACHE=y
CONFIG_SPIFFS_CACHE_WR=y
CONFIG_SPIFFS_CACHE_STATS=
CONFIG_SPIFFS_PAGE_CHECK=y
CONFIG_SPIFFS_GC_MAX_RUNS=10
CONFIG_SPIFFS_GC_STATS=
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=32
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=4
CONFIG_SPIFFS_USE_MTIME=
CONFIG_SPIFFS_DBG=y
CONFIG_SPIFFS_API_DBG=y
CONFIG_SPIFFS_GC_DBG=y
CONFIG_SPIFFS_CACHE_DBG=y
CONFIG_SPIFFS_CHECK_DBG=y
CONFIG_SPIFFS_TEST_VISUALISATION=y
```

negativekelvin, 2018-03-18:
SPIFFS_OBJ_META_LEN: 0

CONFIG_SPIFFS_META_LENGTH=4

Try 0.2.3
https://github.com/igrr/mkspiffs/releases

igrr, 2018-03-18:
Yes unfortunately there are a few parameters (such as NAME_LEN and META_LEN) which SPIFFS does not record/check. We are all eagerly waiting for SPIFFS 0.4 which should resolve these issues. For now use these commands to compare configuration of the app and mkspiffs.

chmorgan, 2018-03-18:
Ahh gotcha, that makes sense. I didn't realize SPIFFS_META_LENGTH was the same as SPIFFS_OBJ_META_LEN.

@igrr should I use 0.2.3 as @negativekelvin suggested or stick with mkspiffs latest master? assuming that I'd set those options in both cases. I'm going to stick with latest master until I hear otherwise and test now.

chmorgan, 2018-03-18:
@igrr, @negativekelvin setting CPPFLAGS="-DSPIFFS_OBJ_META_LEN=4" when building has gotten spiffs working, thanks for the help!

Now onto why I'm seeing aws connection timeout errors.

closed 2018-03-18 by chmorgan

igrr, 2018-03-18:
@chmorgan at the moment the latest master is equal to 0.2.3 tag, so both should work. For future reference, build of mkspiffs version compatible with the default esp-idf config is done here: https://github.com/igrr/mkspiffs/blob/112d65368043f01dd86813cb84e41b1159a34f6b/build_all_configs.sh#L23-L24.

ronnye-roger, 2020-12-21:
Hello,
I am working with a WROOM-02 module, based on ESP8266 chip, using IDF and I'm also trying to make SPIFFS work but I cannot read a file after having prepared the partition content in a .bin file by using mkspiffs and having flashed it to my target SPIFFS partition. The file system is correctly mounted and I successfully run stat to get file information, but I can't read the file content at all.
I checked mkspiffs --version and it says

mkspiffs ver. 0.2.3
Build configuration name: esp-idf
SPIFFS ver. 0.3.7-5-gf5e26c4
Extra build flags: -DSPIFFS_OBJ_META_LEN=4
SPIFFS configuration:
  SPIFFS_OBJ_NAME_LEN: 32
  SPIFFS_OBJ_META_LEN: 4
  SPIFFS_USE_MAGIC: 1
  SPIFFS_USE_MAGIC_LENGTH: 1
  SPIFFS_ALIGNED_OBJECT_INDEX_TABLES: 0

then I checked my sdkconfig and got

CONFIG_SPIFFS_MAX_PARTITIONS=3
CONFIG_SPIFFS_CACHE=y
CONFIG_SPIFFS_CACHE_WR=y
# CONFIG_SPIFFS_CACHE_STATS is not set
CONFIG_SPIFFS_PAGE_CHECK=y
CONFIG_SPIFFS_GC_MAX_RUNS=10
# CONFIG_SPIFFS_GC_STATS is not set
CONFIG_SPIFFS_PAGE_SIZE=256
CONFIG_SPIFFS_OBJ_NAME_LEN=32
CONFIG_SPIFFS_USE_MAGIC=y
CONFIG_SPIFFS_USE_MAGIC_LENGTH=y
CONFIG_SPIFFS_META_LENGTH=4
CONFIG_SPIFFS_USE_MTIME=y
# CONFIG_SPIFFS_DBG is not set
# CONFIG_SPIFFS_API_DBG is not set
# CONFIG_SPIFFS_GC_DBG is not set
# CONFIG_SPIFFS_CACHE_DBG is not set
# CONFIG_SPIFFS_CHECK_DBG is not set
# CONFIG_SPIFFS_TEST_VISUALISATION is not set

and they seem to agree with each other.
I really can't find what's the problem that avoid me to read the file content.

Ronnye