espressif/esp-idf #1733: printf / esp:log in a task

Reiner1210, 2018-03-17:
Hello,,

In the following code the printf statement before the delay does not work.
If I replace it with ESP_LOGI I get a crash

I (288) test: Counter: 0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x40083eca  PS      : 0x00060533  A0      : 0x80084164  A1      : 0x3ffb5c50  
0x40083eca: prvNotifyQueueSetContainer at /home/reiner/data/programm/esp/esp-idf/components/freertos/./queue.c:2037

A2      : 0x3ffb5f1c  A3      : 0x00000000  A4      : 0xffffffff  A5      : 0x00000020  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000001  A9      : 0x3ffb5c40  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060320  A15     : 0x00000005  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000037  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x40083eca:0x3ffb5c50 0x40084161:0x3ffb5c80 0x40084285:0x3ffb5cc0 0x40082c8a:0x3ffb5ce0 0x40082d53:0x3ffb5d10 0x400dfaba:0x3ffb5d30 0x400dfd99:0x3ffb5d50 0x4000bd83:0x3ffb5d70 0x4000117d:0x3ffb5d90 0x400592fe:0x3ffb5db0 0x4005937a:0x3ffb5dd0 0x40058bbf:0x3ffb5df0 0x400d60f9:0x3ffb5e20 0x400db8c2:0x3ffb5e50 0x400dba05:0x3ffb6160 0x40082a35:0x3ffb6190 0x400d1f04:0x3ffb61e0
0x40083eca: prvNotifyQueueSetContainer at /home/reiner/data/programm/esp/esp-idf/components/freertos/./queue.c:2037

0x40084161: xQueueGenericSend at /home/reiner/data/programm/esp/esp-idf/components/freertos/./queue.c:2037

0x40084285: xQueueGiveMutexRecursive at /home/reiner/data/programm/esp/esp-idf/components/freertos/./queue.c:2037

0x40082c8a: lock_release_generic at /home/reiner/data/programm/esp/esp-idf/components/newlib/./locks.c:203

0x40082d53: _lock_release_recursive at /home/reiner/data/programm/esp/esp-idf/components/newlib/./locks.c:215

0x400dfaba: uart_write at /home/reiner/data/programm/esp/esp-idf/components/vfs/./vfs_uart.c:117

0x400dfd99: esp_vfs_write at /home/reiner/data/programm/esp/esp-idf/components/vfs/./vfs.c:274 (discriminator 4)

0x400d60f9: __sprint_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:437

0x400db8c2: _vfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1782 (discriminator 1)

0x400dba05: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c:39

0x40082a35: esp_log_write at /home/reiner/data/programm/esp/esp-idf/components/log/./log.c:215

0x400d1f04: do_it at /home/reiner/data/programm/esp/hello_world/main/./hello_world_main.c:65 (discriminator 1)

void do_it(void *pvParameter)
{
	int i = 0;
    while(1)
	{
	   // ESP_LOGI("test", "Counter: %d\n", i++); --- CRASH
           //printf("Counter: %d\n", i++); -- DO NOT WORK
           vTaskDelay(1000 / portTICK_PERIOD_MS);
          //printf("Counter: %d\n", i++); --  WORK
    }
    vTaskDelete(NULL);
}

void app_main()
{
	ESP_LOGI("test", "xxxx\n");
	TaskHandle_t do_it_task;
	if (xTaskCreate(&do_it, "do_it", configMINIMAL_STACK_SIZE, NULL, 5, &do_it_task) != pdPASS)
		printf("Unable to create task");
}

igrr, 2018-03-17:
Please try changing configMINIMAL_STACK_SIZE to something larger (e.g. 4096) and see if that resolves the issue.

vijaygill, 2018-03-17:
@igrr I too got the same error, changed the MINIMAL_STACK_SIZE to 4096 and it worked. Going forward, does that mean I have to set it for every task which uses logging?

Reiner1210, 2018-03-17:
Yes all now works fine - even the printf calls beforew and after.
What is the recommended stack size - do we have a define for it ?

Thanks
Reiner

closed 2018-03-17 by Reiner1210

chegewara, 2018-03-17:
@vijaygill Yes, printf function is stack consuming function and require bigger than minimal  stack size.