espressif/esp-idf #1734: "wifi:out of memory" when using esp_wifi_80211_tx.

jixunx, 2018-03-17:
Refer to this post of viewtopic.php? F=19&amp;t=3017&amp;p=15265&amp;hilit=sniffer#p15265.I wrote a code:
`/**
 * Copyright (c) 2018, jason <jason@rdyheath.com>
 * ESP32
 * esp_wifi_80211_tx.
 */
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_wifi_types.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include "lwip/err.h"
#include "esp_log.h"

#define USERDATASTZE 961
#define PACK_NUM_PER_FRAME 120

//rates:
/*
0 - B 1Mb CCK
1 - B 2Mb CCK
2 - B 5.5Mb CCK
3 - B 11Mb CCK
4 - XXX Not working. Should be B 1Mb CCK SP
5 - B 2Mb CCK SP
6 - B 5.5Mb CCK SP
7 - B 11Mb CCK SP

8 - G 48Mb ODFM
9 - G 24Mb ODFM
10 - G 12Mb ODFM
11 - G 6Mb ODFM
12 - G 54Mb ODFM
13 - G 36Mb ODFM
14 - G 18Mb ODFM
15 - G 9Mb ODFM

16 - N 6.5Mb MCS0
17 - N 13Mb MCS1
18 - N 19.5Mb MCS2
19 - N 26Mb MCS3
20 - N 39Mb MCS4
21 - N 52Mb MCS5
22 - N 58Mb MCS6
23 - N 65Mb MCS7

24 - N 7.2Mb MCS0 SP
25 - N 14.4Mb MCS1 SP
26 - N 21.7Mb MCS2 SP
27 - N 28.9Mb MCS3 SP
28 - N 43.3Mb MCS4 SP
29 - N 57.8Mb MCS5 SP
30 - N 65Mb MCS6 SP
31 - N 72Mb MCS7 SP
*/
#define RATE_MCS4_SP 28
typedef union {
     uint8_t fix_rate;
     uint8_t b5;
     uint8_t b4;
 
     struct {
         uint8_t b3;
         uint8_t b2;
     } b1;
 
     struct {
         uint32_t a1;
         uint8_t  a2;
         uint8_t  a3;
         uint8_t  a4;
         uint8_t  a5;
         struct {
             uint8_t a6;
             uint8_t a7;
         } a8[4];
         uint8_t a9;
         uint8_t a10;
         uint8_t a11;
         uint8_t a12;
     } a13;
 
 } wifi_internal_rate_t;

uint8_t packetH[32+USERDATASTZE]={
0x80, 0x00, //Frame Control 
0x00, 0x00, //Duration
0xff, 0xff, 0xff, 0xff, 0xff, 0xff, //Receiver Address= Destination Address 
0x30, 0xAE, 0xA4, 0x05, 0x17, 0x64, //Transmitter Address= Source Address
0x01, 0x02, 0x03, 0x04, 0x05, 0x06, //BSSID
0x00, 0x00,                         //Sequence control
0x01, 0x02, 0x02, 0x03,0x03,0x03,0x04,0x04, //Body
};

extern   esp_err_t esp_wifi_80211_tx(wifi_interface_t ifx, const void *buffer, int len, bool en_sys_seq);  
extern   esp_err_t esp_wifi_internal_set_rate(int a, int b, int c, wifi_internal_rate_t *d);

static esp_err_t event_handler(void *ctx, system_event_t *event);

static void wifi_init(void)
{
   nvs_flash_init();
    tcpip_adapter_init();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
   ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
   //ESP_ERROR_CHECK( esp_wifi_set_country(&wifi_country) ); /* set country for channel range [1, 13] */
   ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_NULL) );
    ESP_ERROR_CHECK( esp_wifi_start() );
   wifi_internal_rate_t rate;  
   rate.fix_rate = RATE_MCS4_SP;
   esp_wifi_internal_set_rate(100, 1, 4, &rate);
}

void app_main(void)
{
   //esp_log_level_set("*", ESP_LOG_NONE);
   static unsigned int fail_count=0;
   struct timeval time_start;
   struct timeval time_end;
   struct timeval result_time;
   static uint8_t data[USERDATASTZE];
   /* setup */
   wifi_init();
   while(true)
   {
      gettimeofday(&time_start,NULL);
      for(uint8_t i=0;i<PACK_NUM_PER_FRAME;i++)
      {
         data[0]=i;
         memcpy(packetH+32,data,sizeof(data));
         int result = esp_wifi_80211_tx(WIFI_MODE_STA, packetH, sizeof(packetH), true);
   
         switch (result) 
         {
         /*
            case ERR_OK:
               //Serial.println("ERR_OK"); 
               printf("ESPOK\n");
               break;
         */
            case ERR_MEM:
               //Serial.println("ERR_MEM"); 
               printf("ERR_MEM\n");
               fail_count++;
               break;
            case ERR_IF:
               printf("ERR_IF\n");
               break;
            case ERR_ARG: 
               printf("ERR_ARG\n");
               break;
            //default:
         //      printf("result=%d",result);
         }
      
      }
      
      gettimeofday(&time_end,NULL);
      //caculat the time of sending 115200 bytes(921kbits)
      timersub(&time_end, &time_start, &result_time);
      printf("result_time usec is %ld\n",result_time.tv_usec);
   }
}

esp_err_t
event_handler(void *ctx, system_event_t *event)
{
   return ESP_OK;
}`
log:
E (480597) wifi: out of memory!
E (480600) wifi: out of memory!
E (480603) wifi: out of memory!
E (480605) wifi: out of memory!
E (480608) wifi: out of memory!
E (480611) wifi: out of memory!
E (480614) wifi: out of memory!
E (480617) wifi: out of memory!
E (480620) wifi: out of memory!
E (480623) wifi: out of memory!
E (480626) wifi: out of memory!
E (480629) wifi: out of memory!
E (480631) wifi: out of memory!
E (480634) wifi: out of memory!
E (480637) wifi: out of memory!
result_time usec is 43038

Through the capture from the promiscuous mode of esp32, this code can work normally.But "wifi:out of memory'"need to be handled.My question is :
1`Espressif's engineer told me that the future IDF was released to improve this phenomenon. Does the latest version of IDF on git improve this phenomenon?
2`It looks like "wifi:out of memory" printed in the low layer . But this error does not seem to be passed to the application layer, and this error can not be obtained by the return value of esp_wifi_80211_tx. How to make the "wifi:out of memory" error return to the application layer ，how the application layer monitors "wifi:out of memory error")？

liuzfesp, 2018-03-19:
Hi @jixunx , for your questions:
1.  Yes, you can "git pull" the latest IDF, we already improved esp_wifi_80211_tx. But the API declaration is still in review status.
2.  If esp_wifi_80211_tx print "out of memory", it returns ESP_ERR_WIFI_NO_MEM, not ERR_MEM

jixunx, 2018-03-19:
Thanks for your reply!
Is there a way that the application layer can detect that WiFi TX buffer is enough ,then send data through esp_wifi_80211_tx?

liuzfesp, 2018-03-19:
The only way to detect that TX buffer is not enough is the return value. Generally the application can choose to delay some time if the TX buffer is not enough, and then re-send the packets.

closed 2018-03-20 by igrr

malaimoo, 2018-03-23:
@liuzfesp Hello, Could you give me some information about new "esp_wifi_80211_tx" api usage? After I pull the newest IDF, this function run error :invalid interface 1

jixunx, 2018-03-23:
esp_wifi_80211_tx()  is in libnet80211.Add "extern esp_err_t esp_wifi_80211_tx(wifi_interface_t ifx, const void *buffer, int len, bool en_sys_seq);"

liuzfesp, 2018-03-23:
/**
  * @brief     Send raw ieee80211 data
  *
  * @attention Currently only support for sending beacon/probe request/probe response/action and non-QoS
  *            data frame
  *
  * @param     ifx interface if the Wi-Fi mode is Station, the ifx should be WIFI_IF_STA. If the Wi-Fi
  *            mode is SoftAP, the ifx should be WIFI_IF_AP. If the Wi-Fi mode is Station+SoftAP, the
  *            ifx should be WIFI_IF_STA or WIFI_IF_AP. If the ifx is wrong, the API returns ESP_ERR_WIFI_IF.
  * @param     buffer raw ieee80211 buffer
  * @param     len the length of raw buffer, the len must be <= 1500 Bytes and >= 24 Bytes
  * @param     en_sys_seq indicate whether use the internal sequence number. If en_sys_seq is false, the
  *            sequence in raw buffer is unchanged, otherwise it will be overwritten by WiFi driver with
  *            the system sequence number.
  *            Generally, if esp_wifi_80211_tx is called before the Wi-Fi connection has been set up, both
  *            en_sys_seq==true and en_sys_seq==false are fine. However, if the API is called after the Wi-Fi
  *            connection has been set up, en_sys_seq should be true.
  *
  * @return
  *    - ESP_OK: success
  *    - ESP_ERR_WIFI_IF: Invalid interface
  *    - ESP_ERR_WIFI_ARG: Invalid parameter
  *    - ESP_ERR_WIFI_NO_MEM: out of memory
  */

esp_err_t esp_wifi_80211_tx(wifi_interface_t ifx, const void *buffer, int len, bool en_sys_seq);

malaimoo, 2018-03-23:
@liuzfesp I set the wifi mode to STA `esp_wifi_set_mode(WIFI_MODE_STA)` and send frame using `esp_wifi_80211_tx(WIFI_IF_STA....` but  It always comes to error ` Invalid interface`

liuzfesp, 2018-03-23:
@ malaimoo do you call esp_wifi_start to start the WiFi?