espressif/esp-idf #1739: adc - offset at 0V

Reiner1210, 2018-03-18:
Hello,

I do the following`:
`
`	adc1_config_width(ADC_WIDTH_12Bit);
    adc1_config_channel_atten(ADC1_CHANNEL_0, ADC_ATTEN_DB_6);
	
	esp_adc_cal_characteristics_t *adc_chars = malloc(sizeof(esp_adc_cal_characteristics_t));
    esp_adc_cal_value_t val_type = esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN_DB_6, ADC_WIDTH_BIT_12, 1098, adc_chars);
    /*
	//Check type of calibration value used to characterize ADC
	if (adc2_vref_to_gpio(GPIO_NUM_25) != ESP_OK)
	{
		ESP_LOGE("adc", "adc2_vref_to_gpio failed");
	}
	if (val_type == ESP_ADC_CAL_VAL_EFUSE_VREF)
		{ESP_LOGI("adc", "eFuse Vref");}
    else if (val_type == ESP_ADC_CAL_VAL_EFUSE_TP)
		{ESP_LOGI("adc", "Two Point");}
    else
		{ESP_LOGI("adc", "Default");}
	*/
    
	int raw;
	int voltage;
    while(1)
	{
		raw = adc1_get_raw(ADC1_CHANNEL_0);
		voltage = esp_adc_cal_raw_to_voltage(raw, adc_chars);
		ESP_LOGI("adc", "Raw: %d, Voltage: %d", raw, voltage);
        vTaskDelay(500 / portTICK_PERIOD_MS);
    }
	vTaskDelete(NULL);
`

the 1098 V was measured with a multimeter, val_type = Default

At 0 V at the input I get a raw value 0 and a voltage at 0.170 mV
I (318283) adc: Raw: 0, Voltage: 107
Measurement at the pin says 0V

When there is 1.56V at the Pin I get the following
I (449783) adc: Raw: 3347, Voltage: 1551

So I think adc needs to be calibrated before using. 

I expected that is also 0 !

```

```

negativekelvin, 2018-03-18:
https://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/adc.html#adc-calibration

If you look at the graph it shows the adc can't really measure below 100-150mv

Reiner1210, 2018-03-18:
Ok,

but also in the mid range there is a (constant) difference between 
measured and displayed values.

PS: Why does the ADC not work with low voltages ?

Reiner

closed 2018-03-19 by Reiner1210