espressif/esp-idf #1741: aws thing_shadow example returning -28 (MQTT_REQUEST_TIMEOUT_ERROR)

chmorgan, 2018-03-19:
I ran into this in my application. I set up the thing_shadow example with my device credentials, iot endpoint, and wifi settings and the same issue is occurring.

Error -28 is listed as MQTT_REQUEST_TIMEOUT_ERROR on http://aws-iot-device-sdk-embedded-c-docs.s3-website-us-east-1.amazonaws.com/aws__iot__error_8h.html#a329da5c3a42979b72561e28e749f6921a4500aeaeb0a31f88752de87e73fe3de0

```
I (253) wifi: wifi power manager task: 0x3ffc6fd4 prio: 21 stack: 2560
I (263) shadow: Setting WiFi configuration SSID xxxxxxxxxx...
I (343) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (343) wifi: mode : sta (30:ae:a4:5d:95:04)
I (343) shadow: AWS IoT SDK Version 2.2.1-
I (1073) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
I (2053) wifi: state: init -> auth (b0)
I (2053) wifi: state: auth -> assoc (0)
I (2063) wifi: state: assoc -> run (10)
I (2093) wifi: connected with mylinksys, channel 6
I (2653) event: sta ip: 10.0.1.19, mask: 255.255.255.0, gw: 10.0.1.1
I (2653) shadow: Shadow Init
I (2653) shadow: Shadow Connect
I (2653) aws_iot: Loading embedded CA root certificate ...
I (5063) wifi: pm start, type:0

E (27653) shadow: aws_iot_shadow_connect returned error -28, aborting...
abort() was called at PC 0x400d28e7 on core 1
```

What might be going on here? The certificate and device are active on the AWS IoT portal.

Can someone at espressif verify that this example still works with the latest esp-idf master (7e268ad)?

chmorgan, 2018-03-21:
@mahavirj would you be someone that could advise?

mahavirj, 2018-03-21:
@chmorgan 
Thanks for notifying, I can definitely try to help you. I tried on latest esp-idf master (https://github.com/espressif/esp-idf/commit/84788230392d0918d3add78d9ccf8c2bb7de3152) but I did not encounter any issue.

I am presuming that you mean vanilla thing_shadow example with default configuration from IDF, please see below log,
```
I (274) shadow: AWS IoT SDK Version 2.2.1-
I (394) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1384) wifi: state: init -> auth (b0)
I (1384) wifi: state: auth -> assoc (0)
I (1394) wifi: state: assoc -> run (10)
I (1404) wifi: connected with XXXX, channel 1
I (3114) event: sta ip: 192.168.1.127, mask: 255.255.255.0, gw: 192.168.1.1
I (3114) shadow: Shadow Init
I (3114) shadow: Shadow Connect
I (4394) wifi: pm start, type:0

I (11374) shadow: =======================================================================================
I (11374) shadow: On Device: window state false
I (11374) shadow: Update Shadow: {"state":{"reported":{"temperature":25.500000,"windowOpen":false}}, "clientToken":"myesp32-0"}
I (14214) shadow: *****************************************************************************************
I (14214) shadow: Stack remaining for task 'aws_iot_task' is 1244 bytes
I (15224) shadow: Update accepted
```
Could you please help to provide more information, e.g. SDK configuration that you are using?

chmorgan, 2018-03-21:
Hi @mahavirj.

Attached. Let me update to latest esp-idf and retry.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/1833468/sdkconfig.txt)

chmorgan, 2018-03-21:
Doh, maybe I'm missing a security policy. I just spotted that in the top level README file. Let me check that.

chmorgan, 2018-03-21:
@mahavirj alright, after putting a policy in place I'm able to push. That error message wasn't particularly helpful. I appreciate you retesting.

closed 2018-03-21 by chmorgan